Heltec-Aaron-Lee/WiFi_Kit_series #2: unable to print WiFi.localIP().toString() to oled screen

SmUrF3R5, 2017-11-11:
display.print(WiFi.localIP().toString(),3,0);

when i compile with this code get this error

/tmp/arduino_modified_sketch_829310/AdvancedWebServer.ino: In function 'void setup()':
AdvancedWebServer:129: error: no matching function for call to 'OLED::print(String, int, int)'
   display.print(WiFi.localIP().toString(),3,0);
                                              ^
/tmp/arduino_modified_sketch_829310/AdvancedWebServer.ino:129:46: note: candidate is:
In file included from /tmp/arduino_modified_sketch_829310/AdvancedWebServer.ino:35:0:
/home/arduino-1.8.5/hardware/heltec/esp8266/libraries/OLED/OLED.h:32:8: note: void OLED::print(char*, uint8_t, uint8_t)
   void print(char *s, uint8_t r=0, uint8_t c=0);
        ^
/home/arduino-1.8.5/hardware/heltec/esp8266/libraries/OLED/OLED.h:32:8: note:   no known conversion for argument 1 from 'String' to 'char*'
exit status 1
no matching function for call to 'OLED::print(String, int, int)'

stelb, 2017-11-13:
because print takes a pointer to char and toString returns an Object String.

display.print(WiFi.localIP().toString().c_str(),3,0);

should work

see also:
https://forum.arduino.cc/index.php?topic=228884.msg3102705#msg3102705

SmUrF3R5, 2017-11-14:
hmm yeah sorry I am new to this platform and C. It seems i have to do this
display.print((char*) WiFi.localIP().toString().c_str(),3,0);

closed 2017-11-14 by SmUrF3R5