Heltec-Aaron-Lee/WiFi_Kit_series #4: WiFiScan Example

davidbitton, 2017-11-20:
The WiFiScan example is throwing the following in the Serial Monitor:

```

Exception (3):
epc1=0x40100473 epc2=0x00000000 epc3=0x00000000 excvaddr=0x4001022c depc=0x00000000

ctx: cont 
sp: 3ffef6d0 end: 3ffefa20 offset: 01a0

>>>stack>>>
3ffef870:  feefeffe feefeffe feefeffe feefeffe  
3ffef880:  feefeffe feefeffe feefeffe 3ffef980  
3ffef890:  00000484 00000484 000003fe 401004f4  
3ffef8a0:  00000000 00001000 000003fe 401079b8  
3ffef8b0:  40004b31 3ffef8e0 0000001c 40228c1d  
3ffef8c0:  40105212 40228d05 3fff0b0c 000003ff  
3ffef8d0:  000003fd 3ffef980 3fff0b0c 000003fd  
3ffef8e0:  ffffff01 55aa55aa 00000009 0000001c  
3ffef8f0:  0000001c 000000ef 000000ef 000003ff  
3ffef900:  402290f4 3fff0b0c 3fff0b0c 000000ff  
3ffef910:  00000001 3ffef9a0 4022927b 00000008  
3ffef920:  3fff0b0c 000000ff 3ffef980 00000000  
3ffef930:  3fff0bcc 3ffef9e1 00000001 40229308  
3ffef940:  3ffef980 3fff0b0c 3fffdad0 3ffee9ec  
3ffef950:  3ffef9a0 3fff6e44 3fff0b0c 3fffdad0  
3ffef960:  40229344 3ffee9c0 00000000 feefeffe  
3ffef970:  402021f0 feefeffe feefeffe feefeffe  
3ffef980:  feefef00 feefeffe feefeffe 00000000  
3ffef990:  402018cc 00000100 00000001 40201380  
3ffef9a0:  40201b00 3fff0a04 3fff09d4 40201c88  
3ffef9b0:  0001c200 0000001c 00000003 3fff09ec  
3ffef9c0:  feefeffe feefeffe feefeffe 3ffee9ec  
3ffef9d0:  40228956 00000001 3ffee7c8 3fffdad0  
3ffef9e0:  4020204b 0000001c 00000000 feefeffe  
3ffef9f0:  feefeffe 3ffee7c8 3ffee9c0 40201ece  
3ffefa00:  feefeffe 00000000 3ffee9e4 40202b84  
3ffefa10:  feefeffe feefeffe 3ffeea00 40100718  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset
```

What should i be looking for?

Heltec-Aaron-Lee, 2017-11-21:
Witch board you have?

Seems it's have a FLASH problem, can you try download some other program such as OLED or Blinking, firstly we should confirm the FLASH is OK.

davidbitton, 2017-11-21:
It's a Wifi Kit 8. I was able to get it working again by pushing a simple test script. Now we're good.

closed 2018-02-23 by Heltec-Aaron-Lee