Heltec-Aaron-Lee/WiFi_Kit_series #5: Unknown Battery Connector

davidbitton, 2017-11-27:
The connector type for the battery is not published. What is the connector type? Thank you.

Heltec-Aaron-Lee, 2017-11-28:
That's 1.25mm connector
[image]

https://item.taobao.com/item.htm?spm=a1z10.3-c.w4002-17001064404.9.7a36c72w3CJfz&id=558403816604

davidbitton, 2017-11-28:
@Heltec-Aaron-Lee JST does not have a model "1.25". What is the p/n for the header on the board. I'll figure out the rest. Thanks!

DeuxVis, 2017-11-28:
The female half in [that lot](https://www.ebay.fr/itm/20-Paire-Micro-JST-1-25mm-2-Broches-Mâle-Femelle-Connecteur-Avec-Câbles-100mm/182462448865) is what you are looking for, just received and tested they plug fine.

davidbitton, 2017-11-28:
@DeuxVis ok, so then can you tell me which battery (preferably in Amazon) to purchase?

DeuxVis, 2017-11-29:
I haven't done any battery testing yet, but the guy who designed [that printable enclosure](https://www.thingiverse.com/thing:2581459/#comments) has. Have a look in the comment section someone asked him about the battery model.

This said I tend to believe any single cell lipo should do. For reference some [aliexpress seller pictures](https://ae01.alicdn.com/kf/HTB1mfl3cnAKL1JjSZFCq6xFspXaf.jpg) show a 1000mAh one used.

stdunbar, 2017-12-15:
I'm using [this](https://www.aliexpress.com/item/Lolin-ESP32-OLED-V2-0-Pro-ESP32-OLED-wemos-pour-Arduino-ESP32-OLED-WiFi-Modules-Bluetooth/32822105291.html) with [this battery](https://www.adafruit.com/product/2750) and, depending on your needs, I'm running for a couple of hours.

davidbitton, 2017-12-15:
Is there something I need to do in order for the battery to work? Do I need
to wire in a switch?

chrisdreid, 2019-07-24:
When I plug in a battery nothing happens... how do you get the battery to work?

davidbitton, 2019-07-24:
No idea. Never worked for me before.

jimmybenz, 2019-09-15:
I think it’s a Molex PicoBlade, 1.25mm.

mik00, 2019-12-13:
I think Jimmy is right.  See this. https://www.mouser.com/datasheet/2/276/0532617002_PCB_HEADERS-1510014.pdf   The picture shows a 4-pin, but the ...7002 appears to be 2-pin judging from the downloadable step model.