Heltec-Aaron-Lee/WiFi_Kit_series #6: ESP32 WiFi Lora Board - deep-sleep power consumption

rmh78, 2017-11-27:
**Hardware:**

Board: Heltec ESP32 WiFi Lora Board Core 
Installation/update date: 25/nov/2017
IDE name: Arduino IDE
Flash Frequency: 80Mhz
Upload Speed: 921600

**Description:**

I want to set the esp32 to deep sleep mode and measure the power consumption. The esp32 specification says that the power consumption should be in the µA range. My measurement is between 3mA and 10mA in deep-sleep mode. I'm powering the board with 3.7V lipo on the 3.3V input pin.

**Sketch:**

```
#include "esp_sleep.h"

#define uS_TO_S_FACTOR 1000000  /* Conversion factor for micro seconds to seconds */
#define TIME_TO_SLEEP  60       /* Time ESP32 will go to sleep (in seconds) */

void setup() {
  Serial.begin(115200);
  delay(1000); 

  esp_sleep_enable_timer_wakeup(TIME_TO_SLEEP * uS_TO_S_FACTOR);
  esp_sleep_pd_config(ESP_PD_DOMAIN_MAX, ESP_PD_OPTION_OFF);
  esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_OFF);
  esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_SLOW_MEM, ESP_PD_OPTION_OFF);
  esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_FAST_MEM, ESP_PD_OPTION_OFF);
}

void loop(){
  delay(5000);
  esp_deep_sleep_start();
}
```

Heltec-Aaron-Lee, 2017-11-28:
Seems you did't turn off the OLED's charger pump and set OLED into sleep mode, you may test with fellow codes:

display.sendCommand(0x8D); //into charger pump set mode
display.sendCommand(0x10); //turn off charger pump
display.sendCommand(0xAE); //set OLED sleep

urs8000, 2017-11-28:
> display.sendCommand(0x8D); 
what additional library and initialization is needed for the three commands?
thanks for your help & info

rmh78, 2017-11-28:
I changed my sample program and added the lines of code to turn off the OLED charger pump and the OLED itself. But the power consumption stays the same (about 5mA). Any other suggestions to reduce power consumption down in the µA range. It would be great if you can provide us with an arduino sketch.

This is my changed program. Please double-check it.

```
#include <Wire.h>
#include "SSD1306.h"
#include "esp_sleep.h"

#define uS_TO_S_FACTOR 1000000  /* Conversion factor for micro seconds to seconds */
#define TIME_TO_SLEEP  20       /* Time ESP32 will go to sleep (in seconds) */
RTC_DATA_ATTR int bootCount = 0;

SSD1306 display(0x3c, 4, 15);

void setup() 
{
  Serial.begin(115200);
  delay(1000); 

  initOled();
  display.drawString(0, 0, "hello world");
  display.display();
  delay(2000); 

  sendCommand(0x8D); //into charger pump set mode
  sendCommand(0x10); //turn off charger pump
  sendCommand(0xAE); //set OLED sleep

  delay(5000);
  esp_sleep_enable_timer_wakeup(TIME_TO_SLEEP * uS_TO_S_FACTOR);
  esp_sleep_pd_config(ESP_PD_DOMAIN_MAX, ESP_PD_OPTION_OFF);
  esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_OFF);
  esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_SLOW_MEM, ESP_PD_OPTION_OFF);
  esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_FAST_MEM, ESP_PD_OPTION_OFF);
  esp_deep_sleep_start();
}

void loop()
{
}

void initOled() 
{
  pinMode(16,OUTPUT);
  digitalWrite(16, LOW);    // set GPIO16 low to reset OLED
  delay(50); 
  digitalWrite(16, HIGH);   // while OLED is running, must set GPIO16 in high

  display.init();
  display.flipScreenVertically();
  display.setFont(ArialMT_Plain_10);
  display.clear();
}

void sendCommand(unsigned char command)
{
  Wire.beginTransmission(0x3C); // oled adress
  Wire.write(0x80); // command mode
  Wire.write(command);
  Wire.endTransmission();
}
```

urs8000, 2017-11-28:
thanks 

I‘ll try it soon

regards, urs

gbiohazard, 2017-12-03:
Same trouble here, I'm stucked with 10 mA with everything sleeping..
Any solution ?
Thanks

gbiohazard, 2017-12-03:
I removed the oled display and the board is still draining 10mA.

mrsnek, 2017-12-29:
Is there any news about this?

baxtery, 2018-01-14:
i have similar measurements, with LoRa and DHT-> 94 mA,  without Lora active 46 mA and in deep sleep 11  mA, now i haven tried to power via 3.3 v yet.

kovir, 2018-02-15:
@baxtery Have You got DHT sensor (temp and humid) working with OLED?

agirolami87, 2018-03-08:
Same issue, in deep sleep mode 11mA.
Any news ?
Thanks

DeuxVis, 2018-03-08:
This maybe can help reduce the consumption : https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series/issues/17#issuecomment-368258052

"If you power the device from the 3.3V pin the battery charger won't be enabled"

agirolami87, 2018-03-08:
@DeuxVis I already tried in my tests, but without improvements.

adriancuzman, 2018-03-14:
Also, where can we review the hardware design/schematic?

gbiohazard, 2018-03-14:
Hi, 

It is hardware design issue.
TM.

ToninoTarsi, 2018-03-16:
I'm thinking to use this board for my project but 10mA in deep sleep will be a very big problem. 
Did someone solve this problem  ?

adriancuzman, 2018-03-17:
@gbiohazard Thank you

ayhoung, 2018-03-19:
@gbiohazard any clue where the specific issue is?

gbiohazard, 2018-03-19:
Here are the tests I made (initially on https://www.thethingsnetwork.org/forum/t/big-esp32-sx127x-topic-part-1/10247)

**Using Heltec LoRa board:**
Deep sleep 10.2mA
Deep sleep OLED Off 10.2mA
Deep sleep OLED unsoldered 10mA
Deep sleep OLED and BMS (battery managing system (mcp73831 ?)) and LED remvoed : 2.6mA
Deel sleep without : OLED, BMS, LED, CP210x : 2.6mA

**Using Esp32 Core Board (LED removed):**
Deep sleep 4.5µA

**Using ESP32 Core Board (LED removed) and NiceRF LoRa1276 (sx1276 based module):**
Deep sleep 1.76mA

**Analysis:**
The sx127x can reach 1µA in deep sleep, so the total would be 5.5µA. BUT, the crystal used
is a TCXO, which remains powered continuously.
As I explained in thethingsnetwork topic, you'll have to power off the crystal (e.g. using µC pin and Mosfet) to achieve the expected low power consumption in deep sleep mode. If you still want to deep sleep µC while the LoRa is in listing mode (µC wake up by interrupt), the µC pin must be able to hold its state in deep sleep (the esp32 can on specific RTC pins).

See:

> esp_err_t rtc_gpio_hold_en(gpio_num_t gpio_num)
> Enabling HOLD function will cause the pad to latch current values of input enable, output enable, output value, function, drive strength values. This function is useful when going into light or deep sleep mode to prevent the pin configuration from changing.

Sorry for my english (FR).

baxtery, 2018-03-19:
i think we should design a PCB with  a step up, a mosfet, esp32 and an SX1276 and wait for someone in china to copy it and sell it cheap on aliexpress :)

baxtery, 2018-03-19:
but if we power off the sx, does OTAA still work after waking up from deepsleep?

mpatafio, 2018-03-26:
@kovir I did it! use the PIN 22 ;)

goran-mahovlic, 2018-03-28:
I did try to wirite 0x00 to REG_TCXO - and get same consumption.

Do you have some working schematic how should  TXCO powering look?

goran-mahovlic, 2018-03-29:
But then you can check this module it claims StandBy current is 1uA and it does not have pin to enable disable TXCO so it should be somehow possible to disable it with registers...
[](http://www.dorji.com/products-detail.php?ProId=58)

goran-mahovlic, 2018-03-31:
Without removing any hardware parts I am down to 4mA.

schabauerj, 2018-06-19:
Hi Goran,
Thank you so much for taking the time to describe this issue.  I tested with 3 different boards and thought I was crazy.

I have the same issue: If I plug my board into a USB-plug (and the Linux recognizes the new device), the power consumption is around 3mA.  If I only power form Lipo (or a USB Power Bank), it is 11.7mA (with some sensors).  If I power with Computer first and then hotswap to Battery, 3mA remain.  
With 3mA my Battery application could last almost half a year (4x18650 with 3500mA), but 11,7mA is really a pain.  
I thought I would change the startup code and began with disabling the Serial (Serial.end()) port before esp_deep_sleep_start(), but that would have been too nice :-)  Then I digged depper in "/home/nemo/arduino-1.8.5/hardware/espressif/esp32/cores/esp32", but couldn't find the startup code for the serial port.  Do you have any idea?  Or is it hidden in Rom?

goran-mahovlic, 2018-06-20:
There is only one thing I have read after my post in TTN forum, but did not have time to try:
https://www.thethingsnetwork.org/forum/t/big-esp32-sx127x-topic-part-2/11973/568

When you put the ESP32 into deep sleep the GPIO pins go into high impedance mode, which is not good for devices on the SPI bus, such as LoRa.
With the SPI bus high Z the LoRa device (also in sleep mode) takes the sleep current up to 120uA, not good.
If you add pullup resistors on the 3 SPI pins and the LoRa device select pin, sleep current of an ESP32 and LoRa device is 7uA.
Also note that although you should be able to leave the LoRa NRESET pin floating (it has its own internal pullup) if your driving it with the ESP32 then add an external pullup resistor, the ESP32 in deep sleep drags this pin low.

gbiohazard, 2018-06-20:
One important thing to understand is that NSS (SPI Cable Select) is active low. In deep sleep mode, the esp32 releases all outputs state to low, so the LoRa chip becomes active again. The trick is to use the rtc_gpio_hold_en(gpio_num_t gpio_num) function to keep HIGH the NSS pin.

But wait, in deep sleep mode, only some specific pins are available to control: RTC pins.
- GPIO18, used for NSS on the Heltec LoRa 32, isn't an RTC pins. 
- In deep sleep mode, GPIO18 will always go low
- In deep sleep, LoRa chip will always be in standby mode (about 1.5 mA).

With my LoRa1276 (NiceRF) module, it stays <1µA when the esp32 is in deep sleep (GPIO04 NSS)

schabauerj, 2018-06-20:
Chip Select and Active Low is a good point :-)
I will try this out and might be able to safe 1-2mA.  
However, I don't see this connected to the "Plug me into a proper USB host first if you want low power Problem."  Only after the USB-Serial converter is recognized by a host, the deep sleep power drops from around 10 to 3mA.  However, if it restarts on Battery, it will be 10mA again.  
This might be a bad design issue.  I read that the firebeetle-esp32 has a good low power of less than 1mA. Which boards can you recommend?

bradmcl, 2018-07-08:
So, if the schematics at https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series/tree/master/SchematicDiagram are correct, and I'm reading the info on the CP2102 correctly at https://www.silabs.com/documents/public/data-sheets/CP2102-9.pdf and https://www.silabs.com/community/interface/knowledge-base.entry.html/2004/04/20/cp210x_current_draw-OUVM  then it appears that if you start with a USB host, the CP2102 goes active, and then goes to Suspend after the USB is disconnected.  If you just power up on battery, then it stays active looking for a USB connection.  Apparently if there were a way to force the reset pin low you would get the Suspend level of power usage, but this board feeds the 5V bus from VBAT as part of the LDO circuit.  If there was another diode in the LDO circuit between 5V and pin 1 of the U1 LDO ( and possibly a weak pulldown on pin 9 !RST of the CP2102 it would likely eliminate the problem;  alternatively, if there was a way to use software to pull pin 9 !RST of the CP2102 low - by wiring it to an IO pin ( 3.3v vs 5v issue to solve ) then you could also force the state.  So as others have said elsewhere, it is a hardware design issue on this board.

lois-lee, 2018-08-15:
@urs8000 Mine is also not waking up after the esp_deep_sleep_start() call, did you ever fix the problem?

urs8000, 2018-08-17:
@lois-lee    my Heltec-WiFi board is working with the original code above from @rmh78 
Tested with the new Arduino Setup (WIN10/IDE 1.8.5) and board=Heltec_WIFI_Kit_32
I do not longer work with that boards. (also not with lora version)

myszko, 2018-09-01:
According with my tests the issue is related to the USB mode as mentioned previously by @bradmcl. Even with ESP32 on sleep mode and OLED disabled the node still drains 8-10mA. If I connect the USB to a PC and the PC recognizes it and after that disconect the cable, the current reduces to 1,3mA. This happens because the CP2102 goes to SUSPPEND MODE. 
For my case 1,3 mA is fine. 

However the question is if there is any way (SW or HW) to force the CP2102 to susppend mode.

schabauerj, 2018-09-05:
@myszko: How did you measure 1,3mA? I use the UM25 USB Meter and it shows 3mA.  Is there a new version of the board?  Concerning the CP2102 I have no idea how to force it to sleep mode, other than plugging it into a usb-connector and unplugging it.  
@urs8000: My esp32 never had a problem waking up from deepsleep UNTIL I deployed them in a bigger home automation project (12 sensors - not toooo big) and within 2 months two sensors did not wake up one time after 5 minutes of deep sleep.  After hitting the reset button, everything worked fine, but the uneasy feeling remains.  Any ideas?

urs8000, 2018-09-05:
@schabauerj    sorry I've no long term experience with ESP32. How did you wake it up?
Probably using the ULP as a watchdo could help if it does not wake up on your method...

divya0056, 2019-01-04:
@rmh78, 
Used your code and added Lora init , before the serial begin. Removed the power led. Noticed a strange behavior. If the serial speed is low there is more power used.
So with serial. begin(1000000), and LoRa. sleep there got power down to 550 -600uah or  around  0.5mah. it's prob cp2102 using most of the power.

jonashoechst, 2019-01-21:
@divya0056 @myszko 
I got a power consumption of ~3,1 mAh (with suspended CP2102) which is still much higher than your 1,3 / 0,5 mAh. Can you describe all the things you did?

kampto, 2019-01-30:
I have V2 Lora(sx1276) board with nothing connected to it for test purposes, Not calling the Wifi, RF or OLED librarys.  Deep sleep current is 3ma as measured inline of LiPo battery.  Thats as low as I can get.  
I have Wemos ESP32 dev board and in deep sleep it draws <0.1ma.

divya0056, 2019-01-30:
Disconnect the lipo battery power for vcc 3v.

kampto, 2019-01-30:
> Disconnect the lipo battery power for vcc 3v.
> […](#)

Sure, If I avoid everything and just power the chip itself the draw will be low, But this heltec board using LiPo regulator is high at 3ma compared to others. Difficult for long term battery operation

Heltec-Aaron-Lee, 2019-01-30:
@divya0056 @kampto when in deep sleep mode, some GPIO need set input mode. use this example:
[https://github.com/HelTecAutomation/Heltec_ESP32/blob/master/examples/Low_Power/Low_Power.ino](https://github.com/HelTecAutomation/Heltec_ESP32/blob/master/examples/Low_Power/Low_Power.ino)

kampto, 2019-01-31:
> @divya0056 @kampto when in deep sleep mode, some GPIO need set input mode. use this example:
> https://github.com/HelTecAutomation/Heltec_ESP32/blob/master/examples/Low_Power/Low_Power.ino

I tried that sketch exactly and with the correct Heltec_ESP32 package loaded.  The "LoRa.sleep();" and "LoRa.end();"  are not declared so it wont compile.  
Anyway i've been using the Radiohead RF95 library and called a "rf95.sleep();" and set all those GPIO you listed to input. So same thing? But same result, about 3ma in sleep.  Has anyone successfully gotten <1ma in sleep on this board without modifying the HW?

SloMusti, 2019-02-07:
I have tested the above mentioned low power example with Helltec V2 board and managed to reach 1.44mA@3.7V during sleep periods. Any thoughs on further minimizing this to uA range?

leocastro536, 2019-04-04:
yo encontre que poniendo el pin 11 cmd del esp32 se podia llegar a (80-90)uAh, conecte el pin 11 del esp32 a 3.3v y funciono.

frankheisig71, 2019-06-03:
Hi guys, maybe a little late, but I had the same problem with TTGO LoRa32 deep sleep power consumption and I managed it by cutting the LDO from 5V circuit and by soldering the LiPo directly to LDO. 
Fortunately the LDO is wired to 5V-circuit by a more or less good reachable single track coming from right top (on my picture) to the right top corner of pin 1 of LDO. 
After this change CP2102 is powered only by 5V USB and working if USB is plugged. The on board LiPo-charger of course is not working. No problem for me, because I decided to use a discrete charging module with temperature tracking and batterie protection. But you may wire pin 3 of charger and pin 1 of LDO by a shottky diode - should work.
Deep sleep power consumption now is 900uA.

[image]

(sorry for the bad picture quality...)

Skipper-is, 2019-06-11:
Ah that is awesome! I'm currently stuck on 10.4mA during deep sleep. May have to give detaching the lipo charger and cutting the LDO off from the CP2102 to see how much improvement that gives me

Heltec-Aaron-Lee, 2019-06-12:
@Skipper-is Only WiFi LoRa 32 V2 can down to 900uA in deep sleep. The V1 boards because of no low power design.

Anyway, you can choose from our [LoRa Node Family](https://docs.heltec.cn/#/en/products/lora/lora_node/heltec_lora_node_list_eu), all of the new nodes products were well low power designed.

frankheisig71, 2019-06-12:
@Heltec-Aaron-Lee: The described cut-track-method manages to have 900uA on a V1 board. Of course, one may use a V2 board, but ... V1: 1pc 9$ at aliexpress (2pc. w/o OLED - 18$) V2: hard to get for less than 18$ (1pc.)
And - it may help to find a usage for the 28 lying useless in ones drawer..  
:-)

Heltec-Aaron-Lee, 2019-06-13:
@frankheisig71 Oh, I wonder you still can buy a V1 board? We stopped producing it about 1 year ago, Is it possible it's just a bad copy?
[http://community.heltec.cn/t/wifi-kit-8-bad-copy/92](http://community.heltec.cn/t/wifi-kit-8-bad-copy/92)

Skipper-is, 2019-06-13:
I'm using the TTGO LoRa32, or possibly a clone. I think it might be a clone, as it has truly abysmal RSSI even at close range...

Heltec-Aaron-Lee, 2019-06-13:
@Skipper-is 
[image]
 Here is a document about how to test the maximum communication distance: 
[https://docs.heltec.cn/#/en/user_manual/lora_maximum_communication_distance_test](https://docs.heltec.cn/#/en/user_manual/lora_maximum_communication_distance_test)

We spent a lot of time on impendence matching. it's very important for an RF board. You can try our [Wireless Stick Lite](https://heltec.org/project/wireless-stick-lite/), it's the cheapest one currently. Maybe you buy from a reseller can have a cheaper price.

Anyway, another new LoRa board is in design and production, it's based on an all-new platform, with good low power design (5uA) and good RF design. and the price may lower than 10 USD. Currently, it still in debugging and waiting for some certificates. We will release it on August 1st.

Skipper-is, 2019-06-13:
Yea, I might wait for that one. The 4 I've got have an average of RSSI -75, sitting about 1 meter apart on my desk....
Any battery management with the new module? My intention is to link up a solar cell and a 18650 LiPo for power

Heltec-Aaron-Lee, 2019-06-13:
Yes, of course.

Skipper-is, 2019-06-17:
Safety cutoff for the batteries?

Heltec-Aaron-Lee, 2019-06-18:
@Skipper-is you mean when battery is full, the circuit will stop charging? YES

Skipper-is, 2019-06-18:
Or when it is discharged, that it'll shut off before damaging the battery

kampto, 2019-06-18:
It can not. The EN pin on the regulator is not broken out.  Very problematic for remote battery powered sensors. 

Regarding power consumption on the V2 board during deep sleep I could not get it below 10ma after a reset.  

For these reasons and others I switched to another brand Lora board for remote battery/solar sensors. Still running this board as a house powered receiver of other Lora brand boards.

sle118, 2019-07-29:
@frankheisig71 
> Hi guys, maybe a little late, but I had the same problem with TTGO LoRa32 (V1) deep sleep power consumption and I managed it by cutting the LDO from 5V circuit and by soldering the LiPo directly to LDO.

I tried replicating your approach and couldn't get down to your numbers; the best I can achieve is only when plugging/unplugging USB so the serial chip goes in standby. I did cut the USB +5V trace that goes from the USB right next to the LDO's input.  Then I soldered the battery to the 3.3V breakout pin on the board (e.g. bypassing the LDO).

Did you solder the lipo to the LDO's input or to its output pin (bypassing it entirely)?  Also, are you doing anything special to get to sleep?

dahutte, 2019-08-03:
Confirmation on Heltec Aaron Lee comment:   On 3 recently buyed modules ESP32 Heltec Lora V2, I measure 0.9 mA in deepsleep mode, but I dont understand why it is not less. On the shematic, I find the flash chip which can draw such current?  It would be a far progress (mandatory for me) to get lower than ~150uA at deepsleep and the CPU allows this.
On an other side, RF power is limited to 17dbm on my 3 modules. I will be very interested in improvement of rf matching. I wait any info on this on future versions.
Thanks anymore for the huge info I read from this forum.

Heltec-Aaron-Lee, 2019-08-04:
Because the WiFi LoRa 32 uses an external FLASH, the running codes are inside, it's can't sleep during the ESP32 sleeping time.

Heltec-Aaron-Lee, 2019-08-04:
@dahutte Anyway, you can choose from our LoRa Node family, there are some other nodes with better low power design.

[63 more replies]