Heltec-Aaron-Lee/WiFi_Kit_series #9: BAT LED always blinking?

contractorwolf, 2018-01-01:
Hey just started using this board for the first time and noticed that the BAT LED always seems to be blinking when I have the board plugged into USB.  I decided to remove all code (except the Serial Output indicating that the code had started) and the LED is still blinking.  I have no code in the loop and just the initialization for serial in the setup.  I am sure that this is probably to indicated that there is no battery connected but I dont know how to turn it off.  Thanks, and so far this looks like a great board I am just irritated by the constant flashing.

Heltec-Aaron-Lee, 2018-01-02:
Very sorry, we can't control the yellow LED, the LED blink means power on, bright means in charging, it will turn off after the battery filled.

Do you think that's a big problem? Need we fix it in next version?

contractorwolf, 2018-01-02:
Yes, that design is terribly distracting. You need to make that LED be always on for "power on" and maybe flashing for charging. What other electronic device has a flashing LED for power on? It should also be controlled through the code, even if it defaults to on. That is so annoying that I am going to have to unsolder the LED to fix it. Thanks for getting back to me quickly!

MikeyMoMo, 2018-07-25:
Even if you put this in a case, the blinking orange LED slows through the clear plastic bezel over the OLED.  This is totally wrong.  It must be controllable.  Even having it steady on would still put an orange glow around the left end of the plastic OLED bezel and look very bad.  It would make any project look horrible.  Can't imagine anyone thinks this is the right way to design it.

enricogallesio, 2018-09-03:
I agree, this should be fixed somehow to allow disabling the LED without desoldering that

tomasMiddleton, 2018-10-03:
Please allow us to change that setting, it is very annoying/distracting to me as well.

walterheisenberg, 2019-01-15:
I got my board today and powered it on via USB.
After that, I didn't notice the LED was blinking.
After flashing Micropython, the LED is flashing after power on.
Please fix that!
Thx

tomasMiddleton, 2019-01-16:
It should be remembered as well that some people would want the best battery performace they can get, a LED blinking like that can’t be helping.

JZ-SmartThings, 2019-06-29:
This issue should not have been closed... not like it was fixed. Still have the crazy LED that I either need to tape over or de-solder. Please consider a more reasonable alternative to a blinking light when I don't use or care about batteries or charging state.

Heltec-Aaron-Lee, 2019-06-29:
Sorry, I will reopen it.

That's a problem caused by the old battery charge chip MCP73831. We had replaced it with TP405 on the newer boards. the charging status LED keeps a steady light when the battery not connected.

JZ-SmartThings, 2019-06-29:
Thanks a lot @Heltec-Aaron-Lee! I was wondering if you can highlight a difference to look for to make sure that I'm not buying the older board. My guess is the new board revision looks extremely similar to the old and I'd love to know how to tell the difference so that I don't buy the constantly blinking older model.

contractorwolf, 2019-06-29:
so we get free new boards for everyone in this convo right? ;)

Heltec-Aaron-Lee, 2019-07-04:
@contractorwolf Haha, that's a good idea. But I don't have the power to do that 😄.

Anyway, I applied some 20% off coupons for you guys. 

[http://community.heltec.cn/t/20-off-coupon-for-esp32-and-esp8266-products/113](http://community.heltec.cn/t/20-off-coupon-for-esp32-and-esp8266-products/113)

Heltec-Aaron-Lee, 2019-07-04:
@JZ-SmartThings All LoRa products had fixed the blinking LED problems. WiFi Kit 32 and WiFi Kit 8 will be fixed this month.

JZ-SmartThings, 2019-07-04:
@Heltec-Aaron-Lee got it, thank you!

So for now, the current model is not fixed but its planned to be very soon. You should have saved the coupon for us when the fix is here :)

Heltec-Aaron-Lee, 2019-07-05:
@JZ-SmartThings 

Thanks for your suggestion, I will do that.

FlailAway, 2019-07-28:
Where can I buy them to use the Coupon? I am OK with un-soldering the LED for 20% off, I never use the battery socket anyway :)

Heltec-Aaron-Lee, 2019-07-29:
@FlailAway On our home page:
[https://heltec.org/proudct_center/](https://heltec.org/proudct_center/)

Armengedon, 2020-06-01:
I have the lora version and with the factory program the led was not blinking or i cannot remember it doing this. But after installing my own software using the ESP-IDF firmware the led blinks like crazy. Is there some way to control it via firmware that i am not aware? Thank you very much @Heltec-Aaron-Lee