Heltec-Aaron-Lee/WiFi_Kit_series #10: Battery to ADC connection

garthberry, 2018-01-07:
Is there any kind of connection between the battery and one of the ADC pins for monitoring battery level?

seaser0, 2018-02-06:
I have the same question. @garthberry: Do you figured it out?

seaser0, 2018-02-08:
I was running some test and it seams to be A4. When I connect a full LiPo the analog read function shows values around 500 and it will drop below 100 when the battery is empty. I hope that helps.

garthberry, 2018-02-12:
@seaser0 I performed the same test with a bench power supply connected to the battery terminal and I can confirm that out of A1-A7, A4 is the only one that reflects the change in battery voltage. The minimum voltage seems to be about 3.3v, anything less and the analog read function returns 0.

At a guess I would say there is a voltage divider connected between the battery + and A4 pin. @Heltec-Aaron-Lee Any chance the schematics could be made available somewhere?

kovir, 2018-02-27:
sorry for a dumb question.... A4 - which PIN number is this? ( I can't see A4 on pinout diagram)
thanks
Marcel

seaser0, 2018-02-27:
According to this diagram A4 (or ADC1_4) corresponds to GIPO 32.

Heltec Wifi Kit 32 Pinout Diagram:
http://esp32.net/images/Heltec/WIFI-Kit-32/Heltec_WIFI-Kit-32_DiagramPinout.jpg

walterheisenberg, 2019-02-08:
Has anyone code for micropython to read the battery level?
vcc = machine.ADC(machine.Pin(32))
vcc.read()
does not work reliable and gave me values from 500 to under 100 in one minute

mwargan, 2019-07-23:
@walterheisenberg have you figured it out? My readings are all over the place too :/

christiangda, 2020-07-25:
Heltec Schematic:

* [https://heltec-automation-docs.readthedocs.io/en/latest/esp32+arduino/frequently_asked_questions.html#vext-control](https://heltec-automation-docs.readthedocs.io/en/latest/esp32+arduino/frequently_asked_questions.html#vext-control)

Good discussion about that

* [https://hackaday.io/project/27791-esp32-lora-oled-module/log/69600-battery-voltage-measurement](https://hackaday.io/project/27791-esp32-lora-oled-module/log/69600-battery-voltage-measurement)