Heltec-Aaron-Lee/WiFi_Kit_series #18: Make sample code like Serial.Print but send to on board display

Eddiiie, 2018-01-28:
Sorry if this is not the right place to request this.

I am trying to write a function like Serial.Print  that will output to the display.  It is harder than originally thought but maybe making it too difficult.   Scroll the text?   Copy existing display to memory, clear screen, write text, delay, restore display from memory.

Can someone throw something together for me?

I think this would be a handy example.

jared0x90, 2018-01-31:
Hey Eddiiie,

The onboard i2c screen is pretty easy to control actually. Here are some snippets from a project where I'm using it for some very basic data. I also use the Streaming library to make string formatting a bit more user/developer friendly and would recommend checking it out. The library I reference here has scrolling capability built in. I'm not using it here but I have included links to the libs in the comments where they discuss how to do this.... The streaming lib also works with Serial and anything else that has a Print function.

```
#include "Streaming.h"            // Written against Streaming 5        http://arduiniana.org/Streaming/Streaming5.zip
#include "SSD1306Ascii.h"         // Written against ASCIIWire 91a4ef4  https://github.com/jared0x90/SSD1306Ascii/releases
#include "SSD1306AsciiWire.h"

/******************************************************************************
  Settings for OLED display
******************************************************************************/
#define OLED_I2C_SDA              4
#define OLED_I2C_SCL              15
#define OLED_PIN_RESET            16
#define OLED_I2C_ADDRESS          0x3C

SSD1306AsciiWire oled;

void setup() {

  // init OLED
  // set GPIO16 low to reset OLED
  pinMode(OLED_PIN_RESET, OUTPUT);
  digitalWrite(OLED_PIN_RESET, LOW);
  delay(50);  // Wait for OLED to power cycle
  digitalWrite(OLED_PIN_RESET, HIGH); // while OLED is running, must set GPIO16 to high
  Wire.begin(OLED_I2C_SDA, OLED_I2C_SCL);
  oled.begin(&Adafruit128x64, OLED_I2C_ADDRESS);
  oled.setFont(System5x7);
  oled.clear();

  oled << F(
    "Sample Messag\n"
    "Hello World\n"
  );
}
```

jared0x90, 2018-01-31:
You may need to adjust the pin definitions a bit if you aren't using the LoRa board as I have only used that version.