Heltec-Aaron-Lee/WiFi_Kit_series #19: MicroSD Example Code that works with WiFi LoRa

jared0x90, 2018-01-31:
Hello,

I have been scratching my head trying to get the Heltec LoRa WiFi ESP32 board to talk to a MicroSD card properly using the Arduino IDE. I am using a breakout board I have used with a 32u4 & 328p and verified those can write to the SD card. I've also tried a few other breakout boards and cards to rule those out as well as a few different heltec lora esp32 boards (I own about a half dozen of them now). I have tried both your example code (which appears to use the same chip select pin as the onboard LoRa SPI), have tried passing a different CS in SD.begin, have tried using the "mySD" library for the ESP32 where you can specify all lines (MOSI, MISO, SCK and CS) in SD.begin.... all to no avail. When I've tried scoping the MOSI, MISO, SCK & CS line it seems like the ESP32 is attempting to communicate with the MicroSD MUCH faster than it does with the onboard LoRa chip. Do you have a working example for using MicroSD with this LoRa board? I noticed that you use a few unconventional pins compared to the standard ESP32 pinout on the LoRa board. I am curious if that might be causing any potential issues. For instance to get another hardware UART working I had to modify...

Documents\Arduino\hardware\espressif\esp32\cores\esp32\HardwareSerial.cpp

    if(_uart_nr == 1 && rxPin < 0 && txPin < 0) {
        rxPin = 13;
        txPin = 21;
    }

...in order to get a functional second hardware serial line I needed. I really want to use this board as the centerpiece of a project I am working on but this is unfortunately quite the hangup at the moment.

jared0x90, 2018-01-31:
Here is a very basic implementation via the mySD library. 

```
#include <mySD.h>

#define MICROSD_PIN_CHIP_SELECT   4
#define MICROSD_PIN_MOSI          13
#define MICROSD_PIN_MISO          19
#define MICROSD_PIN_SCK           5

File root;

void setup()
{
  Serial.begin(115200);

  Serial.print("Initializing SD card...");
  /* initialize SD library with SPI pins */
  if (!SD.begin(MICROSD_PIN_CHIP_SELECT, MICROSD_PIN_MOSI, MICROSD_PIN_MISO, MICROSD_PIN_SCK)) {
    Serial.println("initialization failed!");
    return;
  }
  Serial.println("initialization done.");
  /* Begin at the root "/" */
  root = SD.open("/");
  if (root) {
    printDirectory(root, 0);
    root.close();
  } else {
    Serial.println("error opening test.txt");
  }
  /* open "test.txt" for writing */
  root = SD.open("test.txt", FILE_WRITE);
  /* if open succesfully -> root != NULL
    then write string "Hello world!" to it
  */
  if (root) {
    root.println("Hello world!");
    root.flush();
   /* close the file */
    root.close();
  } else {
    /* if the file open error, print an error */
    Serial.println("error opening test.txt");
  }
  delay(1000);
  /* after writing then reopen the file and read it */
  root = SD.open("test.txt");
  if (root) {
    /* read from the file until there's nothing else in it */
    while (root.available()) {
      /* read the file and print to Terminal */
      Serial.write(root.read());
    }
    root.close();
  } else {
    Serial.println("error opening test.txt");
  }

  Serial.println("done!");
}

void loop()
{
}

void printDirectory(File dir, int numTabs) {

  while(true) {
     File entry =  dir.openNextFile();
     if (! entry) {
       break;
     }
     for (uint8_t i=0; i<numTabs; i++) {
       Serial.print('\t');   // we'll have a nice indentation
     }
     // Print the name
     Serial.print(entry.name());
     /* Recurse for directories, otherwise print the file size */
     if (entry.isDirectory()) {
       Serial.println("/");
       printDirectory(entry, numTabs+1);
     } else {
       /* files have sizes, directories do not */
       Serial.print("\t\t");
       Serial.println(entry.size());
     }
     entry.close();
   }
}
```

Ldsrc2008, 2018-02-13:
I have a heltec lira oled board,I tried to add a micro sd card with the spi interface,and 21 as ss for sd, not yet liading lira code,tried to pull up the cc is lira,but the sd init always fail, any suggestion to make sd card work together with lira on spi of heltec board?

Grippentech, 2018-03-30:
Any solution to this? I'm having the exact issue right now... enabling SD kills the OLED

jared0x90, 2018-04-04:
@Grippentech 

I haven't personally messed around with this any more but I think part of the issue I was having was not disabling both SPI devices via CS pins before begin() them. I wound up having further issues with the heltec boards that required me to handle some timing stuff in software that seemed unnecessary from other ESP32 examples/projects I have seen. I decided to try the adafruit esp32 feather stack of components and literally plugged them together and the timing issues seemingly fixed themselves. I would check the pinout closely for this board. You might be using a pin reserved for the OLED disable. The SPI pinout is also a bit non standard compared to other ESP32 dev boards from my recollection.

Grippentech, 2018-04-04:
Makes sense yeah, thanks for the reply. I'm trying to connect 3 MPU4250 Acceleromters and an SD card over SPI so that's proving interesting... :D so far looks like I might have to dive into some library code.

Grippentech, 2018-04-14:
Yeah it's a chip select pin issue.  The standard defined SS pin is what interferes. If you add ssPin = 34; or whatever inside the library SDFS::begin defined function in SD.cpp it fixes everything... 

Not elegant but works for the quick prototype I'm doing right now...

lytruns, 2018-05-10:
Hello ... would it be possible to provide the code? : D

jonashoechst, 2019-01-21:
I've got an example project working with LoRa and an SD card. My board does not have an OLED display, thus i cannot test wether this is working, too. However I'm not using any of the OLED connected ports.

Code: http://github.com/jonashoechst/ttgo-lora-sd

adrianomourapeixoto, 2019-08-29:
hello, i'm having trouble connecting an sdcard to esp32 lora from heltec. would anyone have a solution for this?

jonashoechst, 2019-08-29:
I actually worked on TTGO LoRa with SD Card as well and came up with a solution: https://github.com/jonashoechst/ttgo-lora-sd 

My code uses two separate SPI busses for SD and LoRa.

microclimatedev, 2026-04-05:
Has anyone come up with a solution to this problem? It's been 7 years....Im having the same problem. Im trying to add sd card to Heltec V4 Expansion Kit. Any help would be greatly appreciated.