Heltec-Aaron-Lee/WiFi_Kit_series #23: using i2c sensor with OLED

kovir, 2018-02-15:
Hi
I connected DHT31 temp & humidity sensor via i2c (using pin 21 & 22) but it won't work when OLED is initiated (display.init(); ) - I get NaN from DHT31. When I comment out display.init(); sensor works...
How to use i2c with enabled OLED>?

toddlesher, 2018-05-18:
Use the second I2C bus:

```
// Second I2C bus for sensors
TwoWire SensorWire = TwoWire(1);

// Bringup sensor I2C bus BEFORE OLED.
SensorWire.begin(PIN_SDA, PIN_SCL);
```
SDA is pin 21, SCL is pin 22.

DaveCurnow, 2018-08-22:
Not sure about the version you are using, but I eventually managed to get my WiFi Kit 8 working with a I2C IR Temperature sensor. The trick was to bring the display up first and then the sensor.
#include <Adafruit_GFX.h>
#include <Adafruit_SSD1306.h>
#include "Wire.h"
#include <Adafruit_MLX90614.h>
Adafruit_MLX90614 mlx = Adafruit_MLX90614();
#define OLED_RESET 16
Adafruit_SSD1306 display(OLED_RESET);
const int Pushbutton = 13;
const int LED = 12;
int value = 0;

void setup() {
  Serial.begin(9600);
  Serial.println("Adafruit MLX90614 test");
  pinMode (LED, OUTPUT);
  pinMode (Pushbutton, INPUT_PULLUP);

  display.begin(SSD1306_SWITCHCAPVCC, 0x3C);  // initialize with the I2C addr 0x3C (for the 128x32)
  display.display();
  mlx.begin();
  delay(1);
   // Clear the buffer.
  display.clearDisplay(); 
  display.ssd1306_command(SSD1306_DISPLAYOFF);
}

cdoern, 2018-12-02:
did anyone ever figure this out? trying to use an mpu6050 with the heltec esp32 wifi lora