Heltec-Aaron-Lee/WiFi_Kit_series #27: One module doesn't send/receive

onemoreitguy, 2018-03-11:
I bought your two great 2 modules kit.
I managed to setup them using your libraries and I the leds showing the send and receive packets sucessfully.
But when I switch the code between the modules one keeps sending, but the other doesn't report receive.

Is this expected or some sort of malfunction? what can I do?

Heltec-Aaron-Lee, 2018-03-21:
Did you connect a antenna?

onemoreitguy, 2018-03-21:
I did not connected at the first attempt. Now I'm using the antennas
But if there was a damage it would make unusable for both send and receive at the same time, no?
[image]

Heltec-Aaron-Lee, 2018-03-21:
indeed send and recieve use a different path.
https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series/tree/master/SchematicDiagram
here is the schematic diagram release. it may help you check into it.

onemoreitguy, 2018-03-21:
So there could have been a damage on components just for send or receive?
I only see an SPI connectivity between th SX1276 and the ESP32

Heltec-Aaron-Lee, 2018-03-21:
I mean the RF part, LoRa singal out put via RFO_LF/RFO_HF; and input path via RFI_LF/RFI_HF.