Heltec-Aaron-Lee/WiFi_Kit_series #30: Fails to receive or send

holgerlembke, 2018-03-19:
Hi!

I have two Heltec WiFi Lora Esp32 boards, #1 with this small wire wrapped antenna and #2 without this antenna.

I use the Lora example LoraSender/LoraReceiver from this git with band 433E6.

Problem:
#1 send and #2 receive works fine, #2 send and #1 receive does not work at all.

Any idea where the problem is?

Heltec-Aaron-Lee, 2018-03-21:
Antenna is must needed for both board, wothout antenna may broken your device.

onemoreitguy, 2018-03-21:
@holgerlembke, I'm experiencing a similar problem:
- I can use one to send and the other to receive
- But if I change roles they dont' work (no send or receive)
I have both antennas connected now, I didn't have at the first attempt. So i don't know if it damage the capacity to send or receive