Heltec-Aaron-Lee/WiFi_Kit_series #32: OnReceive CallbackCalled

darrylburke, 2018-03-22:
Board: Heltec HTIT-WB32LA
Espressif: ESP32 Arduino 0.0.1 (Latest as of Mar19th 2018)
Heltec: most recent as of Mar 19th 2018)
Arduino IDE: 1.8.5

I've tried several ways to configure the OnReceive Call back, but it never gets called.

 LoRa.onReceive(cbk);
 LoRa.receive();

even tried the suggestions to attach an interrupt to pin 26 DIO0
pinMode(26, INPUT);
  attachInterrupt(digitalPinToInterrupt(26), cbkCallback, CHANGE);

but the callback never gets called.. 
(If I try the polling mechanism there is data and it can be read, so I know it's receiving data from the sender)

ramazanulucay, 2018-03-25:
I have a same problem. How can I fix it.

kadirulucay, 2018-03-30:
Hi Darrylburke,

LORA.CPP__
1) The Spreadingfactor has to be at the same value as the sender.
2) _Spisettings 10E6 make sure it is.
3) Loraclass::SetSpreadingFactor(int sf)
    else part Optimize and threshold same value as the sender make sure it is.

Example Code :

#include <SPI.h>
#include <LoRa.h>

#define SCK     5    
#define MISO    19  
#define MOSI    27   
#define SS      18  
#define RST     14  
#define DI0     26   

#define BAND    433E6  
#define PABOOST true

void setup() {
  Serial.begin(115200);
  while (!Serial);

  Serial.println("LoRa Receiver Callback");
  SPI.begin(SCK,MISO,MOSI,SS);
  LoRa.setPins(SS,RST,DI0);
  if (!LoRa.begin(BAND,PABOOST)) {
    Serial.println("Starting LoRa failed!");
    while (1);
  }

  LoRa.onReceive(onReceive);

  LoRa.receive();
}

void loop() {

}

void onReceive(int packetSize)
{
  Serial.print("Received packet '");
  for (int i = 0; i < packetSize; i++)
  {
    Serial.print(String(LoRa.read(),HEX));
  }
  Serial.print("' with RSSI ");
  Serial.println(LoRa.packetRssi());
}

good luck 🥇