Heltec-Aaron-Lee/WiFi_Kit_series #36: BLE Library problem: c++ exception handling must be enabled

Pravin-master, 2018-03-30:
when I am compiling BLEnotify following error is occuring
In file included from C:\Users\Pravin\Documents\Arduino\hardware\espressif\esp32\libraries\BLE\src/BLEClient.h:18:0,

                 from C:\Users\Pravin\Documents\Arduino\hardware\espressif\esp32\libraries\BLE\src/BLEDevice.h:19,

                 from C:\Users\Pravin\Documents\Arduino\hardware\espressif\esp32\libraries\BLE\examples\BLE_notify\BLE_notify.ino:21:

C:\Users\Pravin\Documents\Arduino\hardware\espressif\esp32\libraries\BLE\src/BLEExceptions.h:13:2: error: #error "C++ exception handling must be enabled within make menuconfig. See Compiler Options > Enable C++ Exceptions."

 #error "C++ exception handling must be enabled within make menuconfig. See Compiler Options > Enable C++ Exceptions."

  ^

exit status 1
Error compiling for board WIFI_LoRa_32.

before that esp_bt.h is not available error prompted, I solved it by renaming all bt.h to esp_bt.h

kadirulucay, 2018-03-30:
You have mistakes in the library.

Pravin-master, 2018-03-30:
Can you help me out for this??

Other part of wifi_lora_32 is working fine. Even simpleBLE is also working fine.

kadirulucay, 2018-04-03:
So what's the problem