Heltec-Aaron-Lee/WiFi_Kit_series #37: JTAG debug capability ?

dnguyen76, 2018-04-03:
Hello

In Espressifn ESP32 doc (http://esp-idf.readthedocs.io/en/latest/api-guides/jtag-debugging/configure-other-jtag.html) , Jtag adapter must be connected to GPIO12, GPIO13, GPIO 14 and GPIO15 ESP32 pins.

But Heltec wifi LoRa board, GPIO14 is used for LoRa_RST and GPIO15 is used for OLED_SCL. In Heltec pinout diagram we can see these two pins must not be used for other purpose unless you know what you are doing.

Does it mean we cannot JTAG debug Heltec board or is there a workaround by assigning other pins to JTAG signals ?

Thanks in advance for your answer

kadirulucay, 2018-04-04:
Hello
Contact me k.ulucay@outlook.com

macedolfm, 2019-03-17:
Hi Guys, did you manage to debug Heltec using a JTAG ?

dnguyen76, 2019-03-18:
Sorry I don't go forward up to now with Heltec board