Heltec-Aaron-Lee/WiFi_Kit_series #38: UART1 and UART2 usage

ralfoide, 2018-04-08:
This isn't really an issue, more like a documentation snippet which I didn't find anywhere else.

I got 2 ESP32 Wifi Kit 32 boards. Fantastic little boards. One of the projects was to interface with an existing RS232 device (aka "serial port"). One of the strengths of the ESP32 is that it has 3 UART modules. So first I tried to see which ones were available on this board.

TL;DR: One must realize that on this particular board, UART1 and UART2 cannot be used and UART0 has limitations.

Source: ESP32 datasheet here
https://www.espressif.com/sites/default/files/documentation/esp32_datasheet_en.pdf

Compared with the Heltec Schematic Diagram for the Wifi module:
https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series/raw/master/SchematicDiagram/WIFI_Kit_32_Schematic_diagram.PDF

What we need to do is look at the ESP32 datasheet to see where U0/U1/U2 are listed and cross-reference that with the Wifi Kit 32 schematic to see how the pins are used.

ESP32 pin number, pin name, alternate function | Wifi Kit 32 Pin name and usage
---- | ---
UART0 | 
38, GPIO19: U0CTS | 19, available
39, GPIO22: U0RTS | 22, available (see note below)
40, U0RXD: GPIO3 | RX, to CP2102 TXD
41, U0TXD: GPIO1 | TX, to CP2102 RXD
UART1 | 
28, SD_DATA_2: GPIO9, U1RXD | FLASH_SHD
29, SD_DATA_3: GPIO10, U1TXD | FLASH_SWP
30, SD_CMD: GPIO11, U1RTS | FLASH_SCS
31, SD_CLK: GPIO6, U1CTS | FLASH_SCK
UART2 | 
25, GPIO16: U2RXD | OLED RST
27, GPIO17: U2TXD | 17, available
32, SD_DATA_0: GPIO7, U2RTS | FLASH_SDO
33, SD_DATA_1: GPIO8, U2CTS | FLASH_SDI

Note: U0RTS is listed in the ESP32 datasheet as ESP32 pin 22, GPIO22. In the Wifi Kit 32 Pinout graphic, it is also listed as GPIO22. However it is listed in the Wifi Kit 32 schematic as pin 21, GPIO15, shared with the OLED SCL.

So, voilà, bottom line:
* UART 1 and UART 2 are not usable on this board as the pins are used by the Flash and the OLED.
* UART 0 is “maybe” usable, as long as the CP2102 is not being used.

I hope this will help others.

nicoloboatto, 2018-04-30:
Hi @ralfoide,

First of all thanks for taking the time to test this out and write this quality post.

I'm currently doing some testing with the WiFi LoRa 32 board myself and can't figure out how to get readings from UART0, even when CP2102 is not in use.
In the code, did you reference the board as "Serial", since UART0 takes the place of USB Serial, or did you set it up as a separate "SoftwareSerial" with custom pins?

Dreanaught, 2018-10-04:
Quite old Issue but it is in fact possible to use the UART2. I did so using HardwareSerial.h
The PIN 16 is only the OLED RESET. It is used when starting things up and might be reclaimed as soon as the OLED has been reset.
I have not yet tried to use both OLED and UART2 in one Project, but they do not interfere each other when only one is used.

djsyl, 2018-10-26:
If use micropython you can change pin rx for sample from machine import UART

defaut object is
# object UART(2, baudrate=9600, bits=8, parity=0, stop=1, tx=17, rx=16, rts=-1, cts=-1, timeout=0, timeout_char=2) is of type UART

now you can make that

uart = UART(2, 9600)                         
uart.init(9600, bits=8, parity=0, stop=1,tx=17,rx=5)

bye bye oled reset :)

have a fun

semaf, 2019-02-02:
What about if you need to use UART2? How to pass OLED_RST to another PIN?

`  pinMode(16, OUTPUT);
  digitalWrite(16, LOW); // set GPIO16 low to reset OLED
  delay(50);
  digitalWrite(16, HIGH);

  SerialGPS.begin(9600, SERIAL_8N1, 16, 17);

  Serial.begin(115200);`

sthex, 2019-03-11:
I have a TTGO Lora with OLED. Guess hardware is much the same. GPIO16 is used for OLED RST and needs to be high when display is in use. I tried to just assign another free pin to USRT2 and it worked like a charm. Also together with an active OLED.

HardwareSerial Serial2(2);
#define RXD2 23   //16 is used for OLED_RST !
#define TXD2 17

void setup() {
  ..
  Serial2.begin(9600, SERIAL_8N1, RXD2, TXD2); 
}
void loop() {
 while (Serial2.available()) {
    Serial.print(char(Serial2.read()));
  }
…
}

Evilgeniusnerd, 2020-05-04:
@sthex you think you can explain more to me on how you got your TTGo Lora board to work with the RockBlock. I have a TTGO T Beam with Lora without  OLED display.

Evilgeniusnerd, 2020-05-04:
Also, has anyone gotten this to work with Heltec ESP32 with Lora Board? I have tried to assign different pins with little to no luck.

Waldmensch1, 2020-09-06:
In my GIT you find a TTN Mapper Project (platformio) which works on ttgo t-beam Rev 1.0 and 1.1 and Heltec Wireless Stick. The 3 environments can be selected in platformio.ini. The GPS on Heltec Wireless Stick uses RX on 23 and TX on 17 as recommended by @sthex
The Project uses the LMIC library for TTN connection

https://github.com/Waldmensch1/tbeam-ttnmapper

navivfr, 2023-11-30:
hello,

What about LORA32 with OLED 
There is a way to add an extra UART ? (in my case a SIM800)

[image]