Heltec-Aaron-Lee/WiFi_Kit_series #45: Is it possible to turn off battery power with a switch?

randallamoore, 2018-05-10:
With Adafruit boards, you can disable the battery charging circuit by tying their EN pin to ground. This allows you to add a slide switch that will turn off the device; the battery doesn't power the device, and does not drain either. With my first prototype using the esp8266, I tried the same thing, but the EN pin is actually Chip Enable. Tying it to ground still allows the battery to supply power to external sensors on the 3.3 regulated pin.

I had to route the negative lead of the battery through a slide-switch before connecting it to the battery socket on the board. Is there a better way to do this?

martinberlin, 2018-11-12:
I would also like to know if there is best way. @Heltec-Aaron-Lee can you please inform us ?
Thanks in advance!

Heltec-Aaron-Lee, 2018-11-20:
@randallamoore you may need series connection an SPST on the battery line.
[image]
[image]