Heltec-Aaron-Lee/WiFi_Kit_series #47: Source Code for "Hello World" OLED demo

a1fred0, 2018-05-14:
Would you be able to point me towards the source code for the "Hello World" application that is installed on WIFI_Kit_32 out of the box?  The one with scan of Wifi that ends "Even scrolling is working"

jhoughjr, 2018-06-30:
I agree. Working code would be nice to have. That preloaded code is the only time I've seen this display working.
I'm avoiding this board in the future but for now I'm stuck with it.

thalaser, 2018-07-02:
Hi,

I got the source code few month ago once I was looking to control under
U8g8 libraries, I just can´t recall where. I will look for it and will post
the link.

Regards,
E.

stickbreaker, 2018-07-15:
@a1fred0 I have the display working just fine with a modified Adafruit_ssd1306.  Here is my modified library [ssd1306](https://github.com/stickbreaker/SSD1306). It requires AdaFruit_GFX, and the current arduino-esp32 core.

Chuck.