Heltec-Aaron-Lee/WiFi_Kit_series #50: How to retrieve battery charge level

ratcashdev, 2018-05-17:
Sicne this board has a li-pol charger, i was wondering, if it's possible to retrieve the charge status (charging/not charging) and also, how to retrieve the current battery percentage/level using Arduino?

antoniosap, 2018-06-29:
you can place a voltage partitor on BAT + and route to pin A0 ?

ratcashdev, 2018-07-03:
Yes, that would be possible. I was just wondering, whether there's something out of the box, since the charging regulator must measure the voltage.

dalefrancis88, 2018-08-29:
Was this ever solved? the last answer doesn't seem to be a real answer

kampto, 2018-11-05:
You can tie in a wire to the battery positive leg and run it thru a voltage divider to an analog input. That's what I have been doing. Some other mfg (Adafruit, Wemos) put an internal V-divider on the board for reading it in the sketch. The esp32 Lora kit Version2 board mentions some battery monitoring but the documentation is poor and theres no schematic for V2 so cant figure out if its real or not.

3KUdelta, 2019-03-03:
Maybe this might help: https://github.com/3KUdelta/heltec_wifi_kit_32_batt_monitor

gavan1, 2021-12-03:
@3KUdelta thank for this link! from your code 

```
 if (val>=320){
    val=320; // 3.7v 500 mAh Batt - estimated max at 320, cut at 320
    }    
                       
  val = (val-100)/2.2;  // calculate percentage  
```

Where does the 2.2 come from?
Basically how should this be adjusted for a 2000 or 2600 mah?

3KUdelta, 2021-12-03:
It seems that I normalized it to 220 (320 when full - 100 = 220) /2.2 equals precentage.
Don't ask me why I did it. It is a long time ago ;-)