Heltec-Aaron-Lee/WiFi_Kit_series #59: display.drawXbm compile error OLED_LoRa_Receiver / OLED_LoRa_Sender

JBSchueler, 2018-09-09:
When I compile the examples `OLED_LoRa_Receiver` / `OLED_LoRa_Sender` I get an error at line 45 / 48.
I am using Arduino 1.8.5 by the way.

```
OLED_LoRa_Receiver:45: error: invalid conversion from 'char*' to 'const uint8_t* {aka const unsigned char*}' [-fpermissive]

   display.drawXbm(0,5,logo_width,logo_height,logo_bits);

                                                       ^
```

Both files have the same line of code at that point
```c
  display.drawXbm(0,5,logo_width,logo_height,logo_bits);
```

If I cast cast logo_bits to uint8_t* it does compile...
```c
  display.drawXbm(0,5,logo_width,logo_height,(uint8_t*)logo_bits);
```
Is this a bug in the code or am I doing something wrong???

JBSchueler, 2018-10-04:
Thank you @solotaker to fix this in commit f971dc8c06b24b261bda4c56536126032cd5b4f0

JBSchueler, 2018-11-15:
Ticket can be closed

closed 2018-11-15 by JBSchueler

Cerunnos75, 2021-07-22:
I use the heltec 0.0.5 board driver and the heltec 1.1.0 libary, but get exactly the same error.
What am I doing wrong?

JBSchueler, 2021-07-22:
check if the `logo_bits` is being cast to `(uint8_t*)`

Cerunnos75, 2021-07-22:
Thanks a lot. That was the missing point .....