Heltec-Aaron-Lee/WiFi_Kit_series #61: Is Vext/pin21/OLED MOSFET supported or present?

alexwhittemore, 2018-09-11:
The code examples for the WiFi Kit 32 OLED version reference Vext, a MOSFET connected to pin 21 that controls the supply rail to the OLED and, if present, LoRa radio. My board for sure doesn't have these - if I set the pin incorrectly, the OLED still functions just fine. But the schematics in this repository also don't include any mention of such a MOSFET - are they out of date? How can I specifically check which hardware revision I have?

hb9eue, 2019-09-08:
I have more or less the same question. I use the board as a TTN tracker with a tiny GPS attached to Vext. That is nice, I can turn the GPS off when not needed.
But I am not sure what else I turn off. In the documentation I read that display and radio are also powered by Vext.
Indeed, the display is notably dimmer when Vext is off, but not completely off.
But why is it connected to Vext as the display library already has a functinon to turn the display off.

I'm more concerned about the radio. Over short distance the radio is still transmitting with Vext off. But is it, like the display, transmitting at reduced power?

Also in some examples I read about a radio PA. So is there maybe an additional RF amplifier on the board that can be controlled? If so, how?

It would be nice if this was clearly docmented instead of having to figure out by try and error. It took me quite a while getting the new board working with the LoraWAN LMIC code which just worked finde for the V1 board.

What I also wonder. When a battery is connected, the charging LED is very very bright until charged. I doubt this LED will survive long time at this brithtness.

hb9eue, 2019-09-08:
I am also quite concerned about Vext operation.
I use the board as a TTN Mapping Node with a tiny 'beitian' GPS connected.
My previous boards used a V.kel GPS which has a logic level 'enable' line, thus can be turned off by any pin. The beitian one has no such enable line, so I use Vext to cut of VCC to the GPS when not needed.
I noticed this dimms the display. According to the schematics, the display is powered only by Vext, so where is that leaking current comming from to keep the display on but dimmed, when Vext is off?
And why is Vext used to power the display anyway as the display itself can be put in power save mode independently.

But my bigger concern is the radio. The RF Switch (used in reverse) is powered by Vext. So if I turn Vext off, what does exactly happen when transmitting? Is the RF Amplifier going to burn because the antenna is disconnected by the switch? Is the RF Power being transmitted back to the input (and burning this)when the switch is unpowered?

Usualy I operate the node this way:

Wait Timer expired:
* Turn Vext on to fire UP GPS
* Turn Display on, display GPS Status
* GPS Fix acquired, check if we did move. If we don't Vext OFF => Wait more.
* If we moved: Vext OFF (we don't need to waste power on the GPS),
* Send LoraWan Packet
* Display LowaWAN Status after Transmit
* Display Off
* Wait Timer

Well as I transmit AFTER I turn Vext Off, I'm a bit concerned now. Is this the right thing to do or do I have to keep Vext ON while TX?