Heltec-Aaron-Lee/WiFi_Kit_series #62: Cannot connect a I2C sensor to ESP32

JazzmanMarburg, 2018-09-22:
Hello.
I am not able to connect an I2C sensor (BME280) to the Heltec Wifi 32 Kit. I have connected the BME280 to Pin 22 => SDA and Pin 21 => SCL. But I get always the response "No I2C Device found". I tried to run a ESP_I2C_Scanner but I get the response: 

- I2CScanner
- SDA Pin = 21
- SCL Pin = 22
- Scanning...
- I2C devices found

Could it be the that the pin layout is wrong?
I am pretty sure the sensor is ok - I have run it on a different Arduino Uno project. I tried a second one with same result. On the other hand the WiFi 32 Kit seems to be ok since I a code running with NTP server time - and it displays the time fine.
Attached you see my project setup and the code of the ESP_I2C_Scanner - do I something wrong?

Warm regards, Jasmin

![setup](https://user-images.gith
[ESP_I2C_Scanner.zip](https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series/files/2407827/ESP_I2C_Scanner.zip)
ubusercontent.com/21978392/45917738-4ead4300-be7c-11e8-87fd-184976b5f11a.jpg)

douglasnavarro, 2018-10-14:
I've been having the same issue with a different i2c sensor. Any luck?

JazzmanMarburg, 2018-10-16:
> Ok! ):
> I have made a little bit of progress being able to at least detect the i2c sensor, as well as the oled display, on the i2c bus using this scanner program<https://github.com/espressif/arduino-esp32/issues/977>. I also used external pull-up resistors.
> But I'm leaving this aside for now. If I by any chance get back on this issue I'll make sure to update you here.
> Thanks!

Congratulation!
This is by far more then I have achieved.
Let us know if any progress comes up.

I wonder why the manufacturer is not commenting on this obviously common problem. Also, that you needed external pull-up resistors is not mentioned anywhere.
Could you show us so far your setup?

Warm regards
Jasmin

tmsd2001, 2018-10-28:
Hello,
I have the same problem, the scanner sketch found both devices, the oled at 0x3C and the MMA8452 at 0x1C, but the sketch cannot communicate with the sensor. Do address scanning other than normal communication?

JazzmanMarburg, 2018-10-28:
I have no idea, what's going on. 
So the problem reading an I2C device with this board is still an open issue. 
Someone will have to contact the manufacturer and asking them for a solution - and pasting the solution here in the forum. 
Otherwise we will have to read from time-to-time this issue coming back for new readers.
Sorry!

travellingkiwi, 2018-11-06:
I have a Heltec 32 Wifi and a BME280. And I see a 76 on bus 2... But also can't read it using the Adafruit libs (No devices found).

I haven't done the pull-ups yet though...

canadaduane, 2018-12-24:
I think I have solved this issue:

I tried a sensor on the default I2C pins, 22 (SCL) & 21 (SDA), and it worked. However, after initializing the OLED display, communication with the sensor on pins 22 & 21 stopped working.

I discovered [here](https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series/blob/master/esp32/libraries/OLED/src/SSD1306Wire.h#L63) that when the OLED is initialized, it tells the Arduino Wire library to start using pins 15 (SCL) and 4 (SDA) for I2C communications.

If I use pins 15, 4 for both the OLED I2C and the sensor I2C, everything works--as long as I initialize the sensor after the OLED display (or, alternatively, initialize the Wire library to use 15, 4 right from the get-go). So basically, using the OLED library has the side-effect of re-assigning the global I2C pins for the esp32 for any Arduino libraries. The simplest solution is to just go with the flow and use the same pins as the OLED for all I2C communications (assuming your I2C sensor device addresses don't collide with the OLED's).

itofficeeu, 2018-12-26:
I can confirm that the above solution of @canadaduane works. At least on Heltec WiFi LoRa 32 v2. This is in a combination of the OLED and the BME280. By random do I here use SoftWire.h, but I guess it can run with the standard Wire.h.

```
#include <Wire.h>
#include <SPI.h>
#include "SSD1306.h"
#include <Adafruit_Sensor.h>
#include <Adafruit_BME280.h>

// BME280
#define SEALEVELPRESSURE_HPA (1032.00)
Adafruit_BME280 bme; // I2C

#define PIN_SDA 4
#define PIN_SCL 15

#include <AsyncDelay.h>
#include <SoftWire.h>
SoftWire sw(PIN_SDA, PIN_SCL);

SSD1306 display(0x3C, 4, 15);

void setup() 
{
  Serial.begin(115200);
  Serial.printf("Starting...\r\n");

  pinMode(16,OUTPUT);
  digitalWrite(16, LOW); // set GPIO16 low to reset OLED
  delay(50);
  digitalWrite(16, HIGH);
  delay(50);
  display.init();
  display.setFont(ArialMT_Plain_10);
  delay(50);
  display.drawString( 0, 0, "Starting up ...");
  display.drawString( 0,20, "- and initializing.");
  display.display();

  bool bme_status;
  bme_status = bme.begin();  
  if (!bme_status) {
      Serial.println("Could not find a valid BME280 sensor, check wiring!");
  }
}

void loop() 
{
  // Print temperature
  float temp = bme.readTemperature();
  Serial.print(temp);
  Serial.print(";");

  float pres = bme.readPressure() / 100.0F;
  Serial.print(pres);
  Serial.print(";");

  // Print humidity
  float humi = bme.readHumidity();
  Serial.print(humi);
  Serial.print(";");

  // Print altitude
  float alti = bme.readAltitude(SEALEVELPRESSURE_HPA);
  Serial.println(alti);
}
```

3KUdelta, 2019-03-20:
Running sample code on itofficeeu's solution with data showed on the display and in serial monitor. 
Thank you, itofficeeu. 
https://github.com/3KUdelta/heltec_wifi_kit_32_bme280
Enjoy

r1k, 2019-07-07:
Thanks, for the information here.

However I've found you can use pins 21/22.  
I'm not using a Heltec ESP32 but a clone with the same pin allocation but a slightly different PCB layout, however it has the same problem.

Iniside `Wire.cpp` it defines the `TwoWire` class. At the bottom of the file it creates two instances of the for 2 separate buses

    TwoWire Wire = TwoWire(0);
    TwoWire Wire1 = TwoWire(1);

(I don't particularly like that it creates these 'hidden' global object instances even though you may never use them.)

The OLED will use `Wire`.  

So for the sensor bus use `Wire1` and initialise it to use pins 21/22 by calling `Wire1.begin(21, 22);` 

Then pass this to the `Adafruit_BME280 bme;` object during the begin call:

`bme.begin(0x76, &Wire1);` 

then it should initialise fine and work on pins 21/22.

My modification to the test code from @itofficeeu above is below, (note the use of `SSD1306Wire` instead of `SSD1306` and the changes mentioned above.)

Hope this helps

    #include <Wire.h>
    #include <SPI.h>
    #include "SSD1306.h"
    #include <Adafruit_Sensor.h>
    #include <Adafruit_BME280.h>

    // BME280
    #define SEALEVELPRESSURE_HPA (1019) //Check Internet for MSL pressure at our place to calc your 
    elevation mASL

    Adafruit_BME280 bme;

    #define PIN_SDA 4
    #define PIN_SCL 15

    SSD1306Wire display(0x3C, PIN_SDA, PIN_SCL);

    float SLpressure_hPa;

    void setup() 
    {

        pinMode(LED, OUTPUT);
        digitalWrite(LED, HIGH);

        pinMode(16,OUTPUT);
        digitalWrite(16, LOW); // set GPIO16 low to reset OLED
        delay(50);
        digitalWrite(16, HIGH);
        delay(50);
        display.init();
        display.setFont(ArialMT_Plain_10);
        delay(50);
        display.drawString( 0, 0, "Starting up ...");
        display.drawString( 0,12, "- and initializing.");
        display.display();
        delay(500);

        bool wireStatus = Wire1.begin(21, 22);
        if (!wireStatus)
        {
            display.clear();
            display.drawString( 0, 0, "Wire1 failed to init");
            display.display();
            delay(3000);
        }

        bool bme_status = bme.begin(0x76, &Wire1);  //address either 0x76 or 0x77
        if (!bme_status) 
        {
            display.clear();
            display.drawString( 0, 0, "No valid BME280 found");
            display.drawString( 0, 12, "Please check wiring!");
            display.display();
        }

        // my best way to get correct indoor temperatures
  
        bme.setSampling(Adafruit_BME280::MODE_FORCED,
                        Adafruit_BME280::SAMPLING_X16,  // temperature
                        Adafruit_BME280::SAMPLING_X1, // pressure
                        Adafruit_BME280::SAMPLING_X1,  // humidity
                        Adafruit_BME280::FILTER_X16,
                        Adafruit_BME280::STANDBY_MS_0_5 );
    }

    void loop() 
    {
        constexpr int ROW_HEIGHT = 15;

        enum side_e {
            left = 0,
            right = 55
        };

        auto Write = [] (const String &msg, int row, enum side_e side) {
            display.drawString(side, row * ROW_HEIGHT, msg);
        };

        bme.takeForcedMeasurement();
        // Print temperature
        const auto temp = bme.readTemperature();

        display.clear();

        Write("Temp.", 0, left);
        Write(String(temp) + (" °C"), 0, right);

        // Print humidity
        const auto humi = bme.readHumidity();

        Write("Humidity", 1, left);
        Write(String(humi) + (" %"), 1, right);

        // Print altitude
        const auto alti = bme.readAltitude(SEALEVELPRESSURE_HPA);

        Write("Altitude", 2, left);
        Write(String(alti) + (" m ASL"), 2, right);

        // Print pressure
        const auto pres = bme.readPressure() / 100.0F;

        Write("Pressure", 3, left);
        Write(String(pres) + (" hPa"), 3, right);

        display.display(); // print display buffer
        delay(1000);
    }

lds000, 2019-07-16:
Taking inspiration from @r1k above, I was able to connect the Adafruit BNO055 to the Heltect OLED ESP 32 wifi kit using the 21/22 pins as well.  This was a very helpful post!

`#include <Wire.h>
#include <Adafruit_Sensor.h>
#include <Adafruit_BNO055.h>
#include "SSD1306.h"

int incoming;
//int LED_BUILTIN = 2;

// the OLED used
#define PIN_SDA 4
#define PIN_SCL 15

//Wire sw(PIN_SDA, PIN_SCL);

SSD1306 display(0x3C, 4, 15, 16); //Set OLED to default address 0x3C, and pins 4 and 15, with reset at pin 16

double xPos = 0, yPos = 0, headingVel = 0;
uint16_t BNO055_SAMPLERATE_DELAY_MS = 10; //how often to read data from the board
uint16_t PRINT_DELAY_MS = 100; // how often to print the data
uint16_t printCount = 0; //counter to avoid printing every 10MS sample

//velocity = accel*dt (dt in seconds)
//position = 0.5*accel*dt^2
double ACCEL_VEL_TRANSITION =  (double)(BNO055_SAMPLERATE_DELAY_MS) / 1000.0;
double ACCEL_POS_TRANSITION = 0.5 * ACCEL_VEL_TRANSITION * ACCEL_VEL_TRANSITION;
double DEG_2_RAD = 0.01745329251; //trig functions require radians, BNO055 outputs degrees

Adafruit_BNO055 bno = Adafruit_BNO055(55, 0x28, &Wire1); //Connect to BNO055, with I2C address 0x28, setting Wire1 from Wire.h

String strMessageBack;

void setup(void)
{

  Serial.begin(115200);

  pinMode(16,OUTPUT);
  digitalWrite(16, LOW); // set GPIO16 low to reset OLED
  delay(50);
  digitalWrite(16, HIGH);
  delay(50);
  display.init();
  display.setFont(ArialMT_Plain_10);
  delay(50);
  display.drawString( 0, 0, "Starting up ...");
  display.drawString( 0, 15, "- and initializing.");
  display.display();

  Wire1.begin(21, 22);
  if (!bno.begin())
  {

        Serial.print("BNO055 not found.");
        delay(1000);
   while (1);
  }

   display.drawString(0, 30, "BNO055 found!");
   display.display();

  pinMode (LED_BUILTIN, OUTPUT);//Specify that LED pin is output
  delay(1000);
  strMessageBack = "";
}

void loop(void)
{

  unsigned long tStart = micros();
  sensors_event_t orientationData , linearAccelData;
  bno.getEvent(&orientationData, Adafruit_BNO055::VECTOR_EULER);
  //  bno.getEvent(&angVelData, Adafruit_BNO055::VECTOR_GYROSCOPE);
  bno.getEvent(&linearAccelData, Adafruit_BNO055::VECTOR_LINEARACCEL);

  xPos = xPos + ACCEL_POS_TRANSITION * linearAccelData.acceleration.x;
  yPos = yPos + ACCEL_POS_TRANSITION * linearAccelData.acceleration.y;

  // velocity of sensor in the direction it's facing
  headingVel = ACCEL_VEL_TRANSITION * linearAccelData.acceleration.x / cos(DEG_2_RAD * orientationData.orientation.x);

  if (printCount * BNO055_SAMPLERATE_DELAY_MS >= PRINT_DELAY_MS) {
        display.clear();
        display.drawString(0,0,String(orientationData.orientation.x));
        display.drawString(0,15,String(orientationData.orientation.y));
        display.drawString(0,30,String(orientationData.orientation.z));
        display.display();
    printCount = 0;
  }
  else {
    printCount = printCount + 1;
  }

  while ((micros() - tStart) < (BNO055_SAMPLERATE_DELAY_MS * 1000))
  {
    //poll until the next sample is ready
  }
}
`

[image]

Epaminondas1966, 2019-07-17:
> Dear r1k is it possible to send a print of Wire.cpp with the modifictaion to work with Wire1?
I'm having problem to modified the wire.cpp
many thanks

Epaminondas1966, 2019-07-17:
Dear @r1k is it possible to send a print of Wire.cpp with the modifictaion to work with Wire1?
I'm having problem to modified the wire.cpp
many thanks

r1k, 2019-07-18:
I made no changes to wire.cpp, Wire1 is already declared in that file.  You just need to initalise it with the correct pin allocation by calling:

`Wire1.begin(21,22);`

and then pass that to the bme object when you create it along with the address to use:

`Adafruit_BME280 bme;`
`bme.begin(0x76, &Wire1);`   <- where 0x76 is the address of the device you're using

If that doesn't work then check you have the correct physical pins, the clone board I am using is different to the actual Heltec board and I didn't realise to begin with

Epaminondas1966, 2019-07-18:
@r1k ok thanks

proffalken, 2019-11-19:
Hey all,

Thanks for all the information here.

I'm working with an older I2C motor driver (https://wiki.wemos.cc/products:retired:motor_shield_v1.0.0 ) and I don't have the option of setting the SDA/SCL pins.

I've tried using pins 15 & 4 as above, however that doesn't appear to have worked either.

Any ideas what I might be doing wrong?

```
#include "Arduino.h"
#include "SPI.h"
#include "U8g2lib.h"
#include "WEMOS_Motor.h"

#ifndef LED_BUILTIN
#define LED_BUILTIN 25
#endif

// setting PWM properties
const int freq = 5000;
const int ledChannel = 0;
const int resolution = 8;

int buzzer = 13;

U8G2_SSD1306_75X64_NONAME_F_HW_I2C  u8g2(U8G2_R0, 16, 15, 4);

Motor M1(0x30,_MOTOR_A, 1000);//Motor A
Motor M2(0x30,_MOTOR_B, 1000);//Motor B

void setup()
{
  // initialize LED digital pin as an output.
  pinMode(LED_BUILTIN, OUTPUT);
  // configure LED PWM functionalitites
    
  // attach the channel to the GPIO to be controlled
  u8g2.begin();
  u8g2.setFont(u8g2_font_6x10_tf);
  u8g2.setFontRefHeightExtendedText();
  u8g2.setDrawColor(1);
  u8g2.setFontPosTop();
  u8g2.setFontDirection(0);
}

void loop()
{
  int cycleDuration = random(100,3000);
  int cyclePauseDuration = random(100,3000);
  int cycleRampUpStep = random(100,1500);
  int cycleRampDownStep = random(100,1500);

  u8g2.clearBuffer();
  u8g2.setCursor(0,8);
  u8g2.print(cycleDuration);
  u8g2.setCursor(0,18);
  u8g2.print(cyclePauseDuration);
  u8g2.setCursor(0,28);
  u8g2.print(cycleRampUpStep);
  u8g2.setCursor(0,38);
  u8g2.print(cycleRampDownStep);
   // increase the LED brightness
  for(int dutyCycle = 0; dutyCycle <= 75; dutyCycle++){   
    // changing the LED brightness with PWM
    M1.setmotor( _CW, dutyCycle);
    M2.setmotor(_CW, dutyCycle);
    
    u8g2.setCursor(0,0);
    u8g2.print(dutyCycle);
    u8g2.sendBuffer();
    delay(cycleRampUpStep);
  }

  // turn the LED on (HIGH is the voltage level)
  digitalWrite(LED_BUILTIN, HIGH);
  // wait for a second
  delay(cyclePauseDuration);

  // turn the LED off by making the voltage LOW
  digitalWrite(LED_BUILTIN, LOW);

  // decrease the LED brightness
  for(int dutyCycle = 75; dutyCycle >= 0; dutyCycle--){
    // changing the LED brightness with PWM
    M1.setmotor( _CW, dutyCycle);
    M2.setmotor(_CW, dutyCycle);
    
    u8g2.setCursor(0,0);
    u8g2.print(dutyCycle);
    u8g2.sendBuffer();
    delay(cycleRampDownStep);
  }
   // wait for a second
  delay(cycleDuration);
}
```
The above code works fine if I comment out the graphics stuff, but as soon as I add it back in, the screen works and the motors don't.

Epaminondas1966, 2019-11-22:
Try to modifie the arduino.h of the Heltec board, all the boards has a declaration pins in this file you can see where is the file inside of your computer in the arduino IDE option, there try to change the I2c pins for others, compile and see if it’s work 

Enviado do Yahoo Mail para iPhone

devincentiis, 2020-02-04:
Hello.
I have used BMx280TwoWire example in Gregor Christandl BMx280MI library on Heltec Wifi Kit32 and work fine with pins 21-22 or 4-15, as long as you call it in setup:

`Wire.begin (4.15);`

https://bitbucket.org/christandlg/bmx280mi/src/master/

You can find it in the Arduino IDE libraries.

Best regards at all

gundrabur, 2021-02-07:
Use the BMP180 pressure sensor with a Heltec WiFi Kit 32 at the same time as the built-in OLED display.
It is possible! I got it to work. Please find a complete solution here:
https://github.com/gundrabur/BMP180-on-Heltec

andressakreutz, 2021-04-14:
Hi!

I have a ESP32 38 pins (https://produto.mercadolivre.com.br/MLB-1717757954-esp-32-38-pinos-_JM) and I want to connect a display OLED 128x32 and a BME280. So, I have connected the display on pins 21 e 22 and the BME280 on pins 4 e 15, but that does'nt work. When I run the code to identify the I2C address from them, I got a message "No I2C devices found". Any tip or help? Anybody who had work with the board ESP32 38 pins?

itofficeeu, 2021-04-14:
Do you know of this tutorial?:
https://randomnerdtutorials.com/esp32-i2c-communication-arduino-ide/
Start with only the BME280 first. Then try the OLED without the BME280. When both run without the other, then try to connect both at the same time.
Note that software from Adafruit has hidden settings, adapted to hardware from Adafruit. You might need to change the address of the hardware in the software, if you us software from Adafruit. Usually done in the library src files, the header file (ending on .h).

andressakreutz, 2021-04-14:
Yes, I was trying to do that with this site; and I also try only with the bme280 and only with the display, but both does'nt work. When I run the code from Random Tutorials to find de I2C adress, I get the message "No I2C devices found". Also when I run the code bellow, the same message of faill comes.

#include <Wire.h>
#include <Adafruit_Sensor.h>
#include <Adafruit_BME280.h>

#define I2C_SDA 21
#define I2C_SCL 22

#define SEALEVELPRESSURE_HPA (1013.25)

TwoWire I2CBME = TwoWire(0);
Adafruit_BME280 bme;

unsigned long delayTime;

void setup() {
  Wire.begin(21, 22); //I try with this line commented and uncommented
  Serial.begin(115200);
  Serial.println(F("BME280 test"));
  I2CBME.begin(I2C_SDA, I2C_SCL, 100000);

  bool status;

  // default settings
  // (you can also pass in a Wire library object like &Wire2)
  status = bme.begin(0x76, &I2CBME);  
  if (!status) {
    Serial.println("Could not find a valid BME280 sensor, check wiring!");
    while (1);
  }

  Serial.println("-- Default Test --");
  delayTime = 1000;

  Serial.println();
}

void loop() { 
  printValues();
  delay(delayTime);
}

void printValues() {
  Serial.print("Temperature = ");
  Serial.print(bme.readTemperature());
  Serial.println(" *C");
  
  // Convert temperature to Fahrenheit
  /*Serial.print("Temperature = ");
  Serial.print(1.8 * bme.readTemperature() + 32);
  Serial.println(" *F");*/
  
  Serial.print("Pressure = ");
  Serial.print(bme.readPressure() / 100.0F);
  Serial.println(" hPa");

  Serial.print("Approx. Altitude = ");
  Serial.print(bme.readAltitude(SEALEVELPRESSURE_HPA));
  Serial.println(" m");

  Serial.print("Humidity = ");
  Serial.print(bme.readHumidity());
  Serial.println(" %");

  Serial.println();
}

I didn't understand which changes I should make in the src file. Following this way: Computer -> Documents -> Arduino -> Libraries -> Softwire -> src -> SoftWire.h, the file that I get is annexed. Is this one who I should change? And if yes, how?

Thanks!!

[SoftWire.txt](https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series/files/6313778/SoftWire.txt)

peterl3233, 2021-09-08:
Brilliant solution: This worked for me on Heltec Wifi Lora v1 .. I used SCL on pin 12 and SDA on pin13

finncat62, 2022-11-19:
Hi, I tried to do the solution here -- using wire1 to make another bus on 21 and 22, in the setup as shown below. I'm using a Heltec-Lora 32 V2. It gets through everything -- wire1 and bme280 initialization and prints "got here" as I wanted, but then when displaying values everything is filled with 0.

So it's connecting to the board but not getting anything. Is the board bad? Or is it something with my code or i2c? Thanks!
```
setup code where I do what was recommended here (full code after this)
void setup() {
  Heltec.begin(true /*DisplayEnable Enable*/, true /*Heltec.Heltec.Heltec.LoRa Disable*/, true /*Serial Enable*/, true /*PABOOST Enable*/, BAND /*long BAND*/);
  
  Heltec.display->init();
  Heltec.display->setFont(ArialMT_Plain_10);
  Heltec.display->setContrast(255);  
  Heltec.display->setTextAlignment(TEXT_ALIGN_CENTER);

  Heltec.display->clear();
  Heltec.display->drawString(30,10,"Hello world");
  Heltec.display->display();

  delay(500);

  Heltec.display->setTextAlignment(TEXT_ALIGN_LEFT);
  while(!Wire1.begin(21,22)){
    Heltec.display->clear();
    Heltec.display->drawString( 0, 0, "Wire1 failed to init");
    Heltec.display->display();
    delay(1000);
  }
  while(!bme.begin(0x76,&Wire1)) {
    Heltec.display->clear();
    Heltec.display->drawString(30,10,"NO BME ERROR )@$");
    Heltec.display->display();
    /*erial.println("Could not find a valid BME280 sensor, check wiring, address, sensor ID!");
    Serial.print("SensorID was: 0x"); Serial.println(bme.sensorID(),16);
    Serial.print("        ID of 0xFF probably means a bad address, a BMP 180 or BMP 085\n");
    Serial.print("   ID of 0x56-0x58 represents a BMP 280,\n");
    Serial.print("        ID of 0x60 represents a BME 280.\n");
    Serial.print("        ID of 0x61 represents a BME 680.\n");*/
    delay(100);
  }
    
  Heltec.display->clear();
  Heltec.display->drawString(30,10,"got here");
  Heltec.display->display();
  delay(2000);
}
```

full code:
`
#include "heltec.h"

#include <Adafruit_BME280.h>
#include <Wire.h>
#include <Adafruit_Sensor.h>

#define BAND    915E6  //you can set band here directly,e.g. 868E6,915E6

#define SEALEVELPRESSURE_HPA (1013.25)

Adafruit_BME280 bme;

#define logo_width 128
#define logo_height 53
const unsigned char logo_bits[] = {
   0x00, 0x00, 0x38, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
  0x00, 0x00, 0x00, 0x00, 0x70, 0x00, 0xF8, 0x00, 0x00, 0x00, 0x00, 0x00, 
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xE0, 0x01, 0xF0, 0x03, 
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
  0xC0, 0x07, 0xF0, 0x07, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
  0x00, 0x00, 0x00, 0x00, 0xC0, 0x0F, 0xF0, 0x0F, 0x00, 0x00, 0x00, 0x00, 
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xC0, 0x1F, 0xF0, 0x1F, 
[... 122 lines cut ...]

void displayValues(Environment_Data dataset){
  readValues(dataset);
  Heltec.display->clear();
  Heltec.display->drawString(10,0,"temp: " + String(dataset.temp) + " c" );
  Heltec.display->drawString(10,15,"pres: " + String(dataset.pres) + " hPa");
  Heltec.display->drawString(10,30,"hum: " + String(dataset.hum) + " %rh");
  Heltec.display->drawString(10,45,"Alt: " + String(dataset.alt) + " m");
  Heltec.display->display();
}
void readValues(Environment_Data dataset){
  dataset.temp = bme.readTemperature();//c
  dataset.pres = (bme.readPressure() / 100.0F);//hpa
  dataset.alt = bme.readAltitude(SEALEVELPRESSURE_HPA);
  dataset.hum = bme.readHumidity();
}
`

It displays
"
temp: 0.00 c
pres: 0.00 hPa
hum: 0.00 %rh
Alt: 0.00 m
"

plumyop, 2024-02-15:
I have a Heltec Lora 32 v2 with a sensor BMP280 and after searching for hours for finding the issue, it finaly works. Thanks for Canadaduane for his message who was really helpfull.  So I connected the two wires of the sensor on the same pin as the Oled screen  (15  &4) and it worked with this code: 

#include <Wire.h>
#include <SPI.h>
#include <Adafruit_BMP280.h>
#include <heltec.h>          // Heltec Library

Adafruit_BMP280 bmp; // I2C

void setup() {
  Heltec.begin(true /*DisplayEnable Enable*/, false /*LoRa Disable*/, true /*Serial Enable*/);

  while ( !Serial ) delay(100);   // wait for native usb
  Serial.println(F("BMP280 test"));
  unsigned status;
  status = bmp.begin(0x76);
  if (!status) {
    Serial.println(F("Could not find a valid BMP280 sensor"));
    while (1) delay(10);
  }

}

void show(int t, int a, int p){
  Heltec.display->clear();                           // clear the display
  Heltec.display->setFont(ArialMT_Plain_10); // Text size
  Heltec.display->drawString(10,10, "Temp:"+String (t)); // draw the string
  Heltec.display->drawString(10,20, "Altitude:"+String (a)); // draw the string
  Heltec.display->drawString(10,30, "Pressure:"+String (p)); // draw the string
  Heltec.display->display();                        // transfer it to display
}

void loop() {
    show(bmp.readTemperature(),bmp.readAltitude(1013.25),bmp.readPressure()/100);
    delay(2000);
}