Heltec-Aaron-Lee/WiFi_Kit_series #65: WIFI_LoRa_32 Module Hangs Up when running for several hours

protopals, 2018-11-02:
Below shows my setup:
                                      
                                              (50 meters)
[LoRa Device (sensor)] <--------------------------> [Heltec WiFi LoRa 32]  <------> (WiFi Network)

Essentially, the Heltec WiFi LoRa serves as a gateway to the cloud.

I am currently experiencing an issue where the Heltec WiFi LoRa module just hangs up after running for several hours. It works okay that first time you power it up in receiving and transmitting data to the LoRa module that is 50 meters away. The amount of data transferred is not really that many -- only about 20 bytes, and only happens once or twice a day. So basically, the Heltec WiFi LoRa module is just waiting for any data packet coming from the external LoRa device. After about a day or so, the Heltec WiFi LoRa hangs up when it receives data packet from the LoRa device. I am just wondering if this issue is already known, and if a fix is available.

I wish I can capture exception code when it happens, but the issue occurs so intermittently -- sometimes within 24 hours or sometimes within 72 hours from the time the module is powered on.

kampto, 2018-11-05:
Is it really hung up or is just the OLED hung up? My OLED hangs up after a few hours but the board is still functioning but OLED is froze.  So to get it going again I have a function in the sketch to reset the OLED after a given time. In my case every 2 hours.
```
void OledReset(){
  if (millis() - OledResetTimer > OledResetDelay) {
//// Reset OLED incase it got hung up
    pinMode(Oled_RST,OUTPUT); digitalWrite(Oled_RST, LOW); delay(100); digitalWrite(Oled_RST, HIGH); // while OLED is running, must set GPIO16 in high
    display.init(); display.flipScreenVertically(); display.setFont(ArialMT_Plain_24); // Initialising the UI will init the display too.
    display.drawString(0, 0, "Reset");
    display.drawString(0, 24, "OLED...");
    display.display(); delay(600); display.clear();
    OledResetTimer = millis(); 
    }
 }
```

kampto, 2018-11-05:
I dont know what triggers my OLED to hang up. Just noticed it was froze up occasionally when I go to look at it. Was figuring it was an i2c issue since its constantly scrolling data. But also could be related to a send / receive event which I have set up to about every couple minutes.

protopals, 2018-11-05:
@kampto I see. Mine is more of static texts (mainly for development purposes) that only changes when I get a new data off the LoRa channel.  I am using the OLED driver that uses the full buffer.

protopals, 2018-11-09:
I have determined that it is the MCU that hangs up and not the OLED module. Hang up happens when receiving data from a LoRa node. Not really sure what's happening here as it occurs intermittently (several days).

KevinMOcean, 2018-12-07:
Use a watchdog. Especially if always works within say a 12 hour period. Just curious, what is the environment temperature you're testing in?