Heltec-Aaron-Lee/WiFi_Kit_series #70: URGENT Windows Defender reports Trojan:Win32/Skeeyah.B!rfn in heltec\esp32\tools\espota.exe

liamkennedy, 2018-12-07:
Following the installation instructions here 
`git clone https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series.git heltec
`
My windows defender reported a trojan present in 
`C:\Users\username\OneDrive\Documents\Arduino\hardware\heltec\esp32\tools\espota.exe`

What the heck?

ddrager, 2018-12-10:
See malware detection from [VirusTotal](https://www.virustotal.com/#/file/5e2ff3daf359cc549dd64ac03b1d349076baf9df09bc0078cd6e99d445597c61/detection). Further analysis from [Hybrid Analysis](https://www.hybrid-analysis.com/sample/5e2ff3daf359cc549dd64ac03b1d349076baf9df09bc0078cd6e99d445597c61). 

Although this looks suspicious, the network addresses it looks to be reaching out to are 87.236.156.136 and 93.184.221.240 which map to 2 CDN providers. It appears the .exe is a compiled python2.7 binary which may just be downloading files to your system for update - the 'ota' name, or "over the air" extension would match up with that analysis. @Heltec-Aaron-Lee can you please provide the source for this file?

xieliwei, 2018-12-11:
The source is available in the general ESP32 repository.

https://github.com/espressif/arduino-esp32/blob/master/tools/espota.py

This looks suspicious, since the OTA tool is for direct connection between the Arduino computer and the ESP32, it should not be accessing anything on the internet. Follow the main thread on the arduino-esp32 issue for updates as well:

https://github.com/espressif/arduino-esp32/issues/2163

buccaneer-jak, 2018-12-25:
24th Dec, 2 Trojans reported by windows Defender