Heltec-Aaron-Lee/WiFi_Kit_series #78: Battery level below 3.7v and board can't start

ed-french, 2018-12-30:
Using 1000mAh Lithium battery: 
https://www.amazon.co.uk/gp/product/B07CYMYMS9

Sketch using WiFi in AP mode from a few seconds after boot.

Battery fully charged- all works OK.

Leaving battery installed and left on charge, battery charges fully, then coasts until below a threshold, then charges again. Removing from charge at low point in that cycle is 3.7V open circuit.

Board repeatedly tries to boot, cannot complete. 3.3V rail shows dips in voltage during boot.

Replaced  OR'ing diode, D1 -> 1N5819 with 1N5817 (lower forward voltage drop) and added 330uF cap to 3.3v line allows board to boot again.

martinberlin, 2019-01-08:
Please make a hardware upgrade. I'm facing the same problem with your boards and I need to use a pololu step up to power the board using 3.3v
I will be willing to test next version when there is an upgrade regarding this.

martinberlin, 2019-01-15:
For me the best way to do this till now: 

I use a pololu step up/down to 3.3 and power the board using 3.3v pin. And then I put a Shotky diode from the board battery plug to  the Lion 3.7 battery so the energy flows only to charge the battery but not to power the board. That has so far the best results for me. 
It's also possible to use a normal diode and get 0.7 volts down from battery to 3.3v pin, but I find the pololu giving longer battery life in my case. It's a pity it does not work like expected, not only on this board, so far all the boards that I use have the battery plug as a UPS system that is not at all prepared to make an embedded device. If Heltec engenieer teams adds this then it will be a great forward.

ed-french, 2019-03-19:
You are right, I've struggled to find any ESP32 boards with good power management. Some do better than others. I suspect this will wait for a nice "PMIC" type IC for these applications that's low enough cost. Ideally it would:

- charge & discharge a lithium or LiPo battery, ideally with a settable charge rate. 

- It would have battery protection to stop over-discharge. It would have a low quiescent current running off battery

- switch off the USB serial (or allow this in code) to reduce current consumption closer to the ESP32's own.

- It would handle using the USB power down to about 4.5V and still charge the battery.

- It would have a low enough drop-out and resistance to help cope with the current spikes from the WiFi without needing a big capacitor.