Heltec-Aaron-Lee/WiFi_Kit_series #81: Packets are not received after some time

elxsi, 2019-01-12:
Hi,

I've two modules (one sender and one receiver). The receiver somehow stops receiving packets after some time (about 15 to 60 minutes). The sender still sends correct packets (tested via a third module, also looked with a rtl-sdr at 433 MHz). After resetting the receiver it also again works for some time. The receiving module is not hanging (i.e. it's still executing code, only the LoRa.parsePacket() function does not return packet sizes > 0). I'm using the following code on the receiver side:

#include <SPI.h>
#include <LoRa.h>
#include "SSD1306.h"
#include <Arduino.h>
#include <Timer.h>

/* Serial baudrate */
#define SERIAL_BAUD 115200

/* OLED display */
#define OLED_SDA      4
#define OLED_SCL      15
#define OLED_RST      16

SSD1306 display(0x3c, OLED_SDA, OLED_SCL);

/* SX1278 (LoRa) */
#define SX1278_SCK    5
#define SX1278_MISO   19
#define SX1278_MOSI   27
#define SX1278_CS     18
#define SX1278_RESET  14
#define SX1278_IRQ    26

/* LoRa */
#define LoRa_SS       18
#define LoRa_RST      14
#define LoRa_DI0      26
#define LoRa_BAND     433E6
#define LoRa_PABOOST  true
#define LoRa_SYNCWORD 0x5F

/* LED PIN */
#define LED_PIN       25

/* Timer */
Timer timer;

/* Setup function */
void setup() {
  /* Begin serial */
  Serial.begin(SERIAL_BAUD);
  
  /* Reset OLED */
  Serial.println("Reseting OLED display");
  pinMode(OLED_RST, OUTPUT);
  digitalWrite(OLED_RST, LOW);
  delay(50); 
  digitalWrite(OLED_RST, HIGH);

  /* Initialize OLED display */
  Serial.println("Initializing OLED display");
  display.init();
  display.flipScreenVertically();
  display.setFont(ArialMT_Plain_10);
  display.setTextAlignment(TEXT_ALIGN_LEFT);
  display.drawString(8, 5,"-= LoRa Receiver =-");
  display.display();

  /* SPI setup */
  Serial.println("Initializing SPI");
  SPI.begin(SX1278_SCK, SX1278_MISO, SX1278_MOSI, SX1278_CS);

  /* Initialize LoRa */
  Serial.println("Initialize LoRa at band: " + String(LoRa_BAND/1000000) + " MHz");
  LoRa.setPins(LoRa_SS, LoRa_RST, LoRa_DI0);
  if (!LoRa.begin(LoRa_BAND, LoRa_PABOOST)) {
    Serial.println("LoRa Init failed - Halted");  
    display.drawString(5, 35,"LoRa Init failed - Halted");
    display.display();
    while(1);
  }
  
  Serial.println("LoRa Init OK");
  display.drawString(5, 15, "LoRa Init OK");
  display.drawString(5, 35, "Waiting for packets...");
  display.display();
  delay(2000);

  /* Set sync word */
  LoRa.setSyncWord(LoRa_SYNCWORD);

  /* Show ESP heap */
  timer.every(1000, espHeap);
}

/* Loop */
void loop() {
  static unsigned long cnt = 0;

  /* Update timer */
  timer.update();
  
  /* Read LoRa packets */
  int packetSize = LoRa.parsePacket();
  if (packetSize) {
    cnt++;

    display.clear();
    display.drawString(8, 5, "-= elxsi LoRa Receiver =-");
    display.drawString(5, 15, "f: " + String(LoRa_BAND/1000000) + " MHz, SW: " + String(LoRa_SYNCWORD, HEX));
    display.drawString(5, 25, "P#: " + String(cnt) + ", Len: " + String(packetSize));
    display.drawString(5, 35, "RSSI: " + String(LoRa.packetRssi()) + ", SNR: " + String(LoRa.packetSnr()) + "dB");
        
    Serial.println("Received LoRa packet (Length: " + String(packetSize) + 
                 " bytes, RSSI: " + String(LoRa.packetRssi()) + ", " +
                 "SNR: " + String(LoRa.packetSnr()) + "db, " +
                 "SW: " + String(LoRa_SYNCWORD, HEX));
    Serial.println("Packet dump:");

    String pkt = "";
    while (LoRa.available()) {
      pkt += ((char) LoRa.read());
    }
    Serial.println(pkt);
    display.drawString(5, 50, pkt);

    display.display();
  }
}

/* Show ESP heap */
void espHeap() {
  Serial.println("Free heap space: " + String(ESP.getFreeHeap()) + " bytes");
}

The serial output of the last few packets shows:

Received LoRa packet (Length: 19 bytes, RSSI: -111, SNR: 9.75db, SW: 5f
Packet dump:
16:24:54 12/01/2019
Free heap space: 370968 bytes
Received LoRa packet (Length: 19 bytes, RSSI: -112, SNR: 10.25db, SW: 5f
Packet dump:
16:24:55 12/01/2019
Free heap space: 370968 bytes
Received LoRa packet (Length: 19 bytes, RSSI: -112, SNR: 6.00db, SW: 5f
Packet dump:
16:24:56 12/01/2019
Free heap space: 370968 bytes
Received LoRa packet (Length: 19 bytes, RSSI: -113, SNR: 10.75db, SW: 5f
Packet dump:
16:24:57 12/01/2019
Free heap space: 370968 bytes
Received LoRa packet (Length: 19 bytes, RSSI: -112, SNR: 6.00db, SW: 5f
Packet dump:
16:24:58 12/01/2019
Free heap space: 370968 bytes
Received LoRa packet (Length: 19 bytes, RSSI: -112, SNR: 6.00db, SW: 5f
Packet dump:
16:24:59 12/01/2019
Free heap space: 370968 bytes
Received LoRa packet (Length: 19 bytes, RSSI: -112, SNR: 11.25db, SW: 5f
Packet dump:
16:25:00 12/01/2019
Free heap space: 370968 bytes
Received LoRa packet (Length: 19 bytes, RSSI: -112, SNR: 10.00db, SW: 5f
Packet dump:
16:25:01 12/01/2019
Free heap space: 370968 bytes
Received LoRa packet (Length: 19 bytes, RSSI: -112, SNR: 6.00db, SW: 5f
Packet dump:
16:25:02 12/01/2019
Free heap space: 370968 bytes
Free heap space: 370968 bytes
Free heap space: 370968 bytes
Free heap space: 370968 bytes
Free heap space: 370968 bytes
Free heap space: 370968 bytes
Free heap space: 370968 bytes
Free heap space: 370968 bytes
Free heap space: 370968 bytes
Free heap space: 370968 bytes
Free heap space: 370968 bytes
Free heap space: 370968 bytes
Free heap space: 370968 bytes
Free heap space: 370968 bytes
Free heap space: 370968 bytes
Free heap space: 370968 bytes
Free heap space: 370968 bytes
Free heap space: 370968 bytes
Free heap space: 370968 bytes
Free heap space: 370968 bytes
Free heap space: 370968 bytes

As you can see, the free heap space is still printed, so the esp is still executing code.
Only the int packetSize = LoRa.parsePacket(); does not return packet sizes greater 0,
so the corresponding if branch does not get executed.

Any ideas?

Thank you!

andrelmbraga, 2019-02-04:
I have the same problem. Did you fix it?

Samuel-ZDM, 2019-03-27:
Hi. You have to see the value of spread factor. In this link https://os.mbed.com/docs/mbed-os/v5.8/tutorials/LoRa-tutorial.html in the subchapter, Some notes on writing firmware.  I think that's it.

Pmartinchile, 2021-02-10:
Same issue here with a Heltec 32 v2  with a LoRa 1278, sounds like a buffer issue