Heltec-Aaron-Lee/WiFi_Kit_series #84: Provide KICAD Footprint for LoRa WiFi 32

proffalken, 2019-01-22:
Hi,

Is it possible to get a KiCAD footprint/component for the LoRa WiFi 32?

I'm working on a PCB at the moment and it would be a lot easier to use your own schematics than trying to work out the distances between the pins for spacing on the board etc.

chevdor, 2019-02-28:
That would be nice indeed, also looking for a part for the `WIFI_LoRa_32`

EdsonMazoni, 2019-02-28:
You can see vídeos in YouTube  by Fernando k     about  ESP 32.
Att.
Edson Mazoni

Obter o Outlook para Android<https://aka.ms/ghei36>

yepher, 2019-05-28:
Anyone have a KiCad file they are willing to share?

proffalken, 2019-06-04:
@Heltec-Aaron-Lee - is this something you can provide please?

Heltec-Aaron-Lee, 2019-10-24:
OK, we will.

DeuxVis, 2019-10-24:
Actually having the footprints for all your boards would really be nice.

Xiao-H95, 2019-10-25:
Hi, this is KiCAD file share link:
https://drive.google.com/open?id=1emSYcXucConRZjR__lWBoOHJZeaCNdYu

phfbertoleti, 2020-09-08:
Hi!

Maybe it's too late, but I've recently done a Kicad 5 library and Footprint for Heltec WiFi LoRa V1 module. I thought it'd be useful for somebody who wants to make PCBs using Heltec module attached to it (as it was usefult o me), and I merged it to official Heltec repository (Aaron Lee, thanks for accepting my pull request). 

Therefore, lib and footprint are available at official Heltec module repository: https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series/tree/master/Kicad5_lib_footprint/Heltec_wifi_LoRa_V1

Enjoy!

mvadu, 2026-08-18:
@Heltec-Aaron-Lee  i don't see this any more. Is there an official place for kicad footprints to cover latest models?