Heltec-Aaron-Lee/WiFi_Kit_series #86: Lora/Wifi Board deficiency's,  V3 wish list

kampto, 2019-01-31:
This esp32 wifi Lora board  V1/V2 has some deficiency's in my opinion...  Please consider for V3 board.

1. The LiPo battery connector is on the bottom so you have to remove the board from your project to change it or power down the board. For next version please put battery connector on top of board and make it a JST 2.0mm pitch so its compatible with most other boards / Battery's.  

2. The 3.3V regulator EN pin is not broken out so you cant shut down the board. This is particularly useful when using a voltage detect circuit to prevent the board from trying to run at battery input voltages <3.3V which can cause lock ups.  For next version break out the EN regulator pin. Tie regulator EN pin to input pin with a 100K res so its pulled up but user can still pull low by grounding EN pin.

3.  No way to read Battery voltage in the sketch at all times.  Recommend have LiPo battery voltage go into a 100K/100K res voltage divider and use a hidden out Analog in pin which can be accessed by the sketch.  Or at the very least have LiPo battery on its on board pin, VBat. 

4. Increase the LiPo charge rate from 100mA to 200mA by changing the program resistor to a 5K.  This is a WiFi/RF board so it uses power, 100ma is to slow.

5. Print the pin labels on top of the board so they can be seen. I know some will be obscured by OLED but its better than nothing.

6. Is it possible for this board to run very low current in deep sleep on LiPo Bat?  Haven't found any success of that yet.  Wheres the current draw? 

7. Not likely to happen but would be great if the RF connector was changed to SMA type connector.

ChristianSasso, 2019-02-22:
2. Pin 13 has connection to the battery, the only problem is that its a ADC2 channel, so it doesn't work while using wifi functions. Battery measurement pin should be at a ADC1 channel.

4. Very important and easy fix.

6. Maybe I am missing something, but on my tests I got as low as 32mA with deep sleep in V2, what is still a very high current. One possible solution would be do some kind of power loop, BTN_0 enable power from the battery using a mosfet to the board, when ESP32 powers on it should activate another mosfet to keep power when the BTN_0 is released. BTN_0 can still be used as a Digital input.

[image]

rayperkins, 2019-04-17:
I'd be keen to see a Schottky added to isolate the CP2102 when powering the circuit from a battery. That way the CP2102 quiescent current doesn't add to the battery drain.

cyberman54, 2019-09-08:
Battery measurement is connected to GPIO13. Thus battery monitor is forced to use ADC2 instead of ADC1. But ADC2 is only usable while ESP32 Wifi driver was not started. Better connect battery pin to an ADC1 pin.