Heltec-Aaron-Lee/WiFi_Kit_series #89: can not compile LoRa examples: 'class Heltec+ESP32 has no member named LoRa'

alongcameastranger, 2019-02-06:
I am using Arduino 1.8.8 and with Heltec ESP32 Dev-Boards library installed from Arduino library manager.
I have Heltec Wifi LoRa V2 board.  The LoRa examples (from File-Examples-Heltec ESP32 Dev-Boards-LoRa) fail to compile and give an error saying Heltec_ESP32 has no member named LoRa.  I can see LoRa in the LoRa.cpp file in a subdirectory of the library. 
Here are some things I tried:
I can get it to compile if I #include "lora/LoRa.h" , change all Heltec.LoRa.* in sketch to LoRa.* , and comment out the lines void Heltec_ESP32::VextON and VextOFF functions in heltec.cpp, but it then dumps the core and reboots continuously when flashed on device, so this doesn't work.  I also tried adding #define WIFI_LoRa_32_V2 and some other things but nothing has been successful.  

Please advise how I can run the LoRa examples in Heltec library.

cypher2001, 2019-02-08:
I have the same issue. Would love some input.

cypher2001, 2019-02-08:
Seems the Heltec.h has some strange IF statements:
#if defined( WIFI_LoRa_32 ) || defined( WIFI_LoRa_32_V2 ) || defined( Wireless_Stick )
I pulled those out and then it pulled in lora.h as well as defined it in the class.  Not sure where those would be defined but it seemed to be the issue.

Boilevin, 2019-02-10:
Exactly the same issue for me.
Do you know how to solve this ?

cypher2001, 2019-02-10:
I actually used different code:

`
/*******************************************************************************
 * Copyright (c) 2015 Thomas Telkamp and Matthijs Kooijman
 * adapted by Curlywurly to use LCD
 * Permission is hereby granted, free of charge, to anyone
 * obtaining a copy of this document and accompanying files,
 * to do whatever they want with them without any restriction,
 * including, but not limited to, copying, modification and redistribution.
 * NO WARRANTY OF ANY KIND IS PROVIDED.
 *
 * This example sends a valid LoRaWAN packet with payload "Hello,
 * world!", using frequency and encryption settings matching those of
 * the The Things Network.
 *
 * This uses ABP (Activation-by-personalisation), where a DevAddr and
 * Session keys are preconfigured (unlike OTAA, where a DevEUI and
 * application key is configured, while the DevAddr and session keys are
 * assigned/generated in the over-the-air-activation procedure).
 *
 * Note: LoRaWAN per sub-band duty-cycle limitation is enforced (1% in
 * g1, 0.1% in g2), but not the TTN fair usage policy (which is probably
 * violated by this sketch when left running for longer)!
 *
 * To use this sketch, first register your application and device with
 * the things network, to set or generate a DevAddr, NwkSKey and
 * AppSKey. Each device should have their own unique values for these
 * fields.
 *
 * Do not forget to define the radio type correctly in config.h.
 *
 *******************************************************************************/

#include <lmic.h>
#include <hal/hal.h>
#include <SPI.h>
#include <U8x8lib.h>

#define BUILTIN_LED 25

// the OLED used
U8X8_SSD1306_128X64_NONAME_SW_I2C u8x8(/* clock=*/ 15, /* data=*/ 4, /* reset=*/ 16);

// LoRaWAN NwkSKey, network session key
// This is the default Semtech key, which is used by the early prototype TTN
// network.
// LoRaWAN NwkSKey, network session key
static const PROGMEM u1_t NWKSKEY[16] = { 0xAA, etc };

// LoRaWAN AppSKey, application session key
static const u1_t PROGMEM APPSKEY[16] = { 0xAA, etc};

static const u4_t DEVADDR = 0xXXXXXXXX ; // <-- Change this address for every node!

// These callbacks are only used in over-the-air activation, so they are
// left empty here (we cannot leave them out completely unless
// DISABLE_JOIN is set in config.h, otherwise the linker will complain).
void os_getArtEui (u1_t* buf) { }
void os_getDevEui (u1_t* buf) { }
void os_getDevKey (u1_t* buf) { }

static uint8_t mydata[] = "Hi!";
static osjob_t sendjob;

// Schedule TX every this many seconds (might become longer due to duty
// cycle limitations).
const unsigned TX_INTERVAL = 10;

// Pin mapping
[... 169 lines cut ...]
    // frequency is not configured here.
    #elif defined(CFG_us915)
    // NA-US channels 0-71 are configured automatically
    // but only one group of 8 should (a subband) should be active
    // TTN recommends the second sub band, 1 in a zero based count.
    // https://github.com/TheThingsNetwork/gateway-conf/blob/master/US-global_conf.json
    LMIC_selectSubBand(1);
    #endif
// Disable link check validation
    LMIC_setLinkCheckMode(0);
// TTN uses SF9 for its RX2 window.
    LMIC.dn2Dr = DR_SF9;
// Set data rate and transmit power for uplink (note: txpow seems to be ignored by the library)
    LMIC_setDrTxpow(DR_SF7,14);
// Start job
    do_send(&sendjob);
    pinMode(BUILTIN_LED, OUTPUT);
    digitalWrite(BUILTIN_LED, LOW);

}

void loop() {
    os_runloop_once();}

`

alongcameastranger, 2019-02-11:
I still hope for comment on this thread maybe from a Heltec rep....but....

I sidestepped this problem by and I got this board (and LoRa functions) to work, basically by doing the pin definitions and using Lora library that this guy does here:
https://robotzero.one/heltec-wifi-lora-32/

affordabletechnology, 2019-02-12:
After purchasing a pair of these LoRa boards, I found one had a dud OLED display, only displaying every other line, but only when you flex the OLDE cable, otherwise it displays **nothing**!

Then I found I can't compile any of their demo's, as I have the identical problem as detailed here.

I also purchased a Heltec WiFi Kit but cannot compile their demos.  I get the same errors as reported by others, and just like here, no response from Heltec (even after Chinese New Year).

I then wasted even more hours, navigating their web site, hoping to d/load either a revised H/W spec or updated code, sadly their site requires an a/c and login, but  demands you have a **Chinese mobile phone** number.  Why the hell do they sell globally if they will only provide support to Chinese residents?

Thankfully, I paid top dollar purchasing through Amazon, so I'm bundling up the lot and returning them and will purchase TTGO modules, which I have since discovered have good comments from users.

Boilevin, 2019-02-13:
Certainly same result for me.
It's not normal that their Demo Don't compile and no response

Heltec-Aaron-Lee, 2019-02-19:
Hi there, here is a document about how to use this frameworks and the newest Heltec-ESP32 library.
[http://docs.heltec.cn/#/en/user_manual/how_to_install_esp32_Arduino](http://docs.heltec.cn/#/en/user_manual/how_to_install_esp32_Arduino)

closed 2019-02-19 by Heltec-Aaron-Lee

jribyat, 2019-08-15:
You'll need to update your communications drivers for the esp32 LoRa boards. You can download the update driver from : https://www.silabs.com/products/development-tools/software/usb-to-uart-bridge-vcp-drivers

TheEarl1, 2020-03-05:
Make sure you have the correct board selected. I was getting this until I changed from the WIFI board to the LORA V2 board under 'Tools -> Board'
You may need to install the Heltec board manager to see the LORA boards

Heltec-Aaron-Lee, 2020-03-05:
Comment `#include "LoRa.h"` in the ino file, because this file already included in the `heltec.h`

howyhowy, 2020-05-22:
I had this error after my board was working and one day it would not compile. I discovered I had two Arduion IDEs installed and I was choosing the old one that did not have the proper board library installed.