Heltec-Aaron-Lee/WiFi_Kit_series #90: Reading battery level on A4 during loop

troyroberson, 2019-02-09:
I can read the battery just fine during the setup() routine, but it always returns zero after it is read ever minute during a loop. Am I missing something?  

Here's the function I'm calling from loop()

`void getBatteryLevel()
{
  measuredvbat = analogRead(VBATPIN);
  //delay(1000);
  measuredvbat *= 2;    // we divided by 2, so multiply back
  measuredvbat *= 3.3;  // Multiply by 3.3V, our reference voltage
  measuredvbat /= 1024; // convert to voltage
}`

ChristianSasso, 2019-02-22:
Are you using wifi?
The battery voltage pin (13) is on ADC2 that has shared functions with wifi on ESP32. Since wifi has higher priority, the ADC2 doesn't work while wifi is active.

leifle, 2020-01-30:
I have the same issue. Having loads of delays and multiple readings behind each other does not change a thing. I think Christian Sasso points in the right direction, that it can not be read while the Wifi is on.

It works perfect if I have no Wifi and the thing on USB. But then I really don't need it.

Is there any trick how one can get a reading of the battery voltage while Wifi is on. Or do I have to turn Wifi off, read the voltage and then turn it back on?

Thank you very much in advance...