Heltec-Aaron-Lee/WiFi_Kit_series #92: Libraries LoRa

Samuel-ZDM, 2019-03-27:
Hi. Why does the repository no longer have Lora files?

JBSchueler, 2019-05-04:
@Heltec-Aaron-Lee : Could you please comment on this topic?

After **Auto synchronization to espressif library** 120dec9e5f679801999ab1fbe04ea928f62c1175 on January 15 2019 a lot of libraries and examples have been lost.
Like LoRa, OLED, DHT11, etc.

Is there a reason why those are removed from the repository?

Heltec-Aaron-Lee, 2019-05-06:
Hi there

I had moved all examples about our board to this repository: 
[https://github.com/HelTecAutomation/Heltec_ESP32](https://github.com/HelTecAutomation/Heltec_ESP32)

Because we made a auto sync script in our local server, it made this WiFi_Kit_series auto sync to [https://github.com/espressif/arduino-esp32](https://github.com/espressif/arduino-esp32)

JBSchueler, 2019-05-06:
Nice!
Wasn't aware of this.

Added `https://raw.githubusercontent.com/HelTecAutomation/Heltec_ESP32/master/library.json` to Arduino Board Managers URL's and added the library via library manager.

[image]

[image]

Heltec-Aaron-Lee, 2019-05-06:
@JBSchueler We recommend use this url: [https://docs.heltec.cn/download/package_heltec_esp32_index.json](https://docs.heltec.cn/download/package_heltec_esp32_index.json)

Some document here:
[https://docs.heltec.cn/#/en/user_manual/how_to_install_esp32_Arduino](https://docs.heltec.cn/#/en/user_manual/how_to_install_esp32_Arduino)

JBSchueler, 2019-05-06:
Hi Aaron-Lee,

Thank you very much, very appreciated!

Samuel-ZDM, 2019-05-06:
Thank you very much , @JBSchueler and @Heltec-Aaron-Lee. 

I didn't find this repository.

Hug

Heltec-Aaron-Lee, 2019-07-22:
@Samuel-ZDM Thar's not a repository, we just upload the .json file to our server.