Heltec-Aaron-Lee/WiFi_Kit_series #94: heltec.h:29:5: error: 'SSD1306Wire' does not name a type      SSD1306Wire *display;

Jaxal83, 2019-04-04:
When I try to compile the wifi_kit_8_factorytest example from the heltec examples, it gives me this error. I tried removing the library, restarting the Arduino IDE and reinstalling the library. I am still getting this issue. I am running this on Ubuntu 18.04.2 LTS. I can upload other sketches and everything works just fine on my board, but I can't use the oled. Any help would be most welcome. Thank you.

dshess, 2019-04-12:
Are you sure you have "WiFi_Kit_8" selected as your board?  I was getting this exact error message 15 minutes ago, because I had "Generic 8266 Module" selected.

Jaxal83, 2019-04-15:
Wow. I couldn't find a "WiFi_Kit_8" board in the list, so I selected the nodemcu board. With the nodemcu board selected you can use a different library for the oled and get it to work. Now that you've pointed out that board exists I was able to find it and get it to work with the heltec library. Thank you.

closed 2019-04-15 by Jaxal83