Heltec-Aaron-Lee/WiFi_Kit_series #97: I can't seem to get the SD sketches working on a Heltec Wifi 32.

asalaices, 2019-06-03:
The Esp32 Board shown, shares all the pins with the Heltec Wifi 32, yet I cannot get the same SD Card to work on it. 
Is there any magic needed on Software ? Perhaps SPI.begin ?

asalaices, 2019-06-03:
Its worthy to note that I was able to get the Same SD Card Board working with mySD.h and the stock sketch.
All this while assigning the same pins that I know the System is using !! 
How weird is this . I need to get this working and I am stumped !. Help would be greately Appreciated.

Heltec-Aaron-Lee, 2019-06-29:
@asalaices What pins were you used for the SD card? If you didn't use the default SPI pins. `SPI.begin(SCK,MISO,MOSI,SS);` is needed.

asalaices, 2019-06-30:
I got it working by using the suggested command. Thanks a bunch !

Heltec-Aaron-Lee, 2019-07-04:
Congratulations!!! 👍 

Now can I make this issue closed?

Heltec-Aaron-Lee, 2019-07-22:
Please try this example:
[https://github.com/HelTecAutomation/Heltec_ESP32/blob/master/examples/SD/SD_Time/SD_Time.ino](https://github.com/HelTecAutomation/Heltec_ESP32/blob/master/examples/SD/SD_Time/SD_Time.ino)

closed 2019-07-22 by Heltec-Aaron-Lee