Heltec-Aaron-Lee/WiFi_Kit_series #99: LoRa sender/receiver examples behaving differently on a two boards

bicubic, 2019-06-29:
Hi, I have 2x Wifi LoRa 32 V2 868 mhz boards. 

I noticed that if I flash one (lets call it A) with the LoRa sender example and the other (call it B) with the LoRa receiver example, then things work fine. 

However if I flash the sender example onto B and receiver example onto A, the receiver only picks up one packet and then nothing. The mitigation seems to be to  call `LoRa.receive();` continuously in the main loop of the receiver. 

DumpRegisters showed a difference in 0x53 and 0x5C only between these boards. I haven't found any documentation for what those registers are. 

Any idea of why supposedly identical boards are behaving differently with the same example flashed on them?

Heltec-Aaron-Lee, 2019-06-29:
[https://github.com/HelTecAutomation/Heltec_ESP32/tree/master/examples/Factory_Test/WiFi_LoRa_32FactoryTest](https://github.com/HelTecAutomation/Heltec_ESP32/tree/master/examples/Factory_Test/WiFi_LoRa_32FactoryTest)

Can you try this factory test example? Download the same code in both board.

bicubic, 2019-06-29:
It sort of works? Both are sending and showing an increasing count, and show received data infrequently despite being next to each other and indicating RSSI -15.

Heltec-Aaron-Lee, 2019-06-29:
This code is for the basic hardware test. have both send and receive to let me know the hardware status.

Can you try board A flash receiver and board B flash receiver?

bicubic, 2019-06-29:
I don't have a third transmitter board to test A/B if both are receivers. The only way I've been able to narrow down the issue so far is swapping the images between the boards.

Heltec-Aaron-Lee, 2019-06-29:
Yes, I mean you can try to change the sender and receiver in your two boards.

bicubic, 2019-06-29:
Closing this issue for now. I've gotten consistent behaviour from both boards using interrupt based receive handler rather than polling. If I get a third device to test with, I will reopen.

closed 2019-06-29 by bicubic