Heltec-Aaron-Lee/WiFi_Kit_series #100: Failed to connect to ESP32: Timed out waiting for packet header - When in-circuit

psgcooldog, 2019-07-13:
I can program the WiFi Kit 32 just fine when I remove it from the board I'm developing, but when it's plugged-in, I cannot connect to it, even if I use the PROG and RST buttons.

What pins should I avoid using, or what conditions do I need to meet to be *certain* that I can program the Wifi Kit 32 when it's eventually soldered into a board?

Heltec-Aaron-Lee, 2019-07-22:
Hi, @psgcooldog 
Did you try to refer to this user manual?
[https://docs.heltec.cn/#/en/faq/heltec_esp_series_common_problem_summary?id=cant-download-firmware](https://docs.heltec.cn/#/en/faq/heltec_esp_series_common_problem_summary?id=cant-download-firmware)

psgcooldog, 2019-07-22:
I made some changes, and will have a new prototype PCB in a few days.  I'll come back and update this after I have a chance to assemble and test it.

psgcooldog, 2019-07-26:
The new prototype is behaving much better.

closed 2019-07-26 by psgcooldog