Heltec-Aaron-Lee/WiFi_Kit_series #105: Wifi Stick Lite - Doesn't like compost heaps

Skipper-is, 2019-10-14:
Not really an issue as such, but an observation -
I'm running a Heltec Wifi Stick Lite, along with a solar power management board (DFRobot) as a temperature probe in a compost heap. There has been a fungus tendril growing up inside the case, and - although I cannot see it, into the Heltec module. It is no longer registering as a device on my computer... 
So - turns out, electronics, damp and fungi... don't get along. I could've probably guessed that before really.

kilnageer2, 2019-10-15:
Good man.Skipper-is. I'm using Heltec Lora boards to monitor my hot compost heaps as well :)

Skipper-is, 2019-10-24:
Ah awesome! What temp sensor you using? I've got a DS18B20 (the waterproofed version) on the end of a long stick! I think I should daisy chain a whole load together for a better picture of the heap.

Just received my 4 CubeCell boards! Looking forward to testing them! Just waiting on the arduino board manager to finish downloading on my terrible internet... Maybe this evening it'll done!