Heltec-Aaron-Lee/WiFi_Kit_series #107: WiFi Kit 32 - Issues with multiple inputs

proffalken, 2019-10-22:
Hello,

When I run the following code, then no matter which switch I throw (the `INPUT`s), It seems to pick a random number of servos to change the position of.

Is this a timing thing, or have I missed something important?

```
#include <ESP32Servo.h>

#include <ArduinoJson.h>
#include <U8g2lib.h>

// Setup the OLED
U8G2_SSD1306_128X64_NONAME_F_HW_I2C   u8g2(U8G2_R0, 16, 15, 4);

// Setup the servos
// Top Right
Servo svotr;
// Top Left
Servo svotl;
// Bottom Right
Servo svobr;
// Bottom Left
Servo svobl;

// Setup the states
char trstate[] = "unlocked";
char tlstate[] = "unlocked";
char brstate[] = "unlocked";
char blstate[] = "unlocked";

void setup() {
  Serial.begin(115200);
  // Setup the OLED
  // Initialize the graphics library.
  u8g2.begin();
  u8g2.setFont(u8g2_font_6x10_tf);
  u8g2.setFontRefHeightExtendedText();
  u8g2.setDrawColor(1);
  u8g2.setFontPosTop();
  u8g2.setFontDirection(0);
  // Set up the buttons
  pinMode(34, INPUT); // Top right
  pinMode(38, INPUT); // Top Left
  pinMode(37, INPUT); // Bottom Right
  pinMode(36, INPUT); // Bottom Left
  // Set up the servos
  svotr.attach(12);
  svotl.attach(13);
  svobr.attach(14);
  svobl.attach(2);
  svotr.write(0);
  svotl.write(0);
  svobr.write(0);
  svobl.write(0);
}

void loop() {
  u8g2.clearBuffer();
  if(digitalRead(34) == LOW){
    svotr.write(0);
    strcpy(trstate, "locked");
  } else {
    svotr.write(90);
    strcpy(trstate, "unlocked");
  }
  if(digitalRead(38) == LOW){
    svotl.write(0);
    strcpy(tlstate, "locked");
  } else {
    svotl.write(90);
    strcpy(tlstate, "unlocked");
  }
  if(digitalRead(37) == LOW){
    svobr.write(0);
    strcpy(brstate, "locked");
  } else {
    svobr.write(90);
    strcpy(brstate, "unlocked");
  }
  if(digitalRead(36) == LOW){
    svobl.write(0);
    strcpy(blstate ,"locked");
  } else {
    svobl.write(90);
    strcpy(blstate, "unlocked");
  }
  const size_t capacity = JSON_OBJECT_SIZE(4) + 90;
DynamicJsonDocument doc(capacity);

doc["top_right"] = trstate;
doc["top_left"] = tlstate;
doc["bottom_right"] = brstate;
doc["bottom_left"] = blstate;

serializeJsonPretty(doc, Serial);

// Print the output
u8g2.drawStr(0, 0, "TR:");
u8g2.drawStr(25, 0, trstate);
u8g2.drawStr(0, 10, "TL:");
u8g2.drawStr(25, 10, tlstate);
u8g2.drawStr(0, 20, "BR:");
u8g2.drawStr(25, 20, brstate);
u8g2.drawStr(0, 30, "BL:");
u8g2.drawStr(25, 30, blstate);

// Send the display buffer to the oled.
  
u8g2.sendBuffer();

delay(1000);
}```

DeuxVis, 2019-10-24:
Have you wired pullup resitors to your input pins ? If not you should use 

`pinMode( pinX, INPUT_PULLUP );`

or

`pinMode( pinX, INPUT );
digitalWrite( pinX, HIGH );
`

(Not sure which syntax is supported with esp)

proffalken, 2019-10-24:
I've got resistors in place, but I think they're "pull down" instead.

I'll look at switching them to "pull up", but my main concern is that whilst I've avoided the pins for the OLED etc, I'm aware that multiple pins have multiple "roles" and I'm wondering if I've "clashed" somewhere.