Heltec-Aaron-Lee/WiFi_Kit_series #111: WiFi Kit 8 package description has a garbage

Hieromon, 2019-11-13:
The Arduino package description file for WiFi Kit 8 contains garbage at line: 44.
Please clean up the garbage.

https://docs.heltec.cn/download/package_heltec_esp8266_index.json
```json
{
  "packages": [
     {
      "maintainer": "Heltec Automation(TM)", 
      "help": {
        "online": "http://heltec.cn"
      }, 
      "websiteURL": "https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series", 
      "platforms": [
        {
          "category": "ESP8266", 
          "help": {
            "online": "http://esp8266.github.io/Arduino/versions/2.0.0/"
          }, 
          "url": "https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series/releases/download/0.0.4-rc1/esp8266-0.0.4.zip", 
          "checksum": "SHA-256:00e4a8c6aeee97b7a037831e22de0531103a63dc79544e342822f8e9d99eea73", 
          "name": "Heltec ESP8266 Series Dev-boards", 
          "version": "0.0.4-rc1", 
          "architecture": "esp8266", 
          "archiveFileName": "esp8266-0.0.4.zip", 
          "toolsDependencies": [
            {
              "packager": "Heltec-esp8266", 
              "version": "2.5.0-3-20ed2b9", 
              "name": "esptool"
            }, 
            {
              "packager": "Heltec-esp8266", 
              "version": "2.5.0-3-20ed2b9", 
              "name": "xtensa-lx106-elf-gcc"
            }, 
            {
              "packager": "Heltec-esp8266", 
              "version": "2.5.0-3-20ed2b9", 
              "name": "mkspiffs"
            }
          ], 
          "boards": [
            {
              "name": "WiFi Kit 8"
            }
          ],
          "size": "30059628"
        }��
    {
      "maintainer": "Heltec Automation(TM)", 
      "help": {
        "online": "http://heltec.cn"
      }, 
      "websiteURL": "https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series", 
      "platforms": [
        {
          "category": "ESP8266", 
          "help": {
            "online": "http://esp8266.github.io/Arduino/versions/2.0.0/"
          }, 
          "url": "https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series/releases/download/0.0.1/esp8266-0.0.1.zip", 
          "checksum": "SHA-256:840cbc13531881e8e9c4a645b433a23794ffc941962c1259a02d92899e5efe4c", 
          "name": "Heltec ESP8266 Series Dev-boards", 
          "version": "0.0.1", 
          "architecture": "esp8266", 
          "archiveFileName": "esp8266-0.0.1.zip", 
          "toolsDependencies": [
            {
              "packager": "Heltec-esp8266", 
[... 170 lines cut ...]
              "size": "368554"
            }, 
            {
              "url": "https://github.com/earlephilhower/esp-quick-toolchain/releases/download/2.5.0-3/x86_64-linux-gnu.mkspiffs-7fefeac.tar.gz", 
              "checksum": "SHA-256:4547bfe668090944d50af8dbf09abb5b51eb4a892b4784818f06a2274a630878", 
              "host": "x86_64-pc-linux-gnu", 
              "archiveFileName": "x86_64-linux-gnu.mkspiffs-7fefeac.tar.gz", 
              "size": "52429"
            }, 
            {
              "url": "https://github.com/earlephilhower/esp-quick-toolchain/releases/download/2.5.0-3/x86_64-w64-mingw32.mkspiffs-7fefeac.zip", 
              "checksum": "SHA-256:0dedc3ee977af99ffd160bb304e8bfb68a8545f2d1f2e4a3b534f2ae1a49d2c9", 
              "host": "x86_64-mingw32", 
              "archiveFileName": "x86_64-w64-mingw32.mkspiffs-7fefeac.zip", 
              "size": "350035"
            }
          ]
        }  
      ], 
      "email": "ivan@esp8266.com", 
      "name": "Heltec-esp8266"
    }
  ]
}
```

MartinNohr, 2019-11-15:
I also have noticed this problem, please fix it somebody!

Momo456, 2019-12-01:
Same issue here ....

decobeto, 2019-12-11:
I also have this problem, if someone could fix it!

andyjdow, 2019-12-14:
Ditto. I think there are several other issues:
- I think the file structure in esp8266-0.0.4.zip is incorrect. I get an invalid package error from Arduino. I managed to work around it by correcting the file structure.
- Then when I tried to compile for the Kit 8 board I got:
   exec: "{runtime.tools.python.path}/python": file does not exist
I eventually got round this by copying the python tool directory from the main esp8266.
- Compilation then completes but I get the error during upload. I think it might be related to https://github.com/esp8266/Arduino/issues/5881 and I tried various mods to the package.txt but didn't get anywhere.
So I'm now stuck and can't really use the board. :-( Please fix it!

Hieromon, 2020-01-17:
Apparently, the problem seems to leave unattended.
It seems sensible to work around this problem by using nodeMCU for now.

Heltec-Aaron-Lee, 2020-01-19:
Hi there

please try thi url, it's working now.
[https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series/releases/download/0.0.4/package_heltec_esp8266_index.json](https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series/releases/download/0.0.4/package_heltec_esp8266_index.json)

Flash1705, 2020-01-27:
Hi there

I have problems when using the "0.0.4" release.
See errors messages below.

Arduino: 1.8.10 (Windows 7), Board: "WiFi Kit 8, 80 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), 4MB (FS:2MB OTA:~1019KB), v2 Lower Memory, Disabled, None, Only Sketch, 921600"

Index error: could not find referenced tool name=esptool version=2.5.0-3-20ed2b9 packager=Heltec-esp8266
Index error: could not find referenced tool name=xtensa-lx106-elf-gcc version=2.5.0-3-20ed2b9 packager=Heltec-esp8266
Index error: could not find referenced tool name=mkspiffs version=2.5.0-3-20ed2b9 packager=Heltec-esp8266
Index error: could not find referenced tool name=esptool version=2.5.0-3-20ed2b9 packager=Heltec-esp8266
Index error: could not find referenced tool name=xtensa-lx106-elf-gcc version=2.5.0-3-20ed2b9 packager=Heltec-esp8266
Index error: could not find referenced tool name=mkspiffs version=2.5.0-3-20ed2b9 packager=Heltec-esp8266

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Error resolving FQBN: board

Error compiling for board WiFi Kit 8.

Heltec-Aaron-Lee, 2020-01-27:
@Flash1705 Which example did you use?

Flash1705, 2020-01-27:
I get these error messages when I load the board library.
I have not even tried an example.

Flash1705, 2020-01-28:
I get these error messages when I load the board library.
I have not even tried an example.

-- 

*George Durant **(SACIA Cert BCT Prof)*
*Red Pepper Pictures / MediaCloud Networks | Head Engineer*
Work: 011 446 7006 | Fax: 011 888 8212 | Cell: 083 390 3363
E-Mail: george@redpepper.co.za

*WWW.REDPEPPER.TV <http://WWW.REDPEPPER.TV>*

*58 4th AVENUE, LINDEN, JOHANNESBURG, SOUTH AFRICA*

-- 

Attention: Any views 
expressed in this message are those of the individual sender, except where 
the sender specifically states them to be the view of Red Pepper Pictures 
(Pty) Ltd, its subsidiaries and associates.  NOTICE: - This message 
contains privileged and confidential information intended only for use by 
the addressee named above. If you are not the intended recipient of this 
message, you are hereby notified that you must not disseminate, copy, take 
action in reliance on, or use it in any manner whatsoever. If you have 
received this message in error, please notify Red Pepper Pictures (Pty) 
Ltd, its subsidiaries and associates, immediately, and delete the e-mail. 
Red Pepper Pictures (Pty) Ltd does not accept liability for any damage, 
loss or prejudice arising from this e-mail and/or from the accessing of any 
files that may be attached to this e-mail

Heltec-Aaron-Lee, 2020-01-28:
Can you please upload some picture here? I can' reappear this error at my side.

Flash1705, 2020-01-28:
Hi
See attached images.
I suspect it might be that I already have the ESP8266 board
definition loaded, as I have been working with the Wemos devices.

"http://arduino.esp8266.com/stable/package_esp8266com_index.json"

I would like to work with both board definitions loaded, without having to
delete and reload every time.

Many Thanks
George Durant

-- 

*George Durant **(SACIA Cert BCT Prof)*
*Red Pepper Pictures / MediaCloud Networks | Head Engineer*
Work: 011 446 7006 | Fax: 011 888 8212 | Cell: 083 390 3363
E-Mail: george@redpepper.co.za

*WWW.REDPEPPER.TV <http://WWW.REDPEPPER.TV>*

*58 4th AVENUE, LINDEN, JOHANNESBURG, SOUTH AFRICA*

-- 

Attention: Any views 
expressed in this message are those of the individual sender, except where 
the sender specifically states them to be the view of Red Pepper Pictures 
(Pty) Ltd, its subsidiaries and associates.  NOTICE: - This message 
contains privileged and confidential information intended only for use by 
the addressee named above. If you are not the intended recipient of this 
message, you are hereby notified that you must not disseminate, copy, take 
action in reliance on, or use it in any manner whatsoever. If you have 
received this message in error, please notify Red Pepper Pictures (Pty) 
Ltd, its subsidiaries and associates, immediately, and delete the e-mail. 
Red Pepper Pictures (Pty) Ltd does not accept liability for any damage, 
loss or prejudice arising from this e-mail and/or from the accessing of any 
files that may be attached to this e-mail

Flash1705, 2020-01-28:
[image]
[image]
[image]
[image]
[image]
[image]

Flash1705, 2020-01-28:
Hi
I deleted my existing C:\Users\George\AppData\Local\Arduino15\packages\esp8266 folder then I could install and run the WIFI kit 8 and compile without any issues.
But this limits the flexibility as I am working with both boards and I then need to change the definitions every time I swap boards.
My solution in the mean time is to use the standard esp8266 libraries and select nodeMCU.

Heltec-Aaron-Lee, 2020-02-01:
@Flash1705 please delete `C:\Users\user name\AppData\Local\Arduino15\packages\esp8266` file and try again.

bkw777, 2020-09-11:
Wow, the first report is from almost a year ago, and today I get the same error. All I did was install the board and libraries following the directions right from here: https://heltec.org/wifi_kit_install/  (board manager not git)

And I get these warnings just from opening board manager or libaries manager.
Index error: could not find referenced tool name=esptool version=2.5.0-3-20ed2b9 packager=Heltec-esp8266
Index error: could not find referenced tool name=xtensa-lx106-elf-gcc version=2.5.0-3-20ed2b9 packager=Heltec-esp8266
Index error: could not find referenced tool name=mkspiffs version=2.5.0-3-20ed2b9 packager=Heltec-esp8266
Index error: could not find referenced tool name=esptool version=2.5.0-3-20ed2b9 packager=Heltec-esp8266
Index error: could not find referenced tool name=xtensa-lx106-elf-gcc version=2.5.0-3-20ed2b9 packager=Heltec-esp8266
Index error: could not find referenced tool name=mkspiffs version=2.5.0-3-20ed2b9 packager=Heltec-esp8266

I mean, I'm only following the directions on the heltec web site to the letter. The directions are pretty short and simple, not much opportunity to get anything wrong.

I'm on ubuntu not Windows.

Heltec-Aaron-Lee, 2020-09-11:
Please try this url: [https://resource.heltec.cn/download/package_heltec_esp8266_index.json](https://resource.heltec.cn/download/package_heltec_esp8266_index.json)

bkw777, 2020-09-11:
I cleaned out all references to the prior install, meaning I removed the previous heltec-esp8266 url from Preferences and manually deleted the directories:
 ~/Arduino/libraries/Heltec_ESP8266_Dev-Boards   (library)
 ~/.arduino15/packages/heltec-esp8266/    (board)

After that, no more errors, and no hint of any heltec 8266 support in the menus.
So, clean slate.

Then I added that url in Preferences and re-installed the library and board support.
https://resource.heltec.cn/download/package_heltec_esp8266_index.json

Same errors again:

Index error: could not find referenced tool name=esptool version=2.5.0-3-20ed2b9 packager=Heltec-esp8266
Index error: could not find referenced tool name=xtensa-lx106-elf-gcc version=2.5.0-3-20ed2b9 packager=Heltec-esp8266
Index error: could not find referenced tool name=mkspiffs version=2.5.0-3-20ed2b9 packager=Heltec-esp8266
Index error: could not find referenced tool name=esptool version=2.5.0-3-20ed2b9 packager=Heltec-esp8266
Index error: could not find referenced tool name=xtensa-lx106-elf-gcc version=2.5.0-3-20ed2b9 packager=Heltec-esp8266
Index error: could not find referenced tool name=mkspiffs version=2.5.0-3-20ed2b9 packager=Heltec-esp8266

Also during install for the board support (not libraries) I saw:
Warning: file /home/bkw/.arduino15/packages/heltec-esp8266/tools/python3/3.7.2-post1/python3 links to an absolute path /usr/bin/python3

Not great but not the worst crime in the world. Please excuse a small digression into esp32 even though this is about esp8266, I also have the esp32 support installed for Heltec wifi-kit-32 and while using that I was getting "python not found in $PATH" which is fixed (on Ubuntu) by "apt install python-is-python3"

Both of theses, esp8266 and esp32 support packages should just have install directions that include the requirement to install python3, and for Ubuntu specifically it would need "python-is-python3" which installs a symlink so that running "python" gets python3 instead of python2, (or the code should ask for python3 instead of python), not assume absolute path, and not omit any mention in the directions. But that's a smaller thing and more or less fixable by the user.

I have no idea what to do about these "could not find referenced tool" errors.

bkw777, 2020-09-11:
Compiling a sketch actually works.

So maybe it's just cosmetic and we can just ignore the messages while in library manager or board manager?

I just compiled and flashed the WiFi_Kit_8_FactoryTest example sketch and it worked. It compiled,flashed, and is running on the board.

I know it actually updated the board because I had previously flashed Zimodem to the board using the generic esp8266 support from here:
http://arduino.esp8266.com/stable/package_esp8266com_index.json
https://github.com/bozimmerman/Zimodem
(which worked, the code doesn't support this board without changes for different pin#s and Serial hardware, but it loads and runs.)
That sketch does not have any support for the oled so the oled is blank and there are debug messages on the usb serial. This tells me I definitely replaced the factory sketch.

And from there I switched back to Tools -> Board -> Wifi-Kit-8, and built the WiFi_Kit_8_FactoryTest sketch, and now the oled screen is showing the wifi scan messages from that sketch.

The build messages show that it's loading files from the Heltec 8266 support directories, not the generic 8266 support directories:
<pre>
...
/home/bkw/.arduino15/packages/heltec-esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/xtensa-lx106-elf-ar cru /tmp/arduino_build_957247/core/core.a /tmp/arduino_build_957247/core/spiffs/spiffs_gc.cpp.o
/home/bkw/.arduino15/packages/heltec-esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/xtensa-lx106-elf-ar cru /tmp/arduino_build_957247/core/core.a /tmp/arduino_build_957247/core/spiffs/spiffs_hydrogen.cpp.o
/home/bkw/.arduino15/packages/heltec-esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/xtensa-lx106-elf-ar cru /tmp/arduino_build_957247/core/core.a /tmp/arduino_build_957247/core/spiffs/spiffs_nucleus.cpp.o
/home/bkw/.arduino15/packages/heltec-esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/xtensa-lx106-elf-ar cru /tmp/arduino_build_957247/core/core.a /tmp/arduino_build_957247/core/spiffs_api.cpp.o
/home/bkw/.arduino15/packages/heltec-esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/xtensa-lx106-elf-ar cru /tmp/arduino_build_957247/core/core.a /tmp/arduino_build_957247/core/sqrt32.cpp.o
/home/bkw/.arduino15/packages/heltec-esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/xtensa-lx106-elf-ar cru /tmp/arduino_build_957247/core/core.a /tmp/arduino_build_957247/core/time.cpp.o
/home/bkw/.arduino15/packages/heltec-esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/xtensa-lx106-elf-ar cru /tmp/arduino_build_957247/core/core.a /tmp/arduino_build_957247/core/uart.cpp.o
/home/bkw/.arduino15/packages/heltec-esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/xtensa-lx106-elf-ar cru /tmp/arduino_build_957247/core/core.a /tmp/arduino_build_957247/core/umm_malloc/umm_malloc.cpp.o
Linking everything together...
/home/bkw/.arduino15/packages/heltec-esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/xtensa-lx106-elf-gcc -CC -E -P -DVTABLES_IN_FLASH /home/bkw/.arduino15/packages/heltec-esp8266/hardware/esp8266/0.0.4/tools/sdk/ld/eagle.app.v6.common.ld.h -o /tmp/arduino_build_957247/local.eagle.app.v6.common.ld
/home/bkw/.arduino15/packages/heltec-esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/xtensa-lx106-elf-gcc -fno-exceptions -Wl,-Map -Wl,/tmp/arduino_build_957247/WiFi_Kit_8_FactoryTest.ino.map -g -w -Os -nostdlib -Wl,--no-check-sections -u app_entry -u _printf_float -u _scanf_float -Wl,-static -L/home/bkw/.arduino15/packages/heltec-esp8266/hardware/esp8266/0.0.4/tools/sdk/lib -L/home/bkw/.arduino15/packages/heltec-esp8266/hardware/esp8266/0.0.4/tools/sdk/lib/NONOSDK22x_190703 -L/home/bkw/.arduino15/packages/heltec-esp8266/hardware/esp8266/0.0.4/tools/sdk/ld -L/home/bkw/.arduino15/packages/heltec-esp8266/hardware/esp8266/0.0.4/tools/sdk/libc/xtensa-lx106-elf/lib -Teagle.flash.4m2m.ld -Wl,--gc-sections -Wl,-wrap,system_restart_local -Wl,-wrap,spi_flash_read -o /tmp/arduino_build_957247/WiFi_Kit_8_FactoryTest.ino.elf -Wl,--start-group /tmp/arduino_build_957247/sketch/WiFi_Kit_8_FactoryTest.ino.cpp.o /tmp/arduino_build_957247/libraries/Heltec_ESP8266_Dev-Boards/heltec.cpp.o /tmp/arduino_build_957247/libraries/Heltec_ESP8266_Dev-Boards/oled/OLEDDisplay.cpp.o /tmp/arduino_build_957247/libraries/Heltec_ESP8266_Dev-Boards/oled/OLEDDisplayUi.cpp.o /tmp/arduino_build_957247/libraries/Wire/Wire.cpp.o /tmp/arduino_build_957247/libraries/ESP8266WiFi/ESP8266WiFi.a /tmp/arduino_build_957247/core/core.a -lhal -lphy -lpp -lnet80211 -llwip2-536-feat -lwpa -lcrypto -lmain -lwps -lbearssl -laxtls -lespnow -lsmartconfig -lairkiss -lwpa2 -lstdc++ -lm -lc -lgcc -Wl,--end-group -L/tmp/arduino_build_957247
/home/bkw/.arduino15/packages/heltec-esp8266/tools/python3/3.7.2-post1/python3 /home/bkw/.arduino15/packages/heltec-esp8266/hardware/esp8266/0.0.4/tools/elf2bin.py --eboot /home/bkw/.arduino15/packages/heltec-esp8266/hardware/esp8266/0.0.4/bootloaders/eboot/eboot.elf --app /tmp/arduino_build_957247/WiFi_Kit_8_FactoryTest.ino.elf --flash_mode dio --flash_freq 40 --flash_size 4M --path /home/bkw/.arduino15/packages/heltec-esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin --out /tmp/arduino_build_957247/WiFi_Kit_8_FactoryTest.ino.bin
Creating BIN file "/tmp/arduino_build_957247/WiFi_Kit_8_FactoryTest.ino.bin" using "/tmp/arduino_build_957247/WiFi_Kit_8_FactoryTest.ino.elf"
[... cut ...] 8_FactoryTest.ino.bin 
esptool.py v2.8
Serial port /dev/ttyUSB0
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: f4:cf:a2:68:eb:a4
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 280960 bytes to 204306...
Wrote 280960 bytes (204306 compressed) at 0x00000000 in 4.9 seconds (effective 462.4 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
</pre>

bkw777, 2021-02-18:
> Hello,
> Responding to your issue with the index errors:
> 
> > ....
> > Then I added that url in Preferences and re-installed the library and board support.
> > https://resource.heltec.cn/download/package_heltec_esp8266_index.json
> > Same errors again:
> > Index error: could not find referenced tool name=esptool version=2.5.0-3-20ed2b9 packager=Heltec-esp8266
> > Index error: could not find referenced tool name=xtensa-lx106-elf-gcc version=2.5.0-3-20ed2b9 packager=Heltec-esp8266
> > Index error: could not find referenced tool name=mkspiffs version=2.5.0-3-20ed2b9 packager=Heltec-esp8266
> > Index error: could not find referenced tool name=esptool version=2.5.0-3-20ed2b9 packager=Heltec-esp8266
> > Index error: could not find referenced tool name=xtensa-lx106-elf-gcc version=2.5.0-3-20ed2b9 packager=Heltec-esp8266
> > Index error: could not find referenced tool name=mkspiffs version=2.5.0-3-20ed2b9 packager=Heltec-esp8266
> > .....
> > I have no idea what to do about these "could not find referenced tool" errors.
> 
> The issue appears to be that there are packager entries are "Heltec-esp8266" with an uppercase "H", the name entry in entry at line 4 is lowercase "H". I put a test version of an edited JSON file on a website and changed the URL to point to my test JSON file and I received no errors!
> 
> Here's an example of what a fixed entry looks like:
> 
> ```
>       "toolsDependencies": [
>         {
>           "packager": "heltec-esp8266", 
>           "version": "2.5.0-3-20ed2b9", 
>           "name": "esptool"
>         }, 
> ```
> 
> Hope this helps!

I'll try it. Thanks much!

bill-orange, 2021-06-04:
Same problem here.  At least for me, the index errors have not resulted in an error during compilation.