Heltec-Aaron-Lee/WiFi_Kit_series #114: Esp8266 json doesn't work

decobeto, 2019-12-11:
I can't install documentation libraries because esp8266 json has problems

https://docs.heltec.cn/download/package_heltec_esp8266_index.json

[image]

SauloAislan, 2019-12-12:
Try this: http://arduino.esp8266.com/stable/package_esp8266com_index.json

SauloAislan, 2019-12-12:
This is a problem on line 44 has garbage:

[image]

SauloAislan, 2019-12-12:
See https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series/releases

decobeto, 2019-12-12:
I saw the problem, I tried to work around using the tutorial from this site https://robotzero.one/heltec-wifi-kit-8/, but my esp doesn't seem to work well with the libraries for esp8266

SauloAislan, 2019-12-12:
See: https://docs.heltec.cn/#/en/user_manual/how_to_install_esp8266_Arduino?id=step1-download-arduino-esp8266-support

SauloAislan, 2019-12-12:
What is the model of your board?

SauloAislan, 2019-12-12:
Resolved https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series/issues/110

decobeto, 2019-12-12:
Esp8266 heltec wifi kit 8, I reinstalled the arduino IDE and now it's working

closed 2019-12-12 by decobeto