Heltec-Aaron-Lee/WiFi_Kit_series #116: Heltec LoRa v2 not working with Adafruit Ultimate GPS Breakout v3 using hardware serial

SneezeWeeze, 2019-12-17:
I tested the GPS module with an Uno using SoftwareSerial and it worked like a charm on pins 2 & 4.

Because SoftwareSerial is not supported on the Heltec LoRa v2, I have to use an alternative. There are samples of using the GPS module including those for hardware serial and ic2.

I figured I'd be able to use Serial2 to make it work. Here is the sketch I made. But it doesn't seem to receive any data. I'm either doing something terribly wrong or there are very specific pins that should be used with hardware serial. @Heltec-Aaron-Lee Is it possible that the Heltec LoRa v2 is incompatible with this GPS module.

Here is my sketch:
```
// Test code for Adafruit GPS modules using MTK3329/MTK3339 driver
//
// This code turns on the LOCUS built-in datalogger. The datalogger
// turns off when power is lost, so you MUST turn it on every time
// you want to use it!
//
// Tested and works great with the Adafruit GPS FeatherWing
// ------> https://www.adafruit.com/products/3133
// or Flora GPS
// ------> https://www.adafruit.com/products/1059
// but also works with the shield, breakout
// ------> https://www.adafruit.com/products/1272
// ------> https://www.adafruit.com/products/746
//
// Pick one up today at the Adafruit electronics shop
// and help support open source hardware & software! -ada

#include <Adafruit_GPS.h>

// what's the name of the hardware serial port?

#define GPSSerial Serial2
// Connect to the GPS on the hardware port
Adafruit_GPS GPS(&GPSSerial);

// Set GPSECHO to 'false' to turn off echoing the GPS data to the Serial console
// Set to 'true' if you want to debug and listen to the raw GPS sentences
#define GPSECHO  true

void setup()
{
  //while (!Serial);  // uncomment to have the sketch wait until Serial is ready
  // connect at 115200 so we can read the GPS fast enough and echo without dropping chars
  // also spit it out
  GPSSerial.begin(9600, SERIAL_8N1,23,17);
  Serial.begin(115200);
  delay(1000);
  Serial.println("Adafruit GPS logging data dump!");
  // 9600 NMEA is the default baud rate for MTK - some use 4800
  GPS.begin(9600);

  GPS.sendCommand(PMTK_SET_NMEA_OUTPUT_OFF);

  while (GPSSerial.available())
     GPSSerial.read();

  delay(1000);
  GPS.sendCommand("$PMTK622,1*29");
  Serial.println("----------------------------------------------------");
}

uint32_t updateTime = 1000;

void loop()                     // run over and over again
{
  if (Serial.available()) {
    char c = Serial.read();
    GPSSerial.write(c);
  }
  if (GPSSerial.available()) {
    char c = GPSSerial.read();
    Serial.write(c);
  }
}//loop
```

SauloAislan, 2019-12-20:
Serial2 in Heltec LoRa v2 is 17 and 16.

https://www.weargenius.in/wp-content/uploads/2018/12/df.png

SneezeWeeze, 2019-12-20:
From my understanding, GPSSerial.begin(9600, SERIAL_8N1,23,17) reassigns Serial2, so7 that should not matter. 23 corresponds to RX on the GPS module. 17 corresponds to TX.  "SERIAL_8N1" symbolizes a UART address I think.

There are 4 jumper cables connecting the Heltec board to the GPS module. One is 3.3v. Another ground. The other two are RX and TX. These are the two pins I need to read on Serial2.

SneezeWeeze, 2019-12-20:
@SauloAislan I originally tried hardware pins 16 and 17 but with no success before moving on to reassign the default hardware pins for Serial2.

SauloAislan, 2019-12-22:
See: https://github.com/espressif/arduino-esp32/issues/665