Heltec-Aaron-Lee/WiFi_Kit_series #119: Constant rebooting when trying to connect to TTN: Core  1 panic'ed (StoreProhibited)

ld21, 2020-03-11:
Hi, I just received two Heltect ESP32 LORA devices: One V2 and one Wireless Stick.
I registered them successfully both at your website and registered the device at TTN. Unfortunately, as soon as I upload the following code (using Arduino IDE), the devices keep on rebooting, saying
```
11:55:08.672 -> ESP32 MCU init...
11:55:09.814 -> ESP32 MCU inited OK!
11:55:10.193 -> 
11:55:10.193 -> LoRaWAN EU868 Class A start!
11:55:10.228 -> 
11:55:10.647 -> joining...
11:55:10.647 -> Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
11:55:10.647 -> Core 1 register dump:
11:55:10.647 -> PC      : 0x400d4237  PS      : 0x00060330  A0      : 0x800d3fca  A1      : 0x3ffb1ef0  
11:55:10.683 -> A2      : 0x3ffb1f24  A3      : 0x3ffb1f1c  A4      : 0x00000031  A5      : 0x00000000  
11:55:10.683 -> A6      : 0x3ffb8f22  A7      : 0x00000006  A8      : 0x00000000  A9      : 0x00000001  
11:55:10.716 -> A10     : 0x00000000  A11     : 0x3ffc1d8c  A12     : 0x3ffc03b4  A13     : 0x00000000  
11:55:10.716 -> A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x0000001b  EXCCAUSE: 0x0000001d  
11:55:10.716 -> EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  
11:55:10.749 -> 
11:55:10.749 -> Backtrace: 0x400d4237:0x3ffb1ef0 0x400d3fc7:0x3ffb1f10 0x400d16c3:0x3ffb1f50 0x400d11db:0x3ffb1f90 0x400da3c1:0x3ffb1fb0 0x40088f4d:0x3ffb1fd0
11:55:10.786 -> 
11:55:10.786 -> Rebooting...
```

This is the code I'm using:
```
/*
 * HelTec Automation(TM) LoRaWAN 1.0.2 OTAA example use OTAA, CLASS A
 *
 * Function summary:
 *
 * - use internal RTC(150KHz);
 *
 * - Include stop mode and deep sleep mode;
 *
 * - 15S data send cycle;
 *
 * - Informations output via serial(115200);
 *
 * - Only ESP32 + LoRa series boards can use this library, need a license
 *   to make the code run(check you license here: http://www.heltec.cn/search );
 *
 * You can change some definition in "Commissioning.h" and "LoRaMac-definitions.h"
 *
 * HelTec AutoMation, Chengdu, China.
 * 成都惠利特自动化科技有限公司
 * https://heltec.org
 * support@heltec.cn
 *
 *this project also release in GitHub:
 *https://github.com/HelTecAutomation/ESP32_LoRaWAN
*/

#include <ESP32_LoRaWAN.h>
#include "Arduino.h"

/*Die Chip-ID kriegt man mit
uint64_t chipid;

void setup() {
Serial.begin(57600);
}

void loop() {
chipid=ESP.getEfuseMac();//The chip ID is essentially its MAC address(length: 6 bytes).
Serial.printf("ESP32 Chip ID = %04X",(uint16_t)(chipid>>32));//print High 2 bytes
Serial.printf("%08X\n",(uint32_t)chipid);//print Low 4bytes.

delay(3000);

}
 * */
/*license for Heltec ESP32 LoRaWan, quary your ChipID relevant license: http://resource.heltec.cn/search */
uint32_t  license[4] = {received from heltec};
/* OTAA para*/
uint8_t DevEui[] = { according to TTN }; 
uint8_t AppEui[] = { according to TTN }; 
uint8_t AppKey[] = { according to TTN }; 

/* ABP para*/
uint8_t NwkSKey[] = { 0x15, 0xb1, 0xd0, 0xef, 0xa4, 0x63, 0xdf, 0xbe, 0x3d, 0x11, 0x18, 0x1e, 0x1e, 0xc7, 0xda,0x85 };
uint8_t AppSKey[] = { 0xd7, 0x2c, 0x78, 0x75, 0x8c, 0xdc, 0xca, 0xbf, 0x55, 0xee, 0x4a, 0x77, 0x8d, 0x16, 0xef,0x67 };
uint32_t DevAddr =  ( uint32_t )0x007e6ae1;

/*LoraWan Class, Class A and Class C are supported*/
DeviceClass_t  loraWanClass = CLASS_A;

/*the application data transmission duty cycle.  value in [ms].*/
uint32_t appTxDutyCycle = 15000;

/*OTAA or ABP*/
bool overTheAirActivation = true;

/*ADR enable*/
bool loraWanAdr = true;

/* Indicates if the node is sending confirmed or unconfirmed messages */
bool isTxConfirmed = true;

/* Application port */
uint8_t appPort = 2;

/*!
* Number of trials to transmit the frame, if the LoRaMAC layer did not
* receive an acknowledgment. The MAC performs a datarate adaptation,
* according to the LoRaWAN Specification V1.0.2, chapter 18.4, according
* to the following table:
*
* Transmission nb | Data Rate
* ----------------|-----------
* 1 (first)       | DR
* 2               | DR
* 3               | max(DR-1,0)
* 4               | max(DR-1,0)
* 5               | max(DR-2,0)
* 6               | max(DR-2,0)
* 7               | max(DR-3,0)
* 8               | max(DR-3,0)
*
* Note, that if NbTrials is set to 1 or 2, the MAC will not decrease
* the datarate, in case the LoRaMAC layer did not receive an acknowledgment
*/
uint8_t confirmedNbTrials = 8;

/*LoraWan debug level, select in arduino IDE tools.
* None : print basic info.
* Freq : print Tx and Rx freq, DR info.
* Freq && DIO : print Tx and Rx freq, DR, DIO0 interrupt and DIO1 interrupt info.
* Freq && DIO && PW: print Tx and Rx freq, DR, DIO0 interrupt, DIO1 interrupt, MCU sleep and MCU wake info.
*/
//uint8_t debugLevel = LoRaWAN_DEBUG_LEVEL;
uint8_t debugLevel = LL_ALL;

/*LoraWan region, select in arduino IDE tools*/
//LoRaMacRegion_t loraWanRegion = ACTIVE_REGION; // Ursprünglich stand das hier (Dani Süsstrunk, 11.3.2020)
LoRaMacRegion_t loraWanRegion = LORAMAC_REGION_EU868;

static void prepareTxFrame( uint8_t port )
{
    appDataSize = 4;//AppDataSize max value is 64
    appData[0] = 0x00;
    appData[1] = 0x01;
    appData[2] = 0x02;
    appData[3] = 0x03;
}

// Add your initialization code here
void setup()
{
  if(mcuStarted==0)
  {
    LoRaWAN.displayMcuInit();
  }
  Serial.begin(57600);
  while (!Serial);
  SPI.begin(SCK,MISO,MOSI,SS);
  Mcu.init(SS,RST_LoRa,DIO0,DIO1,license);
  deviceState = DEVICE_STATE_INIT;
}

// The loop function is called in an endless loop
void loop()
{
  switch( deviceState )
  {
    case DEVICE_STATE_INIT:
    {
      LoRaWAN.init(loraWanClass,loraWanRegion);
      break;
    }
    case DEVICE_STATE_JOIN:
    {
      LoRaWAN.displayJoining();
      LoRaWAN.join();
      break;
    }
    case DEVICE_STATE_SEND:
    {
      LoRaWAN.displaySending();
      prepareTxFrame( appPort );
      LoRaWAN.send(loraWanClass);
      deviceState = DEVICE_STATE_CYCLE;
      break;
    }
    case DEVICE_STATE_CYCLE:
    {
      // Schedule next packet transmission
      txDutyCycleTime = appTxDutyCycle + randr( -APP_TX_DUTYCYCLE_RND, APP_TX_DUTYCYCLE_RND );
      LoRaWAN.cycle(txDutyCycleTime);
      deviceState = DEVICE_STATE_SLEEP;
      break;
    }
    case DEVICE_STATE_SLEEP:
    {
      LoRaWAN.displayAck();
      LoRaWAN.sleep(loraWanClass,debugLevel);
      break;
    }
    default:
    {
      deviceState = DEVICE_STATE_INIT;
      break;
    }
  }
}
```

The exception decoder says the following:
[image]

What am I doing wrong?
Thanks for your help,
Dani

archibald-picq, 2020-05-21:
Same issue here.

Digging into the code, seems it is a null pointer structure assignment in `LoRaMacConfirmQueueAdd()` because the `BufferEnd` variable is not initialized. Looking at your code, i'm trying to set the same region `LORAMAC_REGION_EU868` which leads me to the `LoRaMacInitialization()` returning error code `LORAMAC_STATUS_REGION_NOT_SUPPORTED` (which is not checked). Thus not initializing the aforementioned structure.

README say you should select the region in the Arduino IDE drop down menu. Unfortunately this menu doesn't show up for me. Since I'm not able to select it in the UI, I try to guess what define I should include but to no avail.

Still trying to figure out how to declare my region ...

zizou221190, 2021-04-05:
hello,
I have the exact same problem, could you find a solution?

thanks

tusharvb19, 2023-07-05:
Facing same issue...Help needed.

ESP-ROM:esp32s3-20210327
15:19:20.560 -> Build:Mar 27 2021
15:19:20.560 -> rst:0x5 (DSLEEP),boot:0x28 (SPI_FAST_FLASH_BOOT)
15:19:20.560 -> pro cpu reset by JTAG
15:19:20.560 -> SPIWP:0xee
15:19:20.560 -> mode:DIO, clock div:1
15:19:20.560 -> load:0x3fce3808,len:0x43c
15:19:20.560 -> load:0x403c9700,len:0xbec
15:19:20.560 -> load:0x403cc700,len:0x2a3c
15:19:20.560 -> SHA-256 comparison failed:
15:19:20.560 -> Calculated: dcde8d8a4817d9bf5d5d69a7247667264e4e10ac7493514868b61f5aa6146539
15:19:20.560 -> Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
15:19:20.560 -> Attempting to boot anyway...
15:19:20.560 -> entry 0x403c98d8

fbenti, 2023-11-22:
@tusharvb19 did you ever solve your issue?

AllTracking, 2023-11-26:
@tusharvb19 @fbenti It's crazy how Heltech is working, after their library was working fine they modified it to fix a bug related to validating the license of the Library, however, it seems that this modification damaged the main behavior of the LoRaWAN communication.

I took me a while after I saw that the crash happens during the downlink reception, hence I made a trick that works. In the **ESP32_LoRaWAN.cpp**, I defined a bool variable that becomes true at the end of this function as an indication that both UpLink and Downlink finished. In the main LoRaWAN cycle(inside the **main.cpp**) modify the **DEVICE_STATE_SLEEP** so it only goes to sleep after the boolean variable is set to true.

I believe there should be a group complaining about what Heltech did, they damaged the library that works with the V1 and V2 of their hardware leaving their customer only for the other open-source solutions(LMIC..etc).

fbenti, 2023-11-26:
Thanks for the answer, could you please show the code you added/modified? I'm a bit lost since I'm completely new to LoRa

AllTracking, 2023-11-26:
Sure,
In the ESP32_LoRaWAN.h define an Extern variable:

**extern bool LoRaWAN_Session;**      // To avoid the Crash after Receve a packet from the server

In the **ESP32_LoRaWAN.cpp** modify the variable like:

`/*!
 * \brief   MCPS-Indication event function
 *
 * \param   [IN] mcpsIndication - Pointer to the indication structure,
 *               containing indication attributes.
 */
int ackrssi;
static void McpsIndication( McpsIndication_t *mcpsIndication )
{
	if( mcpsIndication->Status != LORAMAC_EVENT_INFO_STATUS_OK )
	{
		return;
	}

	ackrssi = mcpsIndication->Rssi;

	#ifdef __LoRaWAN_Debug_Print__
		lora_printf( "receive data: rssi = %d, snr = %d, datarate = %d\r\n", 
												mcpsIndication->Rssi, 
												(int)mcpsIndication->Snr,
												(int)mcpsIndication->RxDatarate);
	#endif

	switch( mcpsIndication->McpsIndication )
	{
		case MCPS_UNCONFIRMED:
		{
			break;
		}
		case MCPS_CONFIRMED:
		{
			break;
		}
		case MCPS_PROPRIETARY:
		{
			break;
		}
		case MCPS_MULTICAST:
		{
			break;
		}
		default:
			break;
	}

	// Check Multicast
	// Check Port
	// Check Datarate
	// Check FramePending
	if( mcpsIndication->FramePending == true )
	{
		// The server signals that it has pending data to be sent.
		// We schedule an uplink as soon as possible to flush the server.
		OnTxNextPacketTimerEvent( );
	}
	// Check Buffer
	// Check BufferSize
	// Check Rssi
	// Check Snr
	// Check RxSlot
	if( mcpsIndication->RxData == false )
	{
		LoRaWAN_RxData = false;
	}
	else
	{
		downLinkDataHandle(mcpsIndication);
	}

	#ifdef	__LoRaWAN_Debug_Print__
		lora_printf( "McpsIndication: LoRaWAN_Session = true");
	#endif
	/*
		To avoid the Crash after receiving a packet from the server	
		Setting LoRaWAN_Session to true means that Uplink as well as Downlink are done.
	*/
	LoRaWAN_Session = true;			
}`

In your LoRaWAN_Cycle define the **DEVICE_STATE_SLEEP**:

`case DEVICE_STATE_SLEEP:
    {
      if(LoRaWAN_Session)
      {
        Radio.IrqProcess( );
        MySleepRutin();
      }
      else
      {
        LoRaWAN.sleep(loraWanClass, LoRaWAN_DEBUG_LEVEL);
      }
      
      break;
    }`
    
    As I don't use the Heltech LoRaWAN.sleep function(the reason for the crash), I define another one like:
    
`void MySleepRutin(void)
{
 esp_sleep_enable_timer_wakeup( use the tx period here in milliseconds );  // set the timer to wake

  esp_deep_sleep_start();
}`

I hope I didn't miss other codes as I was using this a long time ago, hope it works for you.