Heltec-Aaron-Lee/WiFi_Kit_series #121: update esptool

aparcar, 2020-03-17:
the currently used usptool version 2.6.1 is outdated and uses python 2. please update to 2.8 and male it use python 3

aaronjudd, 2021-06-11:
On macOS Big Sur, the outdated 'esptool' produces a fatal build error:

> Traceback (most recent call last):
>   File "esptool.py", line 57, in <module>
>   File "/Library/Python/2.7/site-packages/PyInstaller/loader/pyimod03_importers.py", line 389, in load_module
>   File "serial/tools/list_ports.py", line 29, in <module>
>   File "/Library/Python/2.7/site-packages/PyInstaller/loader/pyimod03_importers.py", line 389, in load_module
>   File "serial/tools/list_ports_posix.py", line 31, in <module>
>   File "/Library/Python/2.7/site-packages/PyInstaller/loader/pyimod03_importers.py", line 389, in load_module
>   File "serial/tools/list_ports_osx.py", line 32, in <module>
> ValueError: dlsym(RTLD_DEFAULT, kIOMasterPortDefault): symbol not found
> Failed to execute script esptool
> exit status 255
> /Applications/Arduino.app/Contents/Java/arduino-builder returned 255
> Error compiling for board Wireless Stick.
>

koedal, 2021-12-08:
I have the same error on macOS Monterey 12.0.1 still.

`ValueError: dlsym(RTLD_DEFAULT, kIOMasterPortDefault): symbol not found`

viotemp1, 2021-12-13:
Any way to solve this error on Big Sur
dlsym(RTLD_DEFAULT, kIOMasterPortDefault): symbol not found

koedal, 2021-12-13:
I fixed it in three steps:

1) Install command line tools from Xcode.

2) Download esptool.py file from the 2.6 release of the tool because that is what version of the ESP Tool Heltec is using. I then changed the #! to be python3 instead of python so it wouldn't use the Mac system version of Python 2.X and replaced the esptool binary with the esptool.py but removed the .py extension.

3) I used python3 and pip on the Mac to install pyserial
`cd /Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.8/lib/python3.8/site-packages

sudo python3 ./pip install --upgrade pyserial`