Heltec-Aaron-Lee/WiFi_Kit_series #128: Heltec esp32 wifi kit v2 - high Consumption deep sleep

harsA-85, 2020-05-19:
Hi all
I have setup my heltec board using deep sleep but I can't go under 0.0092A during sleep.

I thought deep sleep could power down the board to 10uA ?

Has someone achieved that kind of consumption, even using only the board and no sensors ?

harsA-85, 2020-05-19:
I've tried different things. thinking at first  esp_deep_sleep_start(); wasn't shutting down everything so I added in my code : 
  esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_OFF);
  esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_SLOW_MEM, ESP_PD_OPTION_OFF);
  esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_FAST_MEM, ESP_PD_OPTION_OFF);
  esp_sleep_pd_config(ESP_PD_DOMAIN_XTAL, ESP_PD_OPTION_OFF);
  esp_sleep_pd_config(ESP_PD_DOMAIN_MAX, ESP_PD_OPTION_OFF);

But I guess, it's already done by the deep_sleep because it doesn't change anything. 

Could my huge calculated sleep consumption come from the board powered via USB ?

xaionaro, 2023-12-16:
I'm also impacted by this issue. Any updated?