Heltec-Aaron-Lee/WiFi_Kit_series #129: Problem with tutorial for Ubuntu/Debian

vincegre, 2020-06-04:
Hi

I try to install that library in my Arduino IDE in Ubuntu 18.04 LTS but I have a problem at that step:

`git submodule update --init --recursive && \`

I get that error refering to Azure.... What's the problem there '?

```
vincen@Kubuntu-portable-de-Vincen:~/Arduino/hardware/heltec/esp32$ git submodule update --init --recursive
fatal: URL non trouvée pour le chemin de sous-module 'libraries/AzureIoT' dans .gitmodules
vincen@Kubuntu-portable-de-Vincen:~/Arduino/hardware/heltec/esp32$ 
```

which states that it could not find the URL for the sub-module AzureIoT !

Thanks

Vincèn

Heltec-Aaron-Lee, 2020-06-08:
We had removed the AzureIoT submodule.

This warning can be ignore.

vincegre, 2020-06-09:
Thanks for confirmation I could pass over that fatal error but it would be nice to clean the code so that fatal warning disappears ! And I still have exact same problem after that ! Nothing is added in Arduino IDE to select region and I get these errors when compiling:

```
Arduino: 1.8.12 (Linux), Board: "Heltec WiFi LoRa 32(V2), Disabled, default_8MB, 240MHz (WiFi/BT), QIO, 80MHz, 8MB (64Mb), 921600, None"

/home/vincen/.local/share/umake/ide/arduino/arduino-builder -dump-prefs -logger=machine -hardware /home/vincen/.local/share/umake/ide/arduino/hardware -hardware /home/vincen/.arduino15/packages -hardware /home/vincen/Arduino/hardware -tools /home/vincen/.local/share/umake/ide/arduino/tools-builder -tools /home/vincen/.local/share/umake/ide/arduino/hardware/tools/avr -tools /home/vincen/.arduino15/packages -built-in-libraries /home/vincen/.local/share/umake/ide/arduino/libraries -libraries /home/vincen/Arduino/libraries -fqbn=esp32:esp32:heltec_wifi_lora_32_V2:PSRAM=disabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=8M,UploadSpeed=921600,DebugLevel=none -ide-version=10812 -build-path /tmp/arduino_build_624228 -warnings=default -build-cache /tmp/arduino_cache_974471 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=/home/vincen/.arduino15/packages/esp32/tools/mkspiffs/0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=/home/vincen/.arduino15/packages/esp32/tools/mkspiffs/0.2.3 -prefs=runtime.tools.esptool_py.path=/home/vincen/.arduino15/packages/esp32/tools/esptool_py/2.6.1 -prefs=runtime.tools.esptool_py-2.6.1.path=/home/vincen/.arduino15/packages/esp32/tools/esptool_py/2.6.1 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=/home/vincen/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-80-g6c4433a-5.2.0.path=/home/vincen/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-80-g6c4433a-5.2.0 -verbose /home/vincen/Arduino/libraries/ESP32_LoRaWAN-master/examples/OTAA_OLED/OTAA_OLED.ino
/home/vincen/.local/share/umake/ide/arduino/arduino-builder -compile -logger=machine -hardware /home/vincen/.local/share/umake/ide/arduino/hardware -hardware /home/vincen/.arduino15/packages -hardware /home/vincen/Arduino/hardware -tools /home/vincen/.local/share/umake/ide/arduino/tools-builder -tools /home/vincen/.local/share/umake/ide/arduino/hardware/tools/avr -tools /home/vincen/.arduino15/packages -built-in-libraries /home/vincen/.local/share/umake/ide/arduino/libraries -libraries /home/vincen/Arduino/libraries -fqbn=esp32:esp32:heltec_wifi_lora_32_V2:PSRAM=disabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=8M,UploadSpeed=921600,DebugLevel=none -ide-version=10812 -build-path /tmp/arduino_build_624228 -warnings=default -build-cache /tmp/arduino_cache_974471 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=/home/vincen/.arduino15/packages/esp32/tools/mkspiffs/0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=/home/vincen/.arduino15/packages/esp32/tools/mkspiffs/0.2.3 -prefs=runtime.tools.esptool_py.path=/home/vincen/.arduino15/packages/esp32/tools/esptool_py/2.6.1 -prefs=runtime.tools.esptool_py-2.6.1.path=/home/vincen/.arduino15/packages/esp32/tools/esptool_py/2.6.1 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=/home/vincen/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-80-g6c4433a-5.2.0.path=/home/vincen/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-80-g6c4433a-5.2.0 -verbose /home/vincen/Arduino/libraries/ESP32_LoRaWAN-master/examples/OTAA_OLED/OTAA_OLED.ino
Using board 'heltec_wifi_lora_32_V2' from platform in folder: /home/vincen/.arduino15/packages/esp32/hardware/esp32/1.0.4
Using core 'esp32' from platform in folder: /home/vincen/.arduino15/packages/esp32/hardware/esp32/1.0.4
Detecting libraries used...
/home/vincen/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-g++ -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H -DGCC_NOT_5_2_0=0 -DWITH_POSIX -I/home/vincen/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/config -I/home/vincen/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/app_trace -I/home/vincen/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/app_update -I/home/vincen/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/asio -I/home/vincen/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/bootloader_support -I/home/vincen/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/bt -I/home/vincen/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/coap -I/home/vincen/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/console -I/home/vincen/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/driver -I/home/vincen/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp-tls -I/home/vincen/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp32 -I/home/vincen/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp_adc_cal -I/home/vincen/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp_event -I/home/vincen/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp_http_client -I/home/vincen/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp_ht [... cut ...] cen/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/libsodium -I/home/vincen/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/log -I/home/vincen/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/lwip -I/home/vincen/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/mbedtls -I/home/vincen/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/mdns -I/home/vincen/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include

Heltec-Aaron-Lee, 2020-06-11:
Please install the newest development framework.

vincegre, 2020-06-11:
Yep but no offense but with all these repositories and documentations on your own website it's just a complete mess to understand which documentation is the  good one ! I had to delete all libraries installed in my Arduino IDE to succeed  to get it to work with your framework !! Could you at least close all these repos that are useless now instead of letting them online and confusing completely customers ??

Heltec-Aaron-Lee, 2020-06-11:
Very sorry about that, this is absolutely a document error that causes so much for your development.

I had fixed the document, please refer to the new [readme.md](https://github.com/HelTecAutomation/Heltec_ESP32/blob/master/README.md) in Heltec_ESP32 library, I believe it will make clear sense to you.

Thank you for your feedback. Your feedback makes things become better.