Heltec-Aaron-Lee/WiFi_Kit_series #131: SSD1306 FreeRTOS - Corrupt Display

DougArmstrong, 2020-07-07:
The SSD1306 display on my Wifi32 boards become corrupt when I use FreeRTOS tasks. This might be just unsupported or I might just be doing something obviously wrong.  Assuming not, I prepared a test case that I think isolates the problem. 

Board: Heltec Wifi32 V1 & V2 boards
Build Env: Vmicro/Visual Studio
Problem: If the SSD1306 Heltec driver is called from inside a FreeRTOS task the display is immediately corrupted. Much testing has been done with stack size and task priority.  I the code below I believe I have a simplified test case.  Buildable demo in the ZIP, just changes pasted.

Work Around: If the display written from the start up (Arduino) thread, the display works.

I modified the canned Heltec 1306 Simple demo to create a FreeRTOS task that calls the demos, it crashes the display in the same was it does in my apps.

Thanks in advance for any assistance on this.  I hope the sample code is helpful.

[SSD1306FreeRTOSDemo.zip](https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series/files/4887491/SSD1306FreeRTOSDemo.zip)

The main test snippet follows:

void RunDemo(void* pvParameters) {
    while (true) {
        // clear the display
        Heltec.display->clear();
        // draw the current demo method
        demos[demoMode]();

        Heltec.display->setTextAlignment(TEXT_ALIGN_RIGHT);
        Heltec.display->drawString(10, 128, String(millis()));
        // write the buffer to the display
        Heltec.display->display();

        if (millis() - timeSinceLastModeSwitch > DEMO_DURATION) {
            demoMode = (demoMode + 1) % demoLength;
            timeSinceLastModeSwitch = millis();
        }
        counter++;
        delay(10);
    }
}

TaskHandle_t UpdateDisplayTask_Handle = NULL;
BaseType_t UpdateDisplayTask_RetCode;
void loop()
{

    UpdateDisplayTask_RetCode = xTaskCreate(RunDemo, "OLEDdisplay", 8000,
	NULL, 1, &UpdateDisplayTask_Handle);

    vTaskDelete(NULL);
}

richardjlyon, 2020-10-03:
I have the same issue on a Wemos TTGO-LoRa ESP32 V2_1.6

DougArmstrong, 2020-10-03:
I ran this down.  I think that the memory allocated to the display is freed when the Arduino start up thread is terminated.  If you do not allow the start up thread to terminate all appears well.  The corruption appears to be caused by reuse of that memory. I imagine all kinds of other issues abound if one terminates the startup thread, perhaps someone more knowledgeable can fill in the details.

richardjlyon, 2020-10-03:
Thanks Doug. I've tried inserting a while (true); at the end of setup() but it's still garbled . I'm using the Adafruit_SSD1306 library, I might try some others. (I thought loop() had to run to let Arduino do its own housekeeping, but I might be wrong on that.)

DougArmstrong, 2020-10-03:
Holding in Setup should work too.  I ended up putting messages in a Queue and having the main loop display them, everything else done in other tasks.  In my case setup messages were displaying but once the main thread terminated, things went bad.  I was using the Heltec version of the driver.  I wanted to switch to Adafruit since it is more generic but never got that working.  I recommend throwing a little screen test together and trying it from different points to isolate what breaks it.  Good Luck!

DougArmstrong, 2021-06-15:
> Hii, every one...
>
> I have the same problem, I can't understand why it works in the loop and not in the created task.
>
> the display gets a lot of bugs...

I think the memory buffer gets allocated in the main thread and the memory is freed when the main task terminates but the driver keeps using the buffer after it is reallocated. 

If the main task is not permitted to terminate all is well. Curios to know if you get the same results.

mdg1019, 2021-06-19:
I had the same issue. I was deleting the empty loop() task:

`void loop() {
  vTaskDelete(NULL);
}`

It started working fine by just suspending the loop() task:

`void loop() {
  vTaskSuspend(NULL);
}`