Heltec-Aaron-Lee/WiFi_Kit_series #136: Error following InstallGuide/mac.md

paulvivier, 2020-10-26:
I'm getting an error with these instructions:
https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series/blob/master/InstallGuide/mac.md

Specifically:
`git submodule update --init --recursive`
produces:
`fatal: No url found for submodule path 'libraries/AzureIoT' in .gitmodules`

Can you help?

madmacks59, 2020-10-26:
I am having the same issue. And the Arduino board manager process fails as well.

madmacks59, 2020-11-17:
Is this getting any attention?

BabyFnord, 2021-04-25:
Just as well,
`git submodule update --init --recursive`
returns 

> Cloning into '/Users/AAAA/Documents/Arduino/hardware/heltec/esp8266/libraries/LittleFS/lib/littlefs'...
remote: Not Found
fatal: repository 'https://github.com/esp8266/Arduino/libraries/LittleFS.git/' not found
fatal: clone of 'https://github.com/esp8266/Arduino/libraries/LittleFS.git' into submodule path '/Users/AAAA/Documents/Arduino/hardware/heltec/esp8266/libraries/LittleFS/lib/littlefs' failed
Failed to clone 'esp8266/libraries/LittleFS/lib/littlefs'. Retry scheduled
Cloning into '/Users/AAAA/Documents/Arduino/hardware/heltec/esp8266/libraries/LittleFS/lib/littlefs'...
remote: Not Found
fatal: repository 'https://github.com/esp8266/Arduino/libraries/LittleFS.git/' not found
fatal: clone of 'https://github.com/esp8266/Arduino/libraries/LittleFS.git' into submodule path '/Users/AAAA/Documents/Arduino/hardware/heltec/esp8266/libraries/LittleFS/lib/littlefs' failed
Failed to clone 'esp8266/libraries/LittleFS/lib/littlefs' a second time, aborting

Checking whether https://github.com/esp8266/Arduino/libraries/LittleFS.git exists is negative (Not Found). Then again, https://github.com/esp8266/Arduino/tree/master/libraries/LittleFS does exist, we just need to change that address somewhere. But I'm not savvy enough with this to know, where to change it.

birdybro, 2022-02-15:
Github cannot git recursively to subfolders, that's why. This is probably something that used to be supported, but isn't any longer. @Heltec-Aaron-Lee should update this to point to possibly a custom repository with a mirror of littlefs with only the details they need.

craigcassidy, 2022-11-08:
I am in the same situation and have run though everything I have found in the issues here to try to get this to work. If there is a workaround please inform otherwise these boards are getting returned.

BabyFnord, 2022-11-08:
As birdybro pointed out, this repo _is_ abandoned. All reported issues are years old. My 2¢: Capable folks would rather commit to community projects rather than abandoned niche commercial ones … don't waste your time hoping for support or further development here 🤷🏻‍♂️.

craigcassidy, 2022-11-08:
Thanks @BabyFnord, fair enough.

Baptou88, 2022-11-08:
it's a shame that this repository is not updated, it makes the task complicated for inexperienced developers (including myself)

craigcassidy, 2022-11-08:
This repo is heavily referenced when purchasing the MakerFocus boards online (Amazon and their website) so it is a great disappointment for those of us who just purchased a batch of them. I've wasted a lot of time on it so far but, if I find a path forward today i'll post it here for posterity. If not I can still return them by tomorrow.