Heltec-Aaron-Lee/WiFi_Kit_series #150: Please update to latest espressif code

clarkbremer, 2021-11-24:
I need to use features like:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/bluetooth/esp_gap_bt.html?highlight=esp_bt_gap_read_remote_name#_CPPv427esp_bt_gap_read_remote_name13esp_bd_addr_t
But they are not found in your version.

MicSchi, 2021-12-24:
Hello!

I have a similar problem. I would like to use [TimerInterrupt_Generic](https://github.com/khoih-prog/TimerInterrupt_Generic) but this requires framework-arduinoespressif32 v2.0.1+.

Would be very nice if you can consider update to the current framework. Or maybe someone can point out if there is another way to get around this.

gasagna, 2023-06-04:
Apologies for resurrecting this old post, but wanted to add my 2 cents. @Heltec-Aaron-Lee What is the use of this repo? Why not adding the variants defined in this repo to the official arduino-esp32 repo? there are already some boards there that have been added by users of your products. Having all esp-based Heltec board directly supported within the official repo is a very good thing because it gives access to the latest version of the esp32 core, with bug fixes and new features.