Heltec-Aaron-Lee/WiFi_Kit_series #153: GitHub install directions broken - littlefs submodule fails to init

ObviousInRetrospect, 2022-03-31:
$ git submodule update --init --recursive
Submodule 'esp32/libraries/ESP32_AzureIoT_Arduino' (https://github.com/VSChina/ESP32_AzureIoT_Arduino.git) registered for path 'libraries/ESP32_AzureIoT_Arduino'
Submodule 'esp8266/libraries/ESP8266SdFat' (https://github.com/earlephilhower/ESP8266SdFat.git) registered for path '../esp8266/libraries/ESP8266SdFat'
Submodule 'esp8266/libraries/LittleFS/lib/littlefs' (https://github.com/esp8266/Arduino/libraries/LittleFS.git) registered for path '../esp8266/libraries/LittleFS/lib/littlefs'
Submodule 'esp8266/libraries/SoftwareSerial' (https://github.com/PaulStoffregen/SoftwareSerial.git) registered for path '../esp8266/libraries/SoftwareSerial'
Cloning into '/Users/gbr4/Documents/Arduino/hardware/heltec/esp32/libraries/ESP32_AzureIoT_Arduino'...
Cloning into '/Users/gbr4/Documents/Arduino/hardware/heltec/esp8266/libraries/ESP8266SdFat'...
Cloning into '/Users/gbr4/Documents/Arduino/hardware/heltec/esp8266/libraries/LittleFS/lib/littlefs'...
remote: Not Found
fatal: repository 'https://github.com/esp8266/Arduino/libraries/LittleFS.git/' not found
fatal: clone of 'https://github.com/esp8266/Arduino/libraries/LittleFS.git' into submodule path '/Users/gbr4/Documents/Arduino/hardware/heltec/esp8266/libraries/LittleFS/lib/littlefs' failed
Failed to clone 'esp8266/libraries/LittleFS/lib/littlefs'. Retry scheduled
Cloning into '/Users/gbr4/Documents/Arduino/hardware/heltec/esp8266/libraries/SoftwareSerial'...
Cloning into '/Users/gbr4/Documents/Arduino/hardware/heltec/esp8266/libraries/LittleFS/lib/littlefs'...
remote: Not Found
fatal: repository 'https://github.com/esp8266/Arduino/libraries/LittleFS.git/' not found
fatal: clone of 'https://github.com/esp8266/Arduino/libraries/LittleFS.git' into submodule path '/Users/gbr4/Documents/Arduino/hardware/heltec/esp8266/libraries/LittleFS/lib/littlefs' failed
Failed to clone 'esp8266/libraries/LittleFS/lib/littlefs' a second time, aborting

goblin, 2022-03-31:
Yah, I had the same issue.

From memory, I fixed this with something similar to this:

```
git rm esp8266/libraries/LittleFS/lib/littlefs
git submodule add https://github.com/esp8266/Arduino.git tmp_modules/Arduino
git submodule update --init --recursive
cd esp8266/libraries/LittleFS/lib
ln -s ../../../../tmp_modules/Arduino/libraries/LittleFS/lib/littlefs ./
```

Also, if you're checking out from git, switch to the 0.0.5 tag. Master branch (just a bit ahead of 0.0.6) was making my dev board panic with "Guru meditation errors" when issuing `SerialBT.begin`.

brio50, 2022-05-05:
If using git for windows:

**OPEN GIT FOR WINDOWS AS ADMINISTRATOR**
```
git config --global core.symlinks true
git clone https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series.git ~/Documents/Arduino/hardware/heltec
cd ~/Documents/Arduino/hardware/heltec
git rm esp8266/libraries/LittleFS/lib/littlefs
git submodule add https://github.com/esp8266/Arduino.git tmp_modules/Arduino
git submodule update --init --recursive
cd esp8266/libraries/LittleFS/lib
cmd <<< "mklink /D littlefs ..\..\..\..\tmp_modules\Arduino\libraries\LittleFS\lib\littlefs"
```

default-student, 2023-08-24:
> Yah, I had the same issue.
>
> From memory, I fixed this with something similar to this:
>
> ```
> git rm esp8266/libraries/LittleFS/lib/littlefs
> git submodule add https://github.com/esp8266/Arduino.git [...]

I was unable to upload the sketch as it failed while connecting.
This and choosing the correct PORT fixed it for me.

schwarzandy82-create, 2026-01-08:
is there an updated thread to this ??