Heltec-Aaron-Lee/WiFi_Kit_series #156: RadioLib Build for WiFi_LoRa_32_V3 fails

MrSniffer, 2022-10-31:
Using the current files, RadioLib can not communicate with the SX1262 and initialize it. The problem appears in pins_arduino.h for the WiFi_LoRa_32_V3. It may also be wrong for other Heltec parts; I dunno.

The correct defs for the SPI GPIOs should be:

static const uint8_t SS    = 8;
static const uint8_t MOSI  = 10;
static const uint8_t MISO  = 11;
static const uint8_t SCK   = 9;

N.B. SDA and SCL are not brought out on pins and probably should not be def'd.

JJJS777, 2023-02-02:
Hello @MrSniffer, have you found a solution for this? Thanks in advance!

MrSniffer, 2023-02-02:
I no longer rely on pins_arduino.h, so I don't know if Heltec has corrected it or not. I use my own .h files and #ifdef switches.

MrSniffer, 2023-02-03:
BTW: The file which needs changing is pins_arduino.h in:
C:\Users\yourname.platformio\packages\framework-arduinoespressif32\variants\heltec_WIFI_LoRa_32_V3

JJJS777, 2023-02-03:
thanks for the fast replay. I'm new to this, so maybe im wrong^^

what im trying to do is to run the RadioLib SX126x Transmit and Receive Example to check if everything is working before i start my project.   

The [radiolib ex](https://github.com/jgromes/RadioLib/tree/master/examples/SX126x) says:   

```
SX1262 has the following connections:
NSS pin:   8
DIO1 pin:  14
NRST pin:  12
BUSY pin:  13
SX1262 radio = new Module(8, 14, 12, 13);
```

of course the pins a wrong so i search the right ones in the [schematic Diagram](https://resource.heltec.cn/download/WiFi_LoRa32_V3/HTIT-WB32LA(F)_V3_Schematic_Diagram.pdf), i found: 

```
nss pin: 8
rst pin: 12 //same as nrst?
dio1 pin: 14
sck pin: 9
miso pin: 11
mosi pin: 10
busy pin: 13

```  

because the heltec board doesn’t work with the standard SPI setup one should use the [extended Module class constructor](https://github.com/jgromes/RadioLib/wiki/Basics#non-standard-spi-setup) in the format ```Module(cs, irq, rst, gpio, &spi, spiSettings)```, where ```spi``` is a reference to an instance of ```SPIClass```, and ```spiSettings``` is an instance of class ```SPISettings```.

Maybe this is also why the LoRa chip can’t be initialized? 
Did u use or try the extended Module class constructor?
I don't know how to deal with:
```cpp 
SPISettings spiSettings(2000000, MSBFIRST, SPI_MODE0);
SX1278 radio = new Module(cs, irq, rst, gpio, spi, spiSettings);
```

Maybe u have an idea or can share your thoughts on this (::

MrSniffer, 2023-02-03:
Not clear what is causing your problem. 

I initialize the V3 board with SX1262 radio = new Module(8, 14, 12, 13);
Make sure that you have set 
static const uint8_t SS = 8;    // NOTE: uses SS, not NSS!
static const uint8_t MOSI = 10;
static const uint8_t MISO = 11;
static const uint8_t SCK = 9;
in the correct pins_arduino.h file (location shown above.)
And that you have the right variant in platformio.ini and other settings for making it an S3 chip, 8MB Flash, etc. I use these options:
[env:WSL_SX1262]
platform = espressif32
board = heltec_wireless_stick_lite
board_upload.maximum_size = 1310720
board_upload.maximum_data_size = 327680
board_build.target = esp32s3
board_build.mcu = esp32s3
board_build.variant = heltec_WIFI_LoRa_32_V3
board_build.flash_size = 8MB
board_upload.flash_size = 8MB
board_build.partitions = config_8MB.csv     // my custom partitions.....
board_build.f_flash = 80000000L

I have never used the SPIclass approach so I can't comment on it.

JJJS777, 2023-02-04:
I see - will try that config next. I also got it to work for me with a different board, this is my platformio config

```cpp
[env:esp32dev]
platform = espressif32
board = esp32-s3-devkitc-1
framework = arduino
lib_deps = 
	olikraus/U8g2@^2.34.13
	jgromes/RadioLib@^5.6.0
```

In the Radiolib example ```SPI.begin``` does not get used. 
The pins u suggested worked for me.

```cpp
#define ss 8
#define lora_mosi 10
#define lora_miso 11
#define lora_sck 9  

#define lora_nss 8
#define dio1 14
#define lora_nrst 12
#define lora_busy 13

SX1262 radio = new Module(lora_nss, dio1, lora_nrst, lora_busy);

void setup(){

Serial.begin(9600);
SPI.begin(lora_sck, lora_miso, lora_mosi, lora_nss);  

(...)

}
```

btw, the [pins_arduino.h](https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series/commit/b624b342921ac69caf7fb4fa8ca8c4e38e02748f) file got changed to the right pins a few days ago.
Thanks for your help (::

PR #167 (2023-06-21): Add missing macro values to V3 boards
Without macros defined like `DISPLAY_HEIGHT`, `DISPLAY_WIDTH` and `DIO0`, features on the V3 boards won't compile properly.

This PR corrects the definitions for: `WIFI_LoRa_32_V3`, `Wireless_Stick_V3`, and `WIFI_Kit_32_V3`

This should help fix #156 fix #164 and help with #157 and #161

Thanks!
closed 2023-06-21 by Heltec-Aaron-Lee