Heltec-Aaron-Lee/WiFi_Kit_series #159: Wifi_Lora_32_ v3 bootloop with  version  0.0.7

heikodemlang, 2022-12-22:
Seems to be a problem with the spi interface.
Arduino IDE Version 2.0.3.
After upload a sketch I get this output log:  

_ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x400454d5
SPIWP:0xee
mode:QIO, clock div:2
load:0x3fce3808,len:0x43c
ets_loader.c 78_

mode QIO is not supported should be DIO

mckay6842, 2022-12-29:
Hello @heikodemlang,

have you been able to find a solution / workaround yet?

As a comparison, here is the (working) boot output of the preflashed firmware that comes with the device:

```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x9 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x43c
load:0x403c9700,len:0xbec
load:0x403cc700,len:0x2a3c
SHA-256 comparison failed:
Calculated: dcde8d8a4817d9bf5d5d69a7247667264e4e10ac7493514868b61f5aa6146539
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
entry 0x403c98d8

sending packet "hello 1,Rssi:0" , length 14
TX done......into RX mode

```

wotzlaff, 2023-01-18:
I am facing the same problem. The issue seems to be related to version 2.0.4 of the Espressif ESP32 framework, see also [here](https://github.com/espressif/arduino-esp32/issues/6980#issuecomment-1184275356) and the full discussion there.

My workaround was to replace
https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series/blob/eacfe30c618e6c26007a986286f0531c18f7b665/esp32/boards.txt#L64
by
```
WIFI_LoRa_32_V3.build.boot=dio
```
and downgrade the file [esp32/tools/sdk/esp32s3/bin/bootloader_dio_80m.bin](https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series/blob/eacfe30c618e6c26007a986286f0531c18f7b665/esp32/tools/sdk/esp32s3/bin/bootloader_dio_80m.bin) to version 2.0.3 from [here](https://github.com/espressif/arduino-esp32/blob/2.0.3/tools/sdk/esp32s3/bin/bootloader_dio_80m.bin).

Probably there is an more adequate way to solve the problem. But since I have no idea how Arduino works and I was just trying to reflash the factory example, this seems to be a feasible workaround...

tauesdospi, 2023-01-25:
Hi,
I realized that if you upgrade esptool_py from 3.3.0 to 4.4 , then firmware can be uploaded correctly to V3 device and no more boot loop or errors when working in Arduino.

I think the problem is in the file  "package_esp32_index.template.json" which installs esptool_py version 3.3.0 and should be upgraded to install the esptool_py 4.4 or newer

Regards
Antonio

fred777, 2023-02-28:
I can confirm this issue and resolved it on my Archlinux machine by replacing `~/.arduino15/packages/Heltec-esp32/tools/esptool_py/3.3.0/esptool.py` with a symlink to `/bin/esptool.py` which currently is v4.4

c64emulator, 2023-04-16:
> I can confirm this issue and resolved it on my Archlinux machine by replacing `~/.arduino15/packages/Heltec-esp32/tools/esptool_py/3.3.0/esptool.py` with a symlink to `/bin/esptool.py` which currently is v4.4

This did not work for me on Linux Mint 21.1. Here is how I resolved:
First I tried to `pip install esptool` and linking `~/.arduino15/packages/Heltec-esp32/tools/esptool_py/3.3.0/esptool.py` to `/usr/local/bin/esptool.py`. I got error messages during compilation.
I then downloaded the latest source package [https://github.com/espressif/esptool/archive/refs/tags/v4.5.1.zip],  extracted to `~/.arduino15/packages/Heltec-esp32/tools/esptool_py/esptool-4.5.1`  and symlinked the whole directory (`ln -s esptool-4.5.1 3.3.0`).
Heureka!

gmbo, 2023-04-26:
Vielen Dank für die Info.
Ich hatte genau das Problem mit der Loopschleife.
Der Code zum Beispiel Chip-ID ließ sich mit EspS3Devkit laden und ausführen aber mit der Heltecesp kam der Loop.
Ich habe daraufhin den Ordner esptool_py mit dem aus der espressif esp tools ersetzt. Daraufhin bekam ich den Fehler Python3 kann 3.3.0 nicht finden. Daher habe ich neben der jetzt 4.5.1 einen symbolischen Link mit 
`ln -s 4.5.1 3.3.0`
erzeugt und das Programm startet nach dem Laden einwandfrei und gibt die ID aus.
Das mit der Lizenz habe ich noch nicht verstanden, kann bei Heltec mit der ID aber eine Lizenz erzeugen. Habe aber auch bisher noch nicht weiter getestet, da das Board erst gestern hier ist und alle Programme in die Schleife liefen.

Wie gesagt vielen Dank mir haben diese Infos geholfen.

Thanks for the info.
I had exactly the problem with the loop.
The code for example Chip-ID could be loaded and executed with EspS3Devkit but with the Heltecesp the loop came.
I replaced the folder esptool_py with the one from espressif esp tools. Thereupon I got the error Python3 can not find 3.3.0. So I created a symbolic link next to the now 4.5.1 with 
`ln -s 4.5.1 3.3.0`
and the program starts correctly after loading and displays the ID.
I have not yet understood the license, but I can create a license at Heltec with the ID. But have not yet tested further, because the board is here only yesterday and all programs ran in the loop.

As I said many thanks to me have helped this info.

Translated with www.DeepL.com/Translator (free version)

gmbo, 2023-04-26:
Habe direkt mal das Lorawan Beispiel geladen und auch das klappt sofort.
Ich hatte das wie in dem [Video von Alex ](https://www.aeq-web.com/heltec-lora32-v3-board-arduino-ide-lorawan-setup/) schon eingerichtet und dann eben nur die Schleife gesehen, als Devbord nicht kompilierbar, aber mit dem esptool_py 4.5.1 überträgt es die Daten wie gewünscht.

Have directly times the Lorawan example loaded and also that works immediately.
I had that as in the [Video of Alex ](https://www.aeq-web.com/heltec-lora32-v3-board-arduino-ide-lorawan-setup/) already set up and then just seen the loop, as Devbord not compilable, but with the esptool_py 4.5.1 it transfers the data as desired.

TheDevMinerTV, 2023-06-03:
The following commands fixed the issue for me:
```
mv ~/.arduino15/packages/Heltec-esp32/tools/esptool/3.3.0 ~/.arduino15/packages/Heltec-esp32/tools/esptool/3.3.0.bak
wget https://github.com/espressif/esptool/archive/refs/tags/v4.6.1.zip
unzip v4.6.1.zip
rm v4.6.1.zip
mv esptool-4.6.1 ~/.arduino15/packages/Heltec-esp32/tools/esptool/3.3.0
```

> Ubuntu 23.04

juanjimenez04, 2023-12-05:
Hi,
I am working on a project with Wireless Stick Lite V3 and Arduino IDE 2.2.1 on Windows 10, everything was working very well until a couple of months ago when the problem started. I tried several stick lite (I have 25) and now every time I recompile and upload the code the bootloop appears. I also found that if I upload a small code from one of the examples it works, but if I upload the factory test or my code it doesn't work. As indicated in https://github.com/HelTecAutomation/Heltec_ESP32/issues/106, I have tried replacing version 3.3.0 of esptool with versions 4.2 through 4.6.2, none of them work. I also tried @wotzlaff's solution but it didn't work. I see in all the comments that you have solved it in the Arduino IDE with Linux, not with Windows. Does anyone have information on how to solve it in Windows? I am very worried because I have 25 sensors that use the Wireless Stick Lite to transmit data via LORA and I have to update the code of all of them. Thanks.

juanjimenez04, 2023-12-07:
Thanks @fred777, I will try. I read that PlatformIO is more complete and convenient for this type of complex projects. It seems I can port the code without problems. Is there anything special I should take into account?

commit 83d049f1 (2024-03-19): Update esptool version in Arduino package template to fix #159
commit 90bc78ab (2024-03-19): Merge pull request #240 from aeggerth/master

Update esptool version in Arduino package template to fix #159
closed 2024-03-19 by Heltec-Aaron-Lee