Heltec-Aaron-Lee/WiFi_Kit_series #162: LoraWan102 V3 : comilation error on time.h ==> missing time.c

hupiv, 2023-01-23:
Hello,

I trie to compile with Visual Code for the loraWan102 library and when the programm is linking (ld) I have this error below. I checked and it is missing the LoraWan102/src/driver/time.c 

`Indexing .pio/build/esp32-s3-devkitc-1/libFrameworkArduino.a
Linking .pio/build/esp32-s3-devkitc-1/firmware.elf
/Users/robino/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pio/build/esp32-s3-devkitc-1/liba44/libLoraWan102.a(LoRaMac.c.o):(.literal.ProcessMacCommands+0x30): undefined reference to `TimerGetSysTime'
/Users/robino/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pio/build/esp32-s3-devkitc-1/liba44/libLoraWan102.a(LoRaMac.c.o):(.literal.ProcessMacCommands+0x34): undefined reference to `TimerSetSysTime'
/Users/robino/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pio/build/esp32-s3-devkitc-1/liba44/libLoraWan102.a(LoRaMac.c.o):(.literal.ProcessMacCommands+0x38): undefined reference to `TimerGetCurrentTime'`

On ArduinoIDE it compiles perfectly but it seems it is making link on SDK/libraries and no issue with the time.c missing file but how I could resolve this with Visual Studio Code ?

AbTrax, 2023-01-24:
I have been having this same issue except I am using the V2 wifi Lora board, I am currently trying to find ways to fix it if you do find a way to restore it please update me.

hupiv, 2023-01-24:
on my platformio.ini I force the use of the lib heltec.a (-l heltec -L…/sdk/libraries)
but after this I have an error on undefined Radio

AbTrax, 2023-01-24:
How i fixed it was by fixing the pins_header. but I too am getting the Undefiened Radio error.

hupiv, 2023-01-25:
I don't know for you but on ArduinoIDE 2.0 it works perfectly with the same libs so I tried to find which library has the radio.o but not found ... :-(

AbTrax, 2023-01-25:
yeah, same here i cant find why it doesn't work for platformio.

hupiv, 2023-01-25:
Good news :-) in the file radio.c there is `#if defined(WIFI_LoRa_32_V3)||defined(Wireless_Mini_Shell)` I just comment it (with #end if at the end of the file and it works nows (with warnings... but it linked the project).
Or you can add in the platformio.ini
`-D WIFI_LoRa_32_V3`

hupiv, 2023-01-28:
in the file radio.c there is #if defined(WIFI_LoRa_32_V3)||defined(Wireless_Mini_Shell

closed 2023-01-28 by hupiv